esphome/issues #3142: Updating to 2022.3.0 breaks dsmr

sjakief, 2022-03-16:
### The problem

I updated ESP within Home assistant
this is what I got in the logs:
INFO Reading configuration /config/esphome/slimmelezer.yaml...
ERROR Unable to load component dsmr.sensor:
Traceback (most recent call last):
  File "/esphome/esphome/loader.py", line 162, in _lookup_module
    module = importlib.import_module(f"esphome.components.{domain}")
  File "/usr/lib/python3.9/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 790, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/config/esphome/.esphome/external_components/d0b8a1aa/esphome/components/dsmr/sensor.py", line 32, in <module>
    cv.Optional("energy_delivered_lux"): sensor.sensor_schema(
TypeError: sensor_schema() takes from 0 to 1 positional arguments but 5 were given
INFO Detected timezone 'Europe/Amsterdam'
Failed config

sensor.dsmr: [source /config/esphome/slimmelezer.yaml:102]
  
  Platform not found: 'sensor.dsmr'.

### Which version of ESPHome has the issue?

2022.3.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.3.5

### What platform are you using?

ESP8266

Calimerorulez, 2022-03-16:
Plus one over here...

sjakief, 2022-03-16:
Hi Maurice,
that did the trick, I removed the whole section "external component" and  I was able to update.
Thanks for looking into this problem and thanks for the solution!

mmakaay, 2022-03-16:
@Calimerorulez does that solve it for you too? Then we can close this issue.

Calimerorulez, 2022-03-17:
@mmakaay 

Well, if I remove:

```
external_components:
  - source:
      type: git
      url: https://github.com/mmakaay/esphome
      ref: fix-dsmr-read-chunk-size
    components: [ "dsmr" ]
    refresh: 60s
```

I'm getting an error later in the code'
'platform not found: sensor.dsmr' 

I'm using the external_component for the checksum mismatch of the DSMR telegrams, because the 8266 is too slow sometimes to parse them correctly, as I understood.

jsuanet, 2022-03-17:
@Calimerorulez 

No need to use this external_component anymore, this was a bug fix which is merged in the official version months ago.  

> 'platform not found: sensor.dsmr'

This looks like an indentation error in your YAML, can you post your current yaml file?

Calimerorulez, 2022-03-17:
@jsuanet 

```
---
substitutions:
  device_name: slimmelezer
  device_description: "DIY P1 module to read your smart meter"
    
esphome:
  name: ${device_name}
  comment: "${device_description}"
  platform: ESP8266
  esp8266_restore_from_flash: true
  board: d1_mini
  name_add_mac_suffix: false
  project:
    name: zuidwijk.slimmelezer
    version: "2.0"
  on_boot:
    then:
      - if:
          condition:
            lambda: return id(has_key);
          then:
            - lambda: |-
                std::string key(id(stored_decryption_key), 32);
                id(dsmr_instance).set_decryption_key(key);
          else:
            - logger.log:
                level: info
                format: "Not using decryption key. If you need to set a key use Home Assistant service 'ESPHome:  ${device_name}_set_dsmr_key'"

wifi:
  networks:
    - ssid: !secret wifi_ssid
      password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: ${device_name}

#improv_serial:

#external_components:
#  - source:
#      type: git
#      url: https://github.com/mmakaay/esphome
#      ref: fix-dsmr-read-chunk-size
#    components: [ "dsmr" ]
#    refresh: 60s

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:
  services:
    service: set_dsmr_key
    variables:
      private_key: string
    then:
      - logger.log:
          format: Setting private key %s. Set to empty string to disable
          args: [private_key.c_str()]
      - globals.set:
          id: has_key
          value: !lambda "return private_key.length() == 32;"
      - lambda: |-
          if (private_key.length() == 32)
            private_key.copy(id(stored_decryption_key), 32);
          id(dsmr_instance).set_decryption_key(private_key);

ota:
  password: !secret ota_password

dashboard_import:
  package_import_url: github://zuidwijk/dsmr/slimmelezer.yaml

web_server:
  port: 80

uart:
  baud_rate: 115200
  rx_pin: D7
  rx_buffer_size: 1500

globals:
  - id: has_key
    type: bool
    restore_value: yes
    initial_value: "false"
  - id: stored_decryption_key
    type: char[32]
    restore_value: yes

dsmr:
  id: dsmr_instance
  # For Luxembourg users set here your decryption key
  #decryption_key: !secret decryption_key // enable this when using decryption for Luxembourg; key like '00112233445566778899AABBCCDDEEFF'

sensor:
  - platform: dsmr
    energy_delivered_lux:
      name: "Energy Consumed Luxembourg"
    energy_delivered_tariff1:
      name: "Energy Consumed Tariff 1"
    energy_delivered_tariff2:
      name: "Energy Consumed Tariff 2"
    energy_returned_lux:
      name: "Energy Produced Luxembourg"
    energy_returned_tariff1:
      name: "Energy Produced Tariff 1"
    energy_returned_tariff2:
      name: "Energy Produced Tariff 2"
    power_delivered:
      name: "Power Consumed"
      accuracy_decimals: 3
    power_returned:
      name: "Power Produced"
      accuracy_decimals: 3
    electricity_failures:
      name: "Electricity Failures"
      icon: mdi:alert
    electricity_long_failures:
      name: "Long Electricity Failures"
      icon: mdi:alert
    voltage_l1:
      name: "Voltage Phase 1"
    voltage_l2:
      name: "Voltage Phase 2"
    voltage_l3:
      name: "Voltage Phase 3"
    current_l1:
      name: "Current Phase 1"
    current_l2:
      name: "Current Phase 2"
    current_l3:
      name: "Current Phase 3"
    power_delivered_l1:
      name: "Power Consumed Phase 1"
      accuracy_decimals: 3
    power_delivered_l2:
      name: "Power Consumed Phase 2"
      accuracy_decimals: 3
    power_delivered_l3:
      name: "Power Consumed Phase 3"
      accuracy_decimals: 3
    power_returned_l1:
      name: "Power Produced Phase 1"
      accuracy_decimals: 3
    power_returned_l2:
      name: "Power Produced Phase 2"
      accuracy_decimals: 3
    power_returned_l3:
      name: "Power Produced Phase 3"
      accuracy_decimals: 3
    gas_delivered:
      name: "Gas Consumed"
    gas_delivered_be:
      name: "Gas Consumed Belgium"
  - platform: uptime
    name: "SlimmeLezer Uptime"
  - platform: wifi_signal
    name: "SlimmeLezer Wi-Fi Signal"
    update_interval: 60s

text_sensor:
  - platform: dsmr
    identification:
      name: "DSMR Identification"
    p1_version:
      name: "DSMR Version"
    p1_version_be:
      name: "DSMR Version Belgium"
  - platform: wifi_info
    ip_address:
      name: "SlimmeLezer IP Address"
    ssid:
      name: "SlimmeLezer Wi-Fi SSID"
    bssid:
      name: "SlimmeLezer Wi-Fi BSSID"
  - platform: version
    name: "ESPHome Version"
    hide_timestamp: true
```

The error is in the second line of:

```
sensor:
  - platform: dsmr
```

and is created by removing/commenting out the external_component section.

mmakaay, 2022-03-17:
I tried your exact configuration and it compiled on my 2022.3.0.
Can you try a clean build? (three dots menu -> Clean Build Files, then build again).

Calimerorulez, 2022-03-18:
@mmakaay 

Thank you, that worked :)

Only thing now is, I've got more

```
[10:09:41][E][dsmr:265]: !B2D3
 ^
Checksum mismatch
```
but I don't think it is related..

sjakief, 2022-03-18:
I think that has to do with:
```
# Enable logging
logger:
  baud_rate: 0
```

mmakaay, 2022-03-18:
Yes, @sjakief is correct. The logger baudrate must be set to 0, to not have serial logging interfer with P1 port reading.

Calimerorulez, 2022-03-18:
That seems to have done the trick, thank you all!

mmakaay, 2022-03-18:
Wonderful! Then I will close this issue, since everything is peachy for all 😎

closed 2022-03-18 by mmakaay

pengemetal, 2022-04-04:
Hi, I have the same problem (platform not found sensor.dsmr) with 2022.3.2 ESPhome. Build clean files couldn't solve the problem. What can I do?

jsuanet, 2022-04-04:
Are you still using an external component in your yaml-file, something like this:
```yaml
external_components:
  - source:
      type: git
      url: https://github.com/mmakaay/esphome
      ref: fix-dsmr-read-chunk-size
      components: [ "dsmr" ]
      refresh: 60s
```
If yes, then first remove this from your yaml-file and after that clean your build files again and start the compile.

pengemetal, 2022-04-04:
I have deleted external component. After Clean and Downloding bin file this the result: 
[image]

jsuanet, 2022-04-04:
Please post your current yaml file.

pengemetal, 2022-04-04:
substitutions:
  device_name: slimmelezer
  device_description: "DIY P1 module to read your smart meter"

esphome:
  name: ${device_name}
  comment: '${device_description}'
  platform: ESP8266
  esp8266_restore_from_flash: true
  board: d1_mini
  on_boot:
    then:
      - if:
          condition:
            lambda: return id(has_key);
          then:
            - lambda: |-
                std::string key(id(stored_decryption_key), 32);
                id(dsmr_instance).set_decryption_key(key);
          else:
            - logger.log:
                level: info
                format: "Not using decryption key. If you need to set a key use Home Assistant service 'ESPHome:  ${device_name}_set_dsmr_key'"

wifi:
  networks:
    - ssid: "xxxxxxx"
      password: "xxxxxxx"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "SlimmeLezer Fallback Hotspot"
    password: "slimmelezer"

captive_portal:

# Enable logging
logger:
  baud_rate: 0
  level: VERBOSE

# Enable Home Assistant API
api:
  services:
    service: set_dsmr_key
    variables:
      private_key: string
    then:
      - logger.log:
          format: Setting private key %s. Set to empty string to disable
          args: [private_key.c_str()]
      - globals.set:
          id: has_key
          value: !lambda 'return private_key.length() == 32;'
      - lambda: |-
          if (private_key.length() == 32)
            private_key.copy(id(stored_decryption_key), 32);
          id(dsmr_instance).set_decryption_key(private_key);
ota:

web_server:
  port: 80

uart:
  baud_rate: 115200
  rx_pin: D7
  rx_buffer_size: 1500

globals:
  - id: has_key
    type: bool
    restore_value: yes
    initial_value: 'false'
  - id: stored_decryption_key
    type: char[32]
    restore_value: yes

dsmr:
  id: dsmr_instance
#  decryption_key: !secret decryption_key // enable this when using decryption for Luxembourg; key like '00112233445566778899AABBCCDDEEFF'

sensor:
  - platform: dsmr
    energy_delivered_lux:
      name: dsmr_energy_delivered_lux
    energy_delivered_tariff1:
      name: dsmr_energy_delivered_tariff1
    energy_delivered_tariff2:
      name: dsmr_energy_delivered_tariff2
    energy_returned_lux:
      name: dsmr_energy_returned_lux
    energy_returned_tariff1:
      name: dsmr_energy_returned_tariff1
    energy_returned_tariff2:
      name: dsmr_energy_returned_tariff2
    power_delivered:
      name: power_delivered
      accuracy_decimals: 0
      filters:
        - multiply: 1000
    power_returned:
      name: power_returned
      accuracy_decimals: 0
      filters:
        - multiply: 1000
    electricity_failures:
      name: electricity_failures
      icon: mdi:alert
    electricity_long_failures:
      name: electricity_long_failures
      icon: mdi:alert
    voltage_l1:
      name: voltage_l1
    voltage_l2:
      name: voltage_l2
    voltage_l3:
      name: voltage_l3
    current_l1:
      name: current_l1
    current_l2:
      name: current_l2
    current_l3:
      name: current_l3
    power_delivered_l1:
      name: power_delivered_l1
      accuracy_decimals: 0
      filters:
        - multiply: 1000
    power_delivered_l2:
      name: power_delivered_l2
      accuracy_decimals: 0
      filters:
        - multiply: 1000
    power_delivered_l3:
      name: power_delivered_l3
      accuracy_decimals: 0
      filters:
        - multiply: 1000
    power_returned_l1:
      name: power_returned_l1
      accuracy_decimals: 0
      filters:
        - multiply: 1000
    power_returned_l2:
      name: power_returned_l2
      accuracy_decimals: 0
      filters:
        - multiply: 1000
    power_returned_l3:
      name: power_returned_l3
      accuracy_decimals: 0
      filters:
        - multiply: 1000
#    gas_delivered:
#      name: gas_delivered
#    gas_delivered_be:
#      name: gas_delivered_be
  - platform: uptime
    name: SlimmeLezer Uptime
  - platform: wifi_signal
    name: SlimmeLezer WiFi Signal
    update_interval: 60s

text_sensor:
  - platform: dsmr
    identification:
      name: 'dsmr_identification'
    p1_version:
      name: 'dsmr_p1_version'
    p1_version_be:
      name: 'dsmr_p1_version_be'
  - platform: wifi_info
    ip_address:
      name: SlimmeLezer IP
    ssid:
      name: SlimmeLezer SSID
    bssid:
      name: SlimmeLezer BSSID

jsuanet, 2022-04-04:
Well, your posted yaml missed the first line `substitutions:`  After adding that line, it compiles for me without any error. So it must be something in your environment.

Are you really sure you cleaned the build files (three dots menu -> Clean Build Files) before building it again?

pengemetal, 2022-04-04:
subtitutions is there only missing in copy, sorry!
Yes, I click three dots and after Clean Build Files, but it has a lot of red errors (I paste it before)

jsuanet, 2022-04-04:
Can you try a clean build files again and describe what happens?

pengemetal, 2022-04-04:
[image]
[image]

jsuanet, 2022-04-04:
I only get this error if the `external component:` part is still in the yaml file. Did you save the yaml file after removing that part, or is it still active in another browser window?

pengemetal, 2022-04-04:
Yes, I saved and closed all open HA window. I found a folder esphome/custom component/dsmr, I renamed and after it works! Thank you all of your help!

mmakaay, 2022-04-04:
That would break the build too yes. The `/custom_component` folder is the "old school" way of adding component source to ESPHome. Good that you got it to work 👍