esphome/issues #3143: 2022.3 breaks pulse_meter

stevetz, 2022-03-16:
### The problem

I have a `pulse_meter` measuring my electricity meter. It was working perfectly before updating.

I note that there is a pull in this version https://github.com/esphome/esphome/pull/3082 but this suggests it is not a breaking change, and I have tested both `internal_filter_mode: "EDGE"` and `internal_filter_mode: "PULSE" - pulse doesn't read at all, and edge (which, if default, is exactly the same as my previously working config) still misses some pulses.

### Which version of ESPHome has the issue?

2022.3.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.3.5

### What platform are you using?

ESP8266

### Board

nodemcu

### Component causing the issue

pulse_meter

### Example YAML snippet

```yaml
########## Electricity Meter ##########
sensor:
  - platform: pulse_meter
    name: '${friendly_name} - Electricity Meter'
    icon: mdi:flash-outline
    unit_of_measurement: 'W'
    state_class: measurement
    device_class: power
    pin:
      number: GPIO5
      mode: INPUT_PULLUP
    internal_filter: 75ms
    internal_filter_mode: "EDGE" # with/without this line, and tried "PULSE"
    accuracy_decimals: 1
    filters:
      - multiply: 18.75
    total:
      name: '${friendly_name} - Total Electricity'
      unit_of_measurement: 'kWh'
      icon: mdi:circle-slice-3
      state_class: total_increasing
      device_class: energy
      accuracy_decimals: 3
      filters:
        - multiply: 0.0003125
```

### Anything in the logs that might be useful for us?

```txt
[16:45:59][D][sensor:124]: 'Energy Monitor - Electricity Meter': Sending state 378.15128 W with 1 decimals of accuracy
[16:45:59][D][sensor:124]: 'Energy Monitor - Total Electricity': Sending state 0.00812 kWh with 3 decimals of accuracy
[16:46:02][D][sensor:124]: 'Energy Monitor - Electricity Meter': Sending state 390.08319 W with 1 decimals of accuracy
[16:46:02][D][sensor:124]: 'Energy Monitor - Total Electricity': Sending state 0.00844 kWh with 3 decimals of accuracy
[16:46:05][D][pulse_meter:061]: No pulse detected for 4294s, assuming 0 pulses/min
[16:46:05][D][sensor:124]: 'Energy Monitor - Electricity Meter': Sending state 0.00000 W with 1 decimals of accuracy
[16:46:05][D][sensor:124]: 'Energy Monitor - Total Electricity': Sending state 0.00875 kWh with 3 decimals of accuracy
[16:46:08][D][sensor:124]: 'Energy Monitor - Electricity Meter': Sending state 401.92926 W with 1 decimals of accuracy
[16:46:08][D][sensor:124]: 'Energy Monitor - Total Electricity': Sending state 0.00906 kWh with 3 decimals of accuracy
[16:46:11][D][sensor:124]: 'Energy Monitor - Electricity Meter': Sending state 391.71310 W with 1 decimals of accuracy
[16:46:11][D][sensor:124]: 'Energy Monitor - Total Electricity': Sending state 0.00937 kWh with 3 decimals of accuracy
[16:46:14][D][pulse_meter:061]: No pulse detected for 4294s, assuming 0 pulses/min
[16:46:14][D][sensor:124]: 'Energy Monitor - Electricity Meter': Sending state 0.00000 W with 1 decimals of accuracy
[16:46:14][D][sensor:124]: 'Energy Monitor - Total Electricity': Sending state 0.00969 kWh with 3 decimals of accuracy
[16:46:16][D][sensor:124]: 'Energy Monitor - Electricity Meter': Sending state 402.93695 W with 1 decimals of accuracy
[16:46:17][D][sensor:124]: 'Energy Monitor - Total Electricity': Sending state 0.01000 kWh with 3 decimals of accuracy
```

stevetz, 2022-03-16:
I've restored the add-on and this node to 2022.2.6 - the `pulse_meter` is working as expected, without any of the:
```
[pulse_meter:061]: No pulse detected for 4294s, assuming 0 pulses/min
```
messages at all, again.

lukebaldan, 2022-03-17:
> I've restored the add-on and this node to 2022.2.6 - the `pulse_meter` is working as expected, without any of the:
> 
> ```
> [pulse_meter:061]: No pulse detected for 4294s, assuming 0 pulses/min
> ```
> 
> messages at all, again.

I have encountered the same issue. My power meter is reporting spurious values. Downgrade to 2022.2.6 has resolved the issue.

andreytse, 2022-03-18:
I have the same problem with pulse meter on 2022.3.0

flystyle, 2022-03-18:
Same problem.. 2022.3.0 is the worst release ever, let's name it 2022.3.0-Putin 🤬

TheOfficialMrBlah, 2022-03-19:
I also have the same problem.
The log looks quite similar for me:
`[pulse_meter:061]: No pulse detected for 4294s, assuming 0 pulses/min`

The diagram created from the values does not look good either.
[image]

Version 2022.2.6 works fine.

cstaahl, 2022-03-20:
Hi!
I have struggled to consistently reproduce the issue on my end. I have a suspicion it is caused by the filtering logic being moved out of the ISR that in turn causes overflow in some of the arithmetics.

Please try out the code in [this branch](https://github.com/cstaahl/esphome/tree/fix_pulse_meter_isr) and let me if it resolves the issue.

TheOfficialMrBlah, 2022-03-21:
> Hi! I have struggled to consistently reproduce the issue on my end. I have a suspicion it is caused by the filtering logic being moved out of the ISR that in turn causes overflow in some of the [...]

Hi!

I have good news, I think.
I'm not sure if I installed this version correctly, because I don't know how to choose a branch in "pip install" but it seems to work.
I haven't looked through the log yet but I don't see any more zero values in the graph.

[image]

Sirs0ri, 2022-08-16:
I've been dealing with the same error in a slightly different context (#3453), and had a closer look at the the [line that causes this message](https://github.com/esphome/esphome/blob/dev/esphome/components/pulse_meter/pulse_meter_sensor.cpp#L61):

```ESP_LOGD(TAG, "No pulse detected for %us, assuming 0 pulses/min", time_since_valid_edge_us / 1000000);```

Reversing the formatting, the original value for time_since_valid_edge_us (`4 294 * 1 000 000`) is suspiciously close to `4 294 967 295`, the maximum value that it can hold as a `uint32_t`.

It looks to me like this problem comes from an overflow error due to the usage of `micros()` to track when pulses happen, which (according to the [Arduino Docs](https://www.arduino.cc/reference/en/language/functions/time/micros/)) stores the microseconds _since the board began running the current program and intentionally overflows every ~70 minutes_.

The interval since the last pulse is calculated as the difference between the `micros()` based timestamp of the last pulse and the current `micros()` based timestamp. If `micros()` overflows between two pulses, that difference will be close to UINT32_MAX, thus causing the dreaded `No pulse detected for 4294s` warning and dropping the (valid) measurement.

I suspect that this could be fixed quite easily by detecting the overflow of `micros()` (in that case the current timestamp is lower than the last timestamp) and adjusting the calculation of the time difference accordingly (the actual interval should be UINT32_MAX minus the calculated interval).

stevetz, 2022-08-30:
I accidentally upgraded my energy monitor to 2022.8.0 today, and have lost the snapshot of esphome 2022.2.6 (last working version) so I'm struggling with this again.

I can confirm that even 2022.8.0 regularly drops the pulses:
```
[pulse_meter:061]: No pulse detected for 4294s, assuming 0 pulses/min
```

@cstaahl, @stevebaxter any chance you can take a look at this or just revert back to the last working version?

stevetz, 2022-08-30:
> Hi! I have struggled to consistently reproduce the issue on my end. I have a suspicion it is caused by the filtering logic being moved out of the ISR that in turn causes overflow in some of the [...]

I'm afraid I don't know how to try a different branch - I use the HA addon... and google wasn't much help, sorry if its a banal question... Can you help?

TheOfficialMrBlah, 2022-08-31:
> I'm afraid I don't know how to try a different branch - I use the HA addon... and google wasn't much help, sorry if its a banal question... Can you help?

There is an open pr since March. esphome/esphome#3321
With this one it works for me again.

To use it, just add the following lines to your yaml:
```
external_components:
  - source: github://pr#3321
    components: [pulse_meter]
```

stevetz, 2022-09-02:
Thanks so much @TheOfficialMrBlah ! I've had it running for 12 hours now, and have 0 missing pulses, and the actual energy readings seem consistent with the last working version.

WiktorS, 2022-09-04:
Just my 2 cents,

I'm also affected by this bug, tried `pr#3321` but it didn't worked as expected for me.
I read about `external_components` as I didn't know it before and looks like its fairly easy to downgrade any component to any version.
So I tried this:
```
external_components:
  - source: github://esphome/esphome@2022.2.6
    components: [pulse_meter]
```
Looks like this is the workaround that reverts to good old code before breakage 😃

jesserockz, 2022-09-06:
@WiktorS YMMV with this as internal APIs change and old code would not have updated APIs

closed 2022-10-12 by jesserockz

emeyedeejay, 2022-11-01:
An existing pulse_meter setup has been running well for months but recently started behaving very weirdly. The readins will swing from e.g. 1000W to like 36 000 000W and back again.

I cannot recall making any changes to the setup but is it possible a recent update broke something?

My config:

```
sensor:    # Pulse meter
  - platform: pulse_meter
    name: 'Current Power Consumption'
    id: sensor_current_power_consumption
    unit_of_measurement: 'W'
    state_class: measurement
    device_class: power
    icon: mdi:flash-outline
    accuracy_decimals: 0
    pin: D2
    internal_filter_mode: pulse
    filters:
  # multiply value = (60 / imp value) * 1000
  # - multiply: 60
    - lambda: return x * ((60.0 / 1000) * 1000.0);
```

[image]

Looks like it went nuts on 24 October - doe that match any release date maybe?

[image]

Some log output:

```
INFO Reading configuration /config/esphome/electricity-usage-monitor.yaml...
INFO Detected timezone 'Africa/Johannesburg'
INFO Starting log output from electricity-usage-monitor.local using esphome API
INFO Successfully connected to electricity-usage-monitor.local
[22:42:52][I][app:102]: ESPHome version 2022.10.1 compiled on Oct 26 2022, 12:16:53
[22:42:52][C][wifi:502]: WiFi:
[22:42:52][C][wifi:360]:   Local MAC: B4:E6:2D:13:F0:A9
[22:42:52][C][wifi:361]:   SSID: 'JNET'[redacted]
[22:42:52][C][wifi:362]:   IP Address: 192.168.0.120
[22:42:52][C][wifi:363]:   BSSID: 78:8A:20:F7:3B:46[redacted]
[22:42:52][C][wifi:365]:   Hostname: 'electricity-usage-monitor'
[22:42:52][C][wifi:367]:   Signal strength: -71 dB ▂▄▆█
[22:42:52][C][wifi:371]:   Channel: 6
[22:42:52][C][wifi:372]:   Subnet: 255.255.255.0
[22:42:52][C][wifi:373]:   Gateway: 192.168.0.1
[22:42:52][C][wifi:374]:   DNS1: 192.168.0.1
[22:42:52][C][wifi:375]:   DNS2: 0.0.0.0
[22:42:52][C][logger:275]: Logger:
[22:42:52][C][logger:276]:   Level: DEBUG
[22:42:52][C][logger:277]:   Log Baud Rate: 115200
[22:42:52][C][logger:278]:   Hardware UART: UART0
[22:42:52][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'Home power monitor'
[22:42:52][C][gpio.binary_sensor:016]:   Pin: GPIO4
[22:42:52][C][pulse_meter:069]: Pulse Meter 'Current Power Consumption'
[22:42:52][C][pulse_meter:069]:   Device Class: 'power'
[22:42:52][C][pulse_meter:069]:   State Class: 'measurement'
[22:42:52][C][pulse_meter:069]:   Unit of Measurement: 'W'
[22:42:52][C][pulse_meter:069]:   Accuracy Decimals: 0
[22:42:52][C][pulse_meter:069]:   Icon: 'mdi:flash-outline'
[22:42:52][C][pulse_meter:070]:   Pin: GPIO4
[22:42:52][C][pulse_meter:074]:   Filtering pulses shorter than 13 µs
[22:42:52][C][pulse_meter:076]:   Assuming 0 pulses/min after not receiving a pulse for 300s
[22:42:52][C][total_daily_energy:023]: Total Daily Energy 'Home daily power'
[22:42:52][C][total_daily_energy:023]:   Device Class: 'energy'
[22:42:52][C][total_daily_energy:023]:   State Class: 'total_increasing'
[22:42:52][C][total_daily_energy:023]:   Unit of Measurement: 'kWh'
[22:42:52][C][total_daily_energy:023]:   Accuracy Decimals: 3
[... 31 lines cut ...]
[22:43:07][D][binary_sensor:036]: 'Home power monitor': Sending state ON
[22:43:09][D][sensor:126]: 'Current Power Consumption': Sending state 1449.85913 W with 0 decimals of accuracy
[22:43:09][D][sensor:126]: 'Home daily power': Sending state 1732.92554 kWh with 3 decimals of accuracy
[22:43:09][D][sensor:126]: 'Home total power': Sending state 128.33800 kWh with 3 decimals of accuracy
[22:43:09][D][binary_sensor:036]: 'Home power monitor': Sending state OFF
[22:43:12][D][sensor:126]: 'Current Power Consumption': Sending state 0.00000 W with 0 decimals of accuracy
[22:43:12][D][sensor:126]: 'Home daily power': Sending state 1732.92554 kWh with 3 decimals of accuracy
[22:43:12][D][sensor:126]: 'Home total power': Sending state 128.34399 kWh with 3 decimals of accuracy
[22:43:12][D][binary_sensor:036]: 'Home power monitor': Sending state ON
[22:43:14][D][sensor:126]: 'Home total power': Sending state 128.34599 kWh with 3 decimals of accuracy
[22:43:14][D][binary_sensor:036]: 'Home power monitor': Sending state OFF
[22:43:17][D][sensor:126]: 'Current Power Consumption': Sending state 1401.86914 W with 0 decimals of accuracy
[22:43:17][D][sensor:126]: 'Home daily power': Sending state 1732.92761 kWh with 3 decimals of accuracy
[22:43:17][D][sensor:126]: 'Home total power': Sending state 128.34801 kWh with 3 decimals of accuracy
[22:43:17][D][sensor:126]: 'Current Power Consumption': Sending state 240000.00000 W with 0 decimals of accuracy
[22:43:17][D][sensor:126]: 'Home daily power': Sending state 1732.92993 kWh with 3 decimals of accuracy
[22:43:17][D][sensor:126]: 'Home total power': Sending state 128.34900 kWh with 3 decimals of accuracy
[22:43:17][D][binary_sensor:036]: 'Home power monitor': Sending state ON
[22:43:19][D][binary_sensor:036]: 'Home power monitor': Sending state OFF
[22:43:22][D][sensor:126]: 'Current Power Consumption': Sending state 257142.85938 W with 0 decimals of accuracy
[22:43:22][D][sensor:126]: 'Home daily power': Sending state 1733.29260 kWh with 3 decimals of accuracy
[22:43:22][D][sensor:126]: 'Home total power': Sending state 128.35500 kWh with 3 decimals of accuracy
[22:43:22][D][sensor:126]: 'Current Power Consumption': Sending state 3600000.00000 W with 0 decimals of accuracy

```