esphome/issues #3144: Web_server is missing handle_js_request

nilsmau, 2022-03-16:
### The problem

```
src/esphome/components/web_server/web_server.cpp:402:6: error: no declaration matches 'void esphome::web_server::WebServer::handle_js_request(AsyncWebServerRequest*)'
  402 | void WebServer::handle_js_request(AsyncWebServerRequest *request) {
      |      ^~~~~~~~~
src/esphome/components/web_server/web_server.cpp:402:6: note: no functions named 'void esphome::web_server::WebServer::handle_js_request(AsyncWebServerRequest*)'
In file included from src/esphome/components/web_server/web_server.cpp:3:
src/esphome/components/web_server/web_server.h:33:7: note: 'class esphome::web_server::WebServer' defined here
   33 | class WebServer : public Controller, public Component, public AsyncWebHandler {
      |       ^~~~~~~~~
src/esphome/components/web_server/web_server.cpp: In member function 'virtual void esphome::web_server::WebServer::handleRequest(AsyncWebServerRequest*)':
src/esphome/components/web_server/web_server.cpp:1144:11: error: 'class esphome::web_server::WebServer' has no member named 'handle_js_request'; did you mean 'handle_index_request'?
 1144 |     this->handle_js_request(request);
      |           ^~~~~~~~~~~~~~~~~
      |           handle_index_request
*** [/data/wordclock/.pioenvs/wordclock/src/esphome/components/web_server/web_server.cpp.o] Error 1
```

### Which version of ESPHome has the issue?

2022.3.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.3.5

### What platform are you using?

ESP8266

### Board

nodemcuv2

### Component causing the issue

web_server

### Example YAML snippet

```yaml
web_server:
  port: 80
  version: 1
  #auth:
  #  username: *********
  #  password: *********
  #css_url: ""
  ##css_include: "/config/esphome/webserver-v1.min.css"
  #js_url: ""
  js_include: "/config/esphome/webserver.js"
```

### Anything in the logs that might be useful for us?

```txt
See above
```

### Additional information

Issue does not occur with previous ESPHome version.

zibous, 2022-03-18:
Same problem here:

Config:
```yaml
## ---------------------------------------------------
## WEBSERVER Settings
## ---------------------------------------------------
web_server:
  port: 80
  css_include: "_static/webserver-v1.min.css"
  css_url: ""
  js_include: "_static/webserver-v1.min.js"
  js_url: ""
```

```
Processing esp32app (board: esp32dev; framework: arduino; platform: platformio/espressif32 @ 3.5.0)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Library Manager: Installing esphome/AsyncTCP-esphome @ 1.2.2
Library Manager: AsyncTCP-esphome @ 1.2.2 has been installed!
Library Manager: Installing esphome/ESPAsyncWebServer-esphome @ 2.1.0
Library Manager: ESPAsyncWebServer-esphome @ 2.1.0 has been installed!
Library Manager: Installing dependencies...
Library Manager: Installing ottowinter/ESPAsyncTCP-esphome
Library Manager: ESPAsyncTCP-esphome @ 1.2.3 has been installed!
Library Manager: AsyncTCP-esphome @ 1.2.2 is already installed
Library Manager: Installing Hash
Library Manager: Installing ESP8266WiFi
Library Manager: Installing bblanchon/ArduinoJson @ 6.18.5
Library Manager: ArduinoJson @ 6.18.5 has been installed!
```

**Error:**
```
Compiling .pioenvs/wasserundgas/src/esphome/core/application.cpp.o
src/esphome/components/web_server/web_server.cpp:391:66: error: no 'void esphome::web_server::WebServer::handle_css_request(AsyncWebServerRequest*)' member function declared in class 'esphome::web_server::WebServer'
 void WebServer::handle_css_request(AsyncWebServerRequest *request) {
                                                                  ^
src/esphome/components/web_server/web_server.cpp:402:65: error: no 'void esphome::web_server::WebServer::handle_js_request(AsyncWebServerRequest*)' member function declared in class 'esphome::web_server::WebServer'
 void WebServer::handle_js_request(AsyncWebServerRequest *request) {
                                                                 ^
src/esphome/components/web_server/web_server.cpp: In member function 'virtual void esphome::web_server::WebServer::handleRequest(AsyncWebServerRequest*)':
src/esphome/components/web_server/web_server.cpp:1137:11: error: 'class esphome::web_server::WebServer' has no member named 'handle_css_request'
     this->handle_css_request(request);
           ^
src/esphome/components/web_server/web_server.cpp:1144:11: error: 'class esphome::web_server::WebServer' has no member named 'handle_js_request'
     this->handle_js_request(request);
           ^
Compiling .pioenvs/wasserundgas/src/esphome/core/color.cpp.o
*** [.pioenvs/wasserundgas/src/esphome/components/web_server/web_server.cpp.o] Error 1
```

guevara777, 2022-03-20:
same problem here.

nilsmau, 2022-04-28:
@OttoWinter, @jesserockz can you please take a look at it?

nilsmau, 2022-05-28:
Update: For whatever reason, the **js_include** now works (as of ESPHome `v2022.5.1`) and is not throwing the error in the subject of this post. 
@zibous, @guevara777, @sheminasalam, @buksve is it working at your end as well? If so, I would close the issue.

sheminasalam, 2022-05-29:
```
Example web_server configuration using version 2 - no internet/intranet required:

# Example configuration entry
web_server:
  local: true
```

This works for me.

guevara777, 2022-05-30:
Tested and working again with:

```
web_server:
  local: true
```

closed 2022-06-01 by nilsmau