esphome/issues #3146: BH1750 no longer sends values

KrX3D, 2022-03-16:
### The problem

Hello, since i updated today ESPHome and i saw this:

`BH1750 dynamically calculate options [esphome#3214](https://github.com/esphome/esphome/pull/3214) by [@OttoWinter](https://github.com/OttoWinter) (breaking-change)`

i needed to remove the measurement_duration parameter since i got an error. i did it but now i no longer get any values from my sensor.

the script was working for months without a problem so far

### Which version of ESPHome has the issue?

2022.3.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.3.5

### What platform are you using?

ESP32

### Board

nodemcuv2

### Component causing the issue

/

### Example YAML snippet

```yaml
esp8266:
  board: nodemcuv2
  restore_from_flash: true
  framework:
    version: 2.7.4

preferences:
  flash_write_interval: 1min

i2c:
  sda: 4
  scl: 5
  scan: true
  id: bus_a
  
sensor:
  - platform: bh1750
    name: "${device_name} Helligkeit"
    address: 0x23
    unit_of_measurement: lux
    update_interval: 500ms
    accuracy_decimals: 0
    filters:
      - sliding_window_moving_average:
          window_size: 6
          send_every: 1
```

### Anything in the logs that might be useful for us?

```txt
/
```

### Additional information

/

jsuanet, 2022-03-17:
You are using a very short `update_interval` for this sensor.  When I tried my sensor with this 500ms I also had no data, however with 700ms it worked again. So I think there might be a timing issue now the options are calculated dynamically. What happens if you use a larger `update_interval`?

jsuanet, 2022-03-17:
> but its still odd since it was working for months with 500ms, just since the last esphome update it stoped.

Well, the C++ code for this component is changed. From the description in the PR:  _"Until now for BH1750 you had to set static resolution/measurement_duration values - and so either optimize for low-light or bright scenes. Instead, dynamically calculate those options now by first doing a quick l-resolution measurement, then switch to the optimal mode/mtreg values."_  Maybe this new process is interrupted by the short update_interval. 

`accuracy_decimals: 0` is used for the data which is send to HA, not for what the sensor delivers to the ESP-node...

tomlut, 2022-05-14:
I'm using a 3 second update interval and it still fails. Esphome can find it, just not talk to it:
```
[00:01:39][I][i2c.arduino:054]: Results from i2c bus scan:
[00:01:39][I][i2c.arduino:060]: Found i2c device at address 0x23
[00:01:39][C][status:034]: Status Binary Sensor 'Lux Sensor Status'
[00:01:39][C][status:034]:   Device Class: 'connectivity'
[00:01:39][C][bh1750.sensor:118]: BH1750 'Outside Light Level'
[00:01:39][C][bh1750.sensor:118]:   Device Class: 'illuminance'
[00:01:39][C][bh1750.sensor:118]:   State Class: 'measurement'
[00:01:39][C][bh1750.sensor:118]:   Unit of Measurement: 'lx'
[00:01:39][C][bh1750.sensor:118]:   Accuracy Decimals: 1
[00:01:39][C][bh1750.sensor:119]:   Address: 0x23
[00:01:39][E][bh1750.sensor:121]: Communication with BH1750 failed!
```

I've tried dropping the frequency but that did not help:
```
i2c:
  sda: 21
  scl: 22
  scan: True
  frequency: 10kHz

sensor:
  - platform: bh1750
    name: "Outside Light Level"
    address: 0x23
    update_interval: 3s
```

closed 2022-10-28 by github-actions[bot]