esphome/issues #3147: 2022.3.0 breaks "PVVX MiThermometer"

PWnet, 2022-03-17:
### The problem

After upgrading to 2022.3.0 firmware with "PVVX MiThermometer" sensors cannot be compiled.

### Which version of ESPHome has the issue?

2022.3.0

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2022.3.5

### What platform are you using?

ESP32

### Board

esp32cam

### Component causing the issue

PVVX MiThermometer

### Example YAML snippet

```yaml
substitutions: 
  devicename: esphome-ble-01

esphome:
  name: ${devicename}
  platform: ESP32
  board:   esp32cam
    
wifi:
  networks:
    - ssid: "xxxxxx"
      password: !secret wifi_password
    - ssid: "yyyyyy"
      password: !secret wifi_password
  domain: .abcd.local

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esphome-Ble-01"
    password: "abcdefghi"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: !secret api_password

ota:
  password: !secret ota_password
  
status_led:
  pin:
    number: GPIO33
    inverted: true

  
esp32_ble_tracker:

switch:
  - platform: restart
    name: ${devicename} restart
  - platform: safe_mode
    name: "${devicename} Restart (Safe Mode)"

sensor:
  - platform: wifi_signal
    name: ${devicename} wifi signal
    update_interval: 60s
 
  - platform: uptime
    name: ${devicename} uptime
         
      
  - platform: pvvx_mithermometer
    mac_address: "A4:C1:38:25:A0:05"
    temperature:
      name: "LYWSD03MMC Temperature"
    humidity:
      name: "LYWSD03MMC Humidity"
      accuracy_decimals: 1
    battery_level:
      name: "LYWSD03MMC Battery Level"
    battery_voltage:
      name: "LYWSD03MMC Battery-Voltage"      
      
  - platform: pvvx_mithermometer
    mac_address: "A4:C1:38:E4:56:FB"
    temperature:
      name: "ATC Temperature"
    humidity:
      name: "ATC Humidity"
      accuracy_decimals: 1
    battery_level:
      name: "ATC Battery-Level"
    battery_voltage:
      name: "ATC Battery-Voltage"
      
  - platform: pvvx_mithermometer
    mac_address: "A4:C1:38:73:9C:5A"
    temperature:
      name: "ATC_739C5A Temperature"
    humidity:
      name: "ATC_739C5A Humidity"
      accuracy_decimals: 1
    battery_level:
      name: "ATC_739C5A Battery-Level"
    battery_voltage:
      name: "ATC_739C5A Battery-Voltage"
```

### Anything in the logs that might be useful for us?

```txt
ERROR Unable to load component pvvx_mithermometer.sensor:
Traceback (most recent call last):
  File "/esphome/esphome/loader.py", line 162, in _lookup_module
    module = importlib.import_module(f"esphome.components.{domain}")
  File "/usr/lib/python3.9/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 790, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/config/custom_components/pvvx_mithermometer/sensor.py", line 34, in <module>
    cv.Optional(CONF_TEMPERATURE): sensor.sensor_schema(
TypeError: sensor_schema() takes from 0 to 1 positional arguments but 4 were given
ERROR Unable to load component pvvx_mithermometer.sensor:
Traceback (most recent call last):
  File "/esphome/esphome/loader.py", line 162, in _lookup_module
    module = importlib.import_module(f"esphome.components.{domain}")
  File "/usr/lib/python3.9/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 790, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/config/custom_components/pvvx_mithermometer/sensor.py", line 34, in <module>
    cv.Optional(CONF_TEMPERATURE): sensor.sensor_schema(
TypeError: sensor_schema() takes from 0 to 1 positional arguments but 4 were given
ERROR Unable to load component pvvx_mithermometer.sensor:
Traceback (most recent call last):
  File "/esphome/esphome/loader.py", line 162, in _lookup_module
    module = importlib.import_module(f"esphome.components.{domain}")
  File "/usr/lib/python3.9/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 790, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/config/custom_components/pvvx_mithermometer/sensor.py", line 34, in <module>
    cv.Optional(CONF_TEMPERATURE): sensor.sensor_schema(
TypeError: sensor_schema() takes from 0 to 1 positional arguments but 4 were given
Failed config

sensor.pvvx_mithermometer: [source /config/esphome-ble-01.yaml:72]
  
  Platform not found: 'sensor.pvvx_mithermometer'.
  platform: pvvx_mithermometer
  mac_address: A4:C1:38:25:A0:05
  temperature: 
    name: LYWSD03MMC Temperature
  humidity: 
    name: LYWSD03MMC Humidity
    accuracy_decimals: 1
  battery_level: 
    name: LYWSD03MMC Battery Level
  battery_voltage: 
    name: LYWSD03MMC Battery-Voltage
sensor.pvvx_mithermometer: [source /config/esphome-ble-01.yaml:84]
  
  Platform not found: 'sensor.pvvx_mithermometer'.
  platform: pvvx_mithermometer
  mac_address: A4:C1:38:E4:56:FB
  temperature: 
    name: ATC Temperature
  humidity: 
    name: ATC Humidity
    accuracy_decimals: 1
  battery_level: 
    name: ATC Battery-Level
  battery_voltage: 
    name: ATC Battery-Voltage
sensor.pvvx_mithermometer: [source /config/esphome-ble-01.yaml:96]
  
  Platform not found: 'sensor.pvvx_mithermometer'.
  platform: pvvx_mithermometer
  mac_address: A4:C1:38:73:9C:5A
  temperature: 
    name: ATC_739C5A Temperature
  humidity: 
    name: ATC_739C5A Humidity
    accuracy_decimals: 1
  battery_level: 
    name: ATC_739C5A Battery-Level
  battery_voltage: 
    name: ATC_739C5A Battery-Voltage
```

nagyrobi, 2022-03-20:
Hm. Works for me...

PWnet, 2022-03-21:
Resolved!

I still had the old "pvvx_mithermometer" custom integration (airy10/pvvx_mithermometer) files inside custom_components folder.

After I renamed the "pvvx_mithermometer" and restarted the container I was able to compile the firmware.

markkamp, 2022-03-21:
Thanks for sharing. Same fix helped for me!

nagyrobi, 2022-03-21:
I guess this can be closed...

closed 2022-03-21 by PWnet