esphome/issues #3150: WebServer fails to display main > section:nth-child(1) > esp-entity-table if esp32_ble_tracker is used

jgvary, 2022-03-17:
### The problem

If I have esp32_ble_tracker in my config, the table esp-entity-table and OTA Update fail to load on the webpage.  No errors on logs?

With esp32_ble_tracker

[image]

Without esp32_ble_tracker

[image]

### Which version of ESPHome has the issue?

2022.3.0 Beta

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.3.5

### What platform are you using?

ESP32

### Board

nodemvu

### Component causing the issue

esp32_ble_tracker - Webserver

### Example YAML snippet

```yaml
esphome:
  name: kitchenesp32
  platform: ESP32
  board: nodemcu-32s

wifi:
  ssid: "DEA1"
  password: "********"
  
  manual_ip:
    static_ip: 192.168.118.54
    gateway: 192.168.118.1
    subnet: 255.255.255.0
    dns1: 192.168.118.1
    dns2: 8.8.8.8

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Kitchen Esp32 Fallback Hotspot"
    password: "***********"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:
  reboot_timeout: 5 min
  
ota:

web_server:
  port: 80

esp32_ble_tracker:

i2c:
  sda: 21
  scl: 22
  scan: true

sensor:
  - platform: xiaomi_lywsd03mmc
    mac_address: "A4:C1:38:EC:24:8E"
    bindkey: "eef418daf699a0c188f3bfd17e4565d9"
    temperature:
      name: "Upstairs Refrigerator Temperature"
    humidity:
      name: "Upstairs Refrigerator Humidity"
    battery_level:
      name: "Upstairs Refrigerator Battery Level"
      
  - platform: xiaomi_lywsd03mmc
    mac_address: "A4:C1:38:87:79:32"
    bindkey: "ec0ab090a8c5d9735dcf8b83526d1a69"
    temperature:
      name: "Upstairs Freezer Temperature"
    humidity:
      name: "Upstairs Freezer Humidity"
    battery_level:
      name: "Upstairs Freezer Battery Level"

  - platform: xiaomi_lywsd03mmc
    mac_address: "A4:C1:38:81:AA:A0"
    bindkey: "033be86368e49e367c2368139b79702e"
    temperature:
      name: "Master Bathroom Temperature"
    humidity:
      name: "Master Bathroom Humidity"
    battery_level:
      name: "Master Bathroom Battery Level"
      
  - platform: xiaomi_lywsd03mmc
    mac_address: "A4:C1:38:80:B2:9F"
    bindkey: "14e074c1488f8254b1be64cced8f2a42"
    temperature:
      name: "Kids Bathroom Temperature"
    humidity:
      name: "Kids Bathroom Humidity"
    battery_level:
      name: "Kids Bathroom Battery Level"

  - platform: xiaomi_lywsd03mmc
    mac_address: "A4:C1:38:BF:68:A2"
    bindkey: "b9f163d0b084ea305729c285f978e463"
    temperature:
      name: "Master Bedroom Temperature"
    humidity:
      name: "Master Bedroom Humidity"
    battery_level:
      name: "Master Bedroom Battery Level"
      
  - platform: aht10
    temperature:
      name: "Kitchen Temperature"
      accuracy_decimals: 0
    humidity:
      name: "Kitchen Humidity"
      accuracy_decimals: 0
    update_interval: 5s

binary_sensor:
  - platform: gpio
    pin: GPIO32
    name: "Kitchen Motion"
    device_class: motion
switch:  
  - platform: restart
    name: "Kitchen restart"
```

### Anything in the logs that might be useful for us?

```txt
Nothing
```

### Additional information

YAML snippet of working Webserver

esphome:
  name: kitchenesp32
  platform: ESP32
  board: nodemcu-32s

wifi:
  ssid: "DEA1"
  password: "********"
  
  manual_ip:
    static_ip: 192.168.118.54
    gateway: 192.168.118.1
    subnet: 255.255.255.0
    dns1: 192.168.118.1
    dns2: 8.8.8.8

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Kitchen Esp32 Fallback Hotspot"
    password: "***********"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:
  reboot_timeout: 5 min
  
ota:

web_server:
  port: 80

i2c:
  sda: 21
  scl: 22
  scan: true

sensor:
  - platform: aht10
    temperature:
      name: "Kitchen Temperature"
      accuracy_decimals: 0
    humidity:
      name: "Kitchen Humidity"
      accuracy_decimals: 0
    update_interval: 5s

binary_sensor:
  - platform: gpio
    pin: GPIO32
    name: "Kitchen Motion"
    device_class: motion
switch:  
  - platform: restart
    name: "Kitchen restart"

flystyle, 2022-03-18:
Same problem, still no way to fix it?

stylus1975, 2022-03-20:
I also have the same problem.

TBobsin, 2022-03-21:
I also have the same problem. 
Tryed to chage web_port: 81 no change

jgvary, 2022-03-29:
Seems to be resolved in 2022.3.1 Beta

closed 2022-03-29 by jgvary