esphome/issues #3151: api.connection:083 and Display does not work

minimag450, 2022-03-17:
### The problem

I can't find the problem. I have a very simple skech, but the temperature from the sensor is not shown on the display. I get the error [api.connection:083] ... Connection closed. What I also don't understand are the log files. If I display the log file when the ESP is plugged into the Raspberry, nothing else comes up. The time is also not displayed, only the elapsed time, for example 1:05:30. If I then create the log file via wireless, the clock works after a few seconds (I think when the WLAN connection is established) but the temperature is still not displayed. I have already tried the other answers in the issues, but it still doesn't work. I have also tried a D1 mini and an ESP8266 with the correct PINs, but the temperature is not displayed.

Translated with www.DeepL.com/Translator (free version)

### Which version of ESPHome has the issue?

2022.3.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.3.5

### What platform are you using?

ESP32

### Board

ESP32-WROOM-32

### Example YAML snippet

```yaml
esphome:
  name: anzeige1zoll
  
esp32:
  board: nodemcu-32s
  
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_pass
  

#    ssid: "anzeige1zoll Fallback Hotspot"
#    password: "secretpassword"
    
captive_portal:

logger:
   
api:
   password: !secret password 

ota:
   password: !secret password
  
time:
  - platform: homeassistant
    id: esptime

sensor:
  - platform: homeassistant
    id: aussen_temperatur
    entity_id: sensor.temperaturmelder_funk_2
#    internal: true

color:
  - id: my_red
    red: 90%
    green: 80%
    blue: 80%

font:
[... 341 lines cut ...]
[20:56:05][C][homeassistant.time:011]:   Timezone: 'CET-1CEST,M3.5.0,M10.5.0/3'
[20:56:05][C][ili9341:030]: ili9341
[20:56:05][C][ili9341:030]:   Rotations: 270 °
[20:56:05][C][ili9341:030]:   Dimensions: 320px x 240px
[20:56:05][C][ili9341:031]:   Width: 240, Height: 320,  Rotation: 270
[20:56:05][C][ili9341:032]:   Reset Pin: GPIO33
[20:56:05][C][ili9341:033]:   DC Pin: GPIO27
[20:56:05][C][ili9341:035]:   Backlight Pin: GPIO32
[20:56:05][C][ili9341:036]:   Update Interval: 1.0s
[20:56:05][C][captive_portal:088]: Captive Portal:
[20:56:05][C][mdns:084]: mDNS:
[20:56:05][C][mdns:085]:   Hostname: anzeige1zoll
[20:56:05][C][ota:085]: Over-The-Air Updates:
[20:56:05][C][ota:086]:   Address: anzeige1zoll.local:3232
[20:56:05][C][ota:089]:   Using Password.
[20:56:05][C][api:138]: API Server:
[20:56:05][C][api:139]:   Address: anzeige1zoll.local:6053
[20:56:05][C][api:143]:   Using noise encryption: NO
[20:56:05][C][homeassistant.sensor:030]: Homeassistant Sensor 'aussen_temperatur'
[20:56:05][C][homeassistant.sensor:030]:   State Class: ''
[20:56:05][C][homeassistant.sensor:030]:   Unit of Measurement: ''
[20:56:05][C][homeassistant.sensor:030]:   Accuracy Decimals: 1
[20:56:05][C][homeassistant.sensor:031]:   Entity ID: 'sensor.temperaturmelder_funk_2'
[20:56:56][D][api:102]: Accepted ::FFFF:C0A8:B251
[20:56:56][W][api.connection:083]: Home Assistant 2022.3.5 (::FFFF:C0A8:B251): Connection closed

ThijsVredenbregt, 2022-03-20:
I also have this issue

davidsat, 2022-03-21:
I also have this issue. Started around 3-4 weeks ago. I see "connection closed" messages every minute, and ESPHOME does send the device info to Home Assistant.

HumanSkunk, 2022-03-24:
Same issue here. In the end as a test I flashed the same board with a default config with nothing attached. I can see logs when connected locally, but they always end with "Connection closed". The board I am having issues with has been refreshed a few times with slightly different yaml configs and a different device name.

The board I am having issues with never originally showed up in HA as a device (which every other esp configured through ESPHome has.)

I am going to try a completely fresh board which has never been configured nor connected  tomorrow and see what happens.

ThijsVredenbregt, 2022-03-24:
this started working auto-magically yesterday. Dont know what changed...

rafalyesware, 2022-07-05:
I had this issue as a result of renaming a `esphome-tester` node (which was initially just that) to `basement-minisplit` in the ESPHome dashboard.  I'm not sure how, but the rename created the new `basement-minisplit` entity and still left the old `esphome-tester` one.  It's possible that I then tried to rename the device as well in Settings > Devices, but that part I'm less sure of.

Unfortunately to fix it, I had to hand-edit a bunch of things in `/config/.storage` (device registry, entity registry and/or entities as well as the `esphome.<device-id>` file) to get rid of the duplicate and finish the rename.  Ugh, but at least it's all happy now.

closed 2022-11-19 by github-actions[bot]