esphome/issues #3155: Binary Sensor device_class: 'co' not available.

jasonhun, 2022-03-18:
### The problem

The 'co' device class listed as a valid device class on the documentation is not available.  An error says:

Unknown value 'co', did you mean 'cold'?

[image]

### Which version of ESPHome has the issue?

2022.3.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.3.5

### What platform are you using?

ESP8266

### Board

nodemcu

### Example YAML snippet

```yaml
binary_sensor:
  - platform: gpio
    id: co_alarm
    pin: 
      number: D2
    name: "CO Alarm"
    device_class: co
    filters:
      - delayed_off: 10ms
```

tomadan1968, 2022-03-20:
Hello, t
[image]
[image]
[image]
[image]
he same problem...

kbx81, 2023-07-25:
`carbon_monoxide` is the correct `device_class` to use here -- right now, this is the only exception to what's listed [here](https://developers.home-assistant.io/docs/core/entity/binary-sensor/#available-device-classes).

closed 2023-07-25 by kbx81

ssieb, 2023-07-29:
It's not even really an exception.  Those are Python constant names, not values.  The problem is that this is the one case where the constant name doesn't match the value.