esphome/issues #3156: Error on CMake while installing on a Adafruit ESP32-S2

opintos, 2022-03-18:
### The problem

Impossible to install, CMake crashes on the beginning.

### Which version of ESPHome has the issue?

2022.3.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.3.5

### What platform are you using?

ESP32

### Board

Adafruit ESP32-S2 Feather

### Component causing the issue

CMake

### Example YAML snippet

```yaml
esphome:
  name: dormitorio

esp32:
  board: esp32-s2-saola-1
  framework:
    type: esp-idf

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:
  password: "************************"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
```

### Anything in the logs that might be useful for us?

```txt
INFO Reading configuration /config/esphome/dormitorio.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing dormitorio (board: esp32-s2-saola-1; framework: espidf; platform: platformio/espressif32 @ 3.5.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32S2 240MHz, 320KB RAM, 4MB Flash
 - framework-espidf 3.40302.0 (4.3.2) 
 - tool-cmake 3.16.9 
 - tool-ninja 1.10.2 
 - toolchain-riscv32-esp 8.4.0+2021r2-patch2 
 - toolchain-xtensa-esp32s2 8.4.0+2021r2-patch2
Reading CMake configuration...
-- The C compiler identification is unknown
-- The CXX compiler identification is unknown
-- The ASM compiler identification is unknown
-- Found assembler: /data/cache/platformio/packages/toolchain-xtensa-esp32s2/bin/xtensa-esp32s2-elf-gcc
-- Check for working C compiler: /data/cache/platformio/packages/toolchain-xtensa-esp32s2/bin/xtensa-esp32s2-elf-gcc
-- Check for working C compiler: /data/cache/platformio/packages/toolchain-xtensa-esp32s2/bin/xtensa-esp32s2-elf-gcc -- broken
-- Configuring incomplete, errors occurred!
See also "/data/dormitorio/.pioenvs/dormitorio/CMakeFiles/CMakeOutput.log".
See also "/data/dormitorio/.pioenvs/dormitorio/CMakeFiles/CMakeError.log".

fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
CMake Error at /data/cache/platformio/packages/tool-cmake/share/cmake-3.16/Modules/CMakeTestCCompiler.cmake:60 (message):
  The C compiler

    "/data/cache/platformio/packages/toolchain-xtensa-esp32s2/bin/xtensa-esp32s2-elf-gcc"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: /data/dormitorio/.pioenvs/dormitorio/CMakeFiles/CMakeTmp
    
    Run Build Command(s):/data/cache/platformio/packages/tool-ninja/ninja cmTC_d1c7e && [1/2] Building C object CMakeFiles/cmTC_d1c7e.dir/testCCompiler.c.obj
    FAILED: CMakeFiles/cmTC_d1c7e.dir/testCCompiler.c.obj 
    /data/cache/platformio/packages/toolchain-xtensa-esp32s2/bin/xtensa-esp32s2-elf-gcc   -mlongcalls -o CMakeFiles/cmTC_d1c7e.dir/testCCompiler.c.obj   -c testCCompiler.c
    /bin/sh: 1: /data/cache/platformio/packages/toolchain-xtensa-esp32s2/bin/xtensa-esp32s2-elf-gcc: not found
    ninja: build stopped: subcommand failed.
    
    

  

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  /data/cache/platformio/packages/framework-espidf/tools/cmake/project.cmake:311 (__project)
  CMakeLists.txt:3 (project)

========================= [FAILED] Took 26.38 seconds =========================
```

photoxor, 2022-03-23:
Same error for DIGISHUO ESP-WROOM-32 ESP32S board.

JoeArny, 2022-04-03:
Hi guys, experiencing the exact same issue, have anyone found a solution or workaround? Thanks.

brauhausdc, 2022-04-10:
Also having this same issue. Board is an Adafruit Metro-esp32-s2. 

https://learn.adafruit.com/adafruit-metro-esp32-s2

brauhausdc, 2022-05-12:
Any updates on this -- still seeing the error. Is this on anyone's radar?

JoeArny, 2022-05-12:
I managed to workaround the issue installing esphome in a docker container. Not even tried again in my hassio...

brauhausdc, 2022-05-12:
What platform did you use for Docker? Did you do a full Home Assistant install there, or did you have separate ESPhome / HASS.io installs?

JoeArny, 2022-05-12:
> What platform did you use for Docker? Did you do a full Home Assistant install there, or did you have separate ESPhome / HASS.io installs?

I have a truenas server, just installing esphome there was enough to do the trick.
It's a shame it does not work directly on hassio...

closed 2022-10-26 by github-actions[bot]