esphome/issues #3159: AirThings BLE Sensors - Values 'Unknown'

galambert75, 2022-03-18:
### The problem

I'm trying to use an ESP32 as a BLE hub for my [AirThings Wave 2nd Generation](https://www.amazon.es/gp/product/B081SFPGBV) 

Using the same configuration as provided [here](https://esphome.io/components/sensor/airthings_ble.html) 

Connection is successful, but no values are read.

### Which version of ESPHome has the issue?

2022.3.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.3.5

### What platform are you using?

ESP32

### Board

ESP32-WROOM-32D

### Component causing the issue

AirThings BLE Sensors

### Example YAML snippet

```yaml
esp32_ble_tracker:

ble_client:
  - mac_address: 58:93:D8:8C:4D:A1
    id: airthings01

sensor:
  - platform: airthings_wave_plus
    ble_client_id: airthings01
    update_interval: 5min # default
    temperature:
      name: "WavePlus Temperature"
    radon:
      name: "WavePlus Radon"
    radon_long_term:
      name: "WavePlus Radon Long Term"
    pressure:
      name: "WavePlus Pressure"
    humidity:
      name: "WavePlus Humidity"
    co2:
      name: "WavePlus CO2"
    tvoc:
      name: "WavePlus VOC"
```

### Anything in the logs that might be useful for us?

```txt
[21:15:06][I][app:102]: ESPHome version 2022.3.0 compiled on Mar 18 2022, 21:10:54
[...]
[21:15:06][C][logger:233]: Logger:
[21:15:06][C][logger:234]:   Level: DEBUG
[21:15:06][C][logger:235]:   Log Baud Rate: 115200
[21:15:06][C][logger:236]:   Hardware UART: UART0
[21:15:06][C][airthings_wave_plus:120]:   Humidity 'WavePlus Humidity'
[21:15:06][C][airthings_wave_plus:120]:     Device Class: 'humidity'
[21:15:06][C][airthings_wave_plus:120]:     State Class: 'measurement'
[21:15:06][C][airthings_wave_plus:120]:     Unit of Measurement: '%'
[21:15:06][C][airthings_wave_plus:120]:     Accuracy Decimals: 0
[21:15:06][C][airthings_wave_plus:121]:   Radon 'WavePlus Radon'
[21:15:06][C][airthings_wave_plus:121]:     State Class: 'measurement'
[21:15:06][C][airthings_wave_plus:121]:     Unit of Measurement: 'Bq/m³'
[21:15:06][C][airthings_wave_plus:121]:     Accuracy Decimals: 0
[21:15:06][C][airthings_wave_plus:121]:     Icon: 'mdi:radioactive'
[21:15:06][C][airthings_wave_plus:122]:   Radon Long Term 'WavePlus Radon Long Term'
[21:15:06][C][airthings_wave_plus:122]:     State Class: 'measurement'
[21:15:06][C][airthings_wave_plus:122]:     Unit of Measurement: 'Bq/m³'
[21:15:06][C][airthings_wave_plus:122]:     Accuracy Decimals: 0
[21:15:06][C][airthings_wave_plus:122]:     Icon: 'mdi:radioactive'
[21:15:06][C][airthings_wave_plus:123]:   Temperature 'WavePlus Temperature'
[21:15:06][C][airthings_wave_plus:123]:     Device Class: 'temperature'
[21:15:06][C][airthings_wave_plus:123]:     State Class: 'measurement'
[21:15:06][C][airthings_wave_plus:123]:     Unit of Measurement: '°C'
[21:15:06][C][airthings_wave_plus:123]:     Accuracy Decimals: 2
[21:15:06][C][airthings_wave_plus:124]:   Pressure 'WavePlus Pressure'
[21:15:06][C][airthings_wave_plus:124]:     Device Class: 'pressure'
[21:15:06][C][airthings_wave_plus:124]:     State Class: 'measurement'
[21:15:06][C][airthings_wave_plus:124]:     Unit of Measurement: 'hPa'
[21:15:06][C][airthings_wave_plus:124]:     Accuracy Decimals: 1
[21:15:06][C][airthings_wave_plus:125]:   CO2 'WavePlus CO2'
[21:15:06][C][airthings_wave_plus:125]:     Device Class: 'carbon_dioxide'
[21:15:06][C][airthings_wave_plus:125]:     State Class: 'measurement'
[21:15:06][C][airthings_wave_plus:125]:     Unit of Measurement: 'ppm'
[21:15:06][C][airthings_wave_plus:125]:     Accuracy Decimals: 0
[21:15:06][C][airthings_wave_plus:126]:   TVOC 'WavePlus VOC'
[21:15:06][C][airthings_wave_plus:126]:     State Class: 'measurement'
[21:15:06][C][airthings_wave_plus:126]:     Unit of Measurement: 'ppb'
[21:15:06][C][airthings_wave_plus:126]:     Accuracy Decimals: 0
[21:15:06][C][airthings_wave_plus:126]:     Icon: 'mdi:radiator'
[21:15:06][C][esp32_ble_tracker:709]: BLE Tracker:
[21:15:06][C][esp32_ble_tracker:710]:   Scan Duration: 300 s
[21:15:06][C][esp32_ble_tracker:711]:   Scan Interval: 320.0 ms
[21:15:06][C][esp32_ble_tracker:712]:   Scan Window: 30.0 ms
[21:15:06][C][esp32_ble_tracker:713]:   Scan Type: ACTIVE
[21:15:06][C][ble_client:035]: BLE Client:
[21:15:06][C][ble_client:036]:   Address: 58:93:d8:8c:4d:a1
[...]
[21:15:16][D][ble_client:047]: Found device at MAC address [58:93:D8:8C:4D:A1]
[21:15:16][I][ble_client:085]: Attempting BLE connection to 58:93:d8:8c:4d:a1
[21:15:17][I][airthings_wave_plus:015]: Connected successfully!
[21:15:18][I][ble_client:161]: Service UUID: 0x1800
[21:15:18][I][ble_client:162]:   start_handle: 0x1  end_handle: 0x9
[21:15:18][I][ble_client:347]:  characteristic 0x2A00, handle 0x3, properties 0x2
[21:15:18][I][ble_client:347]:  characteristic 0x2A01, handle 0x5, properties 0x2
[21:15:18][I][ble_client:347]:  characteristic 0x2A04, handle 0x7, properties 0x2
[21:15:18][I][ble_client:347]:  characteristic 0x2AC9, handle 0x9, properties 0x2
[21:15:18][I][ble_client:161]: Service UUID: 0x1801
[21:15:18][I][ble_client:162]:   start_handle: 0xa  end_handle: 0xa
[21:15:18][I][ble_client:161]: Service UUID: B42E4A8E-ADE7-11E4-89D3-123B93F75CBA
[21:15:18][I][ble_client:162]:   start_handle: 0xb  end_handle: 0x16
[21:15:18][I][ble_client:347]:  characteristic B42E4DCC-ADE7-11E4-89D3-123B93F75CBA, handle 0xd, properties 0x2
[21:15:18][I][ble_client:347]:  characteristic B42E50D8-ADE7-11E4-89D3-123B93F75CBA, handle 0x10, properties 0x2c
[21:15:18][I][ble_client:347]:  characteristic B42E538A-ADE7-11E4-89D3-123B93F75CBA, handle 0x14, properties 0x10
[21:15:18][I][ble_client:161]: Service UUID: F000FFC0-0451-4000-B000-000000000000
[21:15:18][I][ble_client:162]:   start_handle: 0x17  end_handle: 0x23
[21:15:18][I][ble_client:347]:  characteristic F000FFC1-0451-4000-B000-000000000000, handle 0x19, properties 0x1c
[21:15:18][I][ble_client:347]:  characteristic F000FFC2-0451-4000-B000-000000000000, handle 0x1d, properties 0x1c
[21:15:18][I][ble_client:347]:  characteristic F000FFC5-0451-4000-B000-000000000000, handle 0x21, properties 0x14
[21:15:18][I][ble_client:161]: Service UUID: 0x180A
[21:15:18][I][ble_client:162]:   start_handle: 0x24  end_handle: 0xffff
[21:15:18][I][ble_client:347]:  characteristic 0x2A23, handle 0x26, properties 0x2
[21:15:18][I][ble_client:347]:  characteristic 0x2A24, handle 0x28, properties 0x2
[21:15:18][I][ble_client:347]:  characteristic 0x2A25, handle 0x2a, properties 0x2
[21:15:18][I][ble_client:347]:  characteristic 0x2A26, handle 0x2c, properties 0x2
[21:15:18][I][ble_client:347]:  characteristic 0x2A27, handle 0x2e, properties 0x2
[21:15:18][I][ble_client:347]:  characteristic 0x2A29, handle 0x30, properties 0x2
[21:15:18][W][airthings_wave_plus:030]: No sensor characteristic found at service B42E1C08-ADE7-11E4-89D3-123B93F75CBA char B42E2A68-ADE7-11E4-89D3-123B93F75CBA
```

systematicallyrandom, 2022-03-22:
I'm having the exact same issue with my Wave. I'm wondering if there is a slight difference between the BLE output of the Wave Plus, which the integration was designed for, and the Wave.

galambert75, 2022-03-24:
There's a custom Home Assistant [addon](https://github.com/custom-components/sensor.airthings_wave) that works for all three (Wave, Wave Plus, Wave Mini). Not sure if this is using active or passive BLE. Unfortunately, I don't have the programming skills to contribute code...

However, I'm looking to use an ESP32 so I don't need to have the Wave within a certain radius of the HA server.

loca5790, 2022-03-29:
I'm having the same issue, the addon didn't work consistently for me.  It would cycle between which device it was going to pick up.  It would be random between my mini and plus and once in a blue moon I would have both.  Migrated to esphome for hopefully reliability and ran into this.

galambert75, 2022-03-29:
Turns out there's already a feature request for this: https://github.com/esphome/feature-requests/issues/1467

I would recommend we all upvote.

closed 2022-03-29 by galambert75