esphome/issues #3161: hexencode() is deprecated, use format_hex_pretty()

USAFPride, 2022-03-19:
### The problem

hexencode() is depreciated. See YAML and compiler output

### Which version of ESPHome has the issue?

 2022.3.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.3.5

### What platform are you using?

ESP32

### Example YAML snippet

```yaml
esphome:
  name: bbq-thermostat
  on_boot:
    priority: -10
    then:
      - lambda: |-
          {
            id(ble_sensor_1).publish_state(false);
            id(ble_sensor_2).publish_state(false);
            id(ble_sensor_3).publish_state(false);
            id(ble_sensor_4).publish_state(false);
            id(ble_sensor_5).publish_state(false);
            id(ble_sensor_6).publish_state(false);
          }

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:
  password: "b19b1b31b964b09bacd5025c72e6f94c"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  fast_connect: on

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Bbq-Thermostat Fallback Hotspot"
    password: "6U4jWGSVv0BN"

captive_portal:

script:
  - id: timer
    then:
      - delay: 120s
      - lambda: |-
          {
            id(ble_sensor_1).publish_state(false);
            id(ble_sensor_2).publish_state(false);
            id(ble_sensor_3).publish_state(false);
            id(ble_sensor_4).publish_state(false);
            id(ble_sensor_5).publish_state(false);
            id(ble_sensor_6).publish_state(false);
          }
  
esp32_ble_tracker:
  on_ble_advertise:
    - mac_address: F0:B5:D1:84:ED:73
      then:
        - script.stop: timer
        - lambda: |-
            if (x.get_name() != "iBBQ") return;
            
            ESP_LOGI("ble_adv", "New BLE device");
            ESP_LOGI("ble_adv", "  address: %s", x.address_str().c_str());
            ESP_LOGI("ble_adv", "  name: %s", x.get_name().c_str());
            ESP_LOGI("ble_adv", "  Advertised service UUIDs:");
            
            for (auto uuid : x.get_service_uuids()) {
              ESP_LOGI("ble_adv", "    - %s", uuid.to_string().c_str());
            }
            
            ESP_LOGI("ble_adv", "  Advertised service data:");
            
            for (auto data : x.get_service_datas()) {
              ESP_LOGI("ble_adv", "    - %s: (length %i)", data.uuid.to_string().c_str(), data.data.size());
            }
            
            ESP_LOGI("ble_adv", "  Advertised manufacturer data:");
            
            for (auto data : x.get_manufacturer_datas()) {
              ESP_LOGI("ble_adv", "    - %s: (%s)", data.uuid.to_string().c_str(), hexencode(data.data).c_str());
              
              if (data.uuid.contains(0, 0)) {
    
                int probe0 = (data.data[9] << 8) + data.data[8];
                int probe1 = (data.data[11] << 8) + data.data[10];
                int probe2 = (data.data[13] << 8) + data.data[12];
                int probe3 = (data.data[15] << 8) + data.data[14];
                int probe4 = (data.data[17] << 8) + data.data[16];
                int probe5 = (data.data[19] << 8) + data.data[18];
                    
                ESP_LOGI("ble_data", "    - %f %f", probe0 / 10.0, probe1 / 10.0);
    
                if (probe0 < 60000) {
                  id(ble_sensor_1).publish_state(probe0 / 10.0);
                } else {
                  id(ble_sensor_1).publish_state(0);                
                }
                if (probe1 < 60000) {
                  id(ble_sensor_2).publish_state(probe1 / 10.0);
                } else {
                  id(ble_sensor_2).publish_state(0);                
                }
                if (probe2 < 60000) {
                  id(ble_sensor_3).publish_state(probe2 / 10.0);
                } else {
                  id(ble_sensor_3).publish_state(0);                
                }
                if (probe3 < 60000) {
                  id(ble_sensor_4).publish_state(probe3 / 10.0);
                } else {
                  id(ble_sensor_4).publish_state(0);                
                }
                if (probe4 < 60000) {
                  id(ble_sensor_5).publish_state(probe4 / 10.0);
                } else {
                  id(ble_sensor_5).publish_state(0);                
                }
                if (probe5 < 60000) {
                  id(ble_sensor_6).publish_state(probe5 / 10.0);
                } else {
                  id(ble_sensor_6).publish_state(0);                
                }
              }
            }
        - script.execute: timer
  
sensor:
  - platform: template
    name: "iBBQ Temperature Probe 1"
    id: ble_sensor_1
    unit_of_measurement: "°C"
    accuracy_decimals: 0

  - platform: template
    name: "iBBQ Temperature Probe 2"
    id: ble_sensor_2
    unit_of_measurement: "°C"
    accuracy_decimals: 0

  - platform: template
    name: "iBBQ Temperature Probe 3" 
    id: ble_sensor_3 # <- Increment the ble_sensor by one
    unit_of_measurement: "°C"
    accuracy_decimals: 0

  - platform: template
    name: "iBBQ Temperature Probe 4"
    id: ble_sensor_4
    unit_of_measurement: "°C"
    accuracy_decimals: 0

  - platform: template
    name: "iBBQ Temperature Probe 5"
    id: ble_sensor_5
    unit_of_measurement: "°C"
    accuracy_decimals: 0

  - platform: template
    name: "iBBQ Temperature Probe 6"
    id: ble_sensor_6
    unit_of_measurement: "°C"
    accuracy_decimals: 0
```

### Anything in the logs that might be useful for us?

```txt
In file included from src/esphome/components/sensor/sensor.h:3:0,
                 from src/esphome/core/application.h:16,
                 from src/esphome/components/api/api_connection.h:4,
                 from src/esphome.h:3,
                 from src/main.cpp:3:
/config/esphome/bbq-thermostat.yaml: In lambda function:
/config/esphome/bbq-thermostat.yaml:83:97: warning: 'std::__cxx11::string esphome::hexencode(const T&) [with T = std::vector<unsigned char>; std::__cxx11::string = std::__cxx11::basic_string<char>]' is deprecated: hexencode() is deprecated, use format_hex_pretty() instead. [-Wdeprecated-declarations]
               ESP_LOGI("ble_adv", "    - %s: (%s)", data.uuid.to_string().c_str(), hexencode(data.data).c_str());
                                                                                                 ^
src/esphome/core/log.h:105:88: note: in definition of macro 'esph_log_i'
   esp_log_printf_(ESPHOME_LOG_LEVEL_INFO, tag, __LINE__, ESPHOME_LOG_FORMAT(format), ##__VA_ARGS__)
                                                                                        ^
/config/esphome/bbq-thermostat.yaml:83:9: note: in expansion of macro 'ESP_LOGI'
               ESP_LOGI("ble_adv", "    - %s: (%s)", data.uuid.to_string().c_str(), hexencode(data.data).c_str());
         ^
In file included from src/esphome/core/preferences.h:6:0,
                 from src/esphome/core/application.h:6,
                 from src/esphome/components/api/api_connection.h:4,
                 from src/esphome.h:3,
                 from src/main.cpp:3:
src/esphome/core/helpers.h:648:13: note: declared here
 std::string hexencode(const T &data) {
             ^
src/esphome/core/helpers.h: In instantiation of 'std::__cxx11::string esphome::hexencode(const T&) [with T = std::vector<unsigned char>; std::__cxx11::string = std::__cxx11::basic_string<char>]':
/config/esphome/bbq-thermostat.yaml:83:9:   required from here
src/esphome/core/helpers.h:649:19: warning: 'std::__cxx11::string esphome::hexencode(const uint8_t*, uint32_t)' is deprecated: hexencode() is deprecated, use format_hex_pretty() instead. [-Wdeprecated-declarations]
   return hexencode(data.data(), data.size());
                   ^
src/esphome/core/helpers.h:644:20: note: declared here
 inline std::string hexencode(const uint8_t *data, uint32_t len) { return format_hex_pretty(data, len); }
                    ^
```

topic2k, 2022-04-23:
Yesterday i came across this warning too. As suggested i replaced `hexencode` with `format_hex_pretty` and the warning was gone and i get the temperatures from the sensors.

ssieb, 2022-08-22:
I don't understand the point of this issue.  The message explains what to do.

closed 2022-08-22 by ssieb