esphome/issues #3162: incompatible pointer type [-Wincompatible-pointer-types]

USAFPride, 2022-03-19:
### The problem

incompatible pointer type [-Wincompatible-pointer-types] warn thrown during compile. 

Prior versions did not throw the warning. Board seems to be unstable (reboots randomly) after update where stable before (may be unrelated).

### Which version of ESPHome has the issue?

 2022.3.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.3.5

### What platform are you using?

ESP32

### Example YAML snippet

```yaml
esphome:
  name: bbq-thermostat
  on_boot:
    priority: -10
    then:
      - lambda: |-
          {
            id(ble_sensor_1).publish_state(false);
            id(ble_sensor_2).publish_state(false);
            id(ble_sensor_3).publish_state(false);
            id(ble_sensor_4).publish_state(false);
            id(ble_sensor_5).publish_state(false);
            id(ble_sensor_6).publish_state(false);
          }

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:
  password: "b19b1b31b964b09bacd5025c72e6f94c"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  fast_connect: on

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Bbq-Thermostat Fallback Hotspot"
    password: "6U4jWGSVv0BN"

captive_portal:

script:
  - id: timer
    then:
      - delay: 120s
      - lambda: |-
          {
            id(ble_sensor_1).publish_state(false);
            id(ble_sensor_2).publish_state(false);
            id(ble_sensor_3).publish_state(false);
            id(ble_sensor_4).publish_state(false);
            id(ble_sensor_5).publish_state(false);
            id(ble_sensor_6).publish_state(false);
          }
  
esp32_ble_tracker:
  on_ble_advertise:
    - mac_address: F0:B5:D1:84:ED:73
      then:
        - script.stop: timer
        - lambda: |-
            if (x.get_name() != "iBBQ") return;
            
            ESP_LOGI("ble_adv", "New BLE device");
            ESP_LOGI("ble_adv", "  address: %s", x.address_str().c_str());
            ESP_LOGI("ble_adv", "  name: %s", x.get_name().c_str());
            ESP_LOGI("ble_adv", "  Advertised service UUIDs:");
            
            for (auto uuid : x.get_service_uuids()) {
              ESP_LOGI("ble_adv", "    - %s", uuid.to_string().c_str());
            }
            
            ESP_LOGI("ble_adv", "  Advertised service data:");
            
            for (auto data : x.get_service_datas()) {
              ESP_LOGI("ble_adv", "    - %s: (length %i)", data.uuid.to_string().c_str(), data.data.size());
            }
            
            ESP_LOGI("ble_adv", "  Advertised manufacturer data:");
            
            for (auto data : x.get_manufacturer_datas()) {
              ESP_LOGI("ble_adv", "    - %s: (%s)", data.uuid.to_string().c_str(), hexencode(data.data).c_str());
              
              if (data.uuid.contains(0, 0)) {
    
                int probe0 = (data.data[9] << 8) + data.data[8];
                int probe1 = (data.data[11] << 8) + data.data[10];
                int probe2 = (data.data[13] << 8) + data.data[12];
                int probe3 = (data.data[15] << 8) + data.data[14];
                int probe4 = (data.data[17] << 8) + data.data[16];
                int probe5 = (data.data[19] << 8) + data.data[18];
                    
                ESP_LOGI("ble_data", "    - %f %f", probe0 / 10.0, probe1 / 10.0);
    
                if (probe0 < 60000) {
                  id(ble_sensor_1).publish_state(probe0 / 10.0);
                } else {
                  id(ble_sensor_1).publish_state(0);                
                }
                if (probe1 < 60000) {
                  id(ble_sensor_2).publish_state(probe1 / 10.0);
                } else {
                  id(ble_sensor_2).publish_state(0);                
                }
                if (probe2 < 60000) {
                  id(ble_sensor_3).publish_state(probe2 / 10.0);
                } else {
                  id(ble_sensor_3).publish_state(0);                
                }
                if (probe3 < 60000) {
                  id(ble_sensor_4).publish_state(probe3 / 10.0);
                } else {
                  id(ble_sensor_4).publish_state(0);                
                }
                if (probe4 < 60000) {
                  id(ble_sensor_5).publish_state(probe4 / 10.0);
                } else {
                  id(ble_sensor_5).publish_state(0);                
                }
                if (probe5 < 60000) {
                  id(ble_sensor_6).publish_state(probe5 / 10.0);
                } else {
                  id(ble_sensor_6).publish_state(0);                
                }
              }
            }
        - script.execute: timer
  
sensor:
  - platform: template
    name: "iBBQ Temperature Probe 1"
    id: ble_sensor_1
    unit_of_measurement: "°C"
    accuracy_decimals: 0

  - platform: template
    name: "iBBQ Temperature Probe 2"
    id: ble_sensor_2
    unit_of_measurement: "°C"
    accuracy_decimals: 0

  - platform: template
    name: "iBBQ Temperature Probe 3" 
    id: ble_sensor_3 # <- Increment the ble_sensor by one
    unit_of_measurement: "°C"
    accuracy_decimals: 0

  - platform: template
    name: "iBBQ Temperature Probe 4"
    id: ble_sensor_4
    unit_of_measurement: "°C"
    accuracy_decimals: 0

  - platform: template
    name: "iBBQ Temperature Probe 5"
    id: ble_sensor_5
    unit_of_measurement: "°C"
    accuracy_decimals: 0

  - platform: template
    name: "iBBQ Temperature Probe 6"
    id: ble_sensor_6
    unit_of_measurement: "°C"
    accuracy_decimals: 0
```

### Anything in the logs that might be useful for us?

```txt
Compiling /data/bbq-thermostat/.pioenvs/bbq-thermostat/FrameworkArduino/esp32-hal-time.c.o
/data/cache/platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-spi.c: In function 'spiTransferBytesNL':
/data/cache/platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-spi.c:922:39: warning: initialization from incompatible pointer type [-Wincompatible-pointer-types]
                 uint8_t * last_out8 = &result[c_longs-1];
                                       ^
/data/cache/platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-spi.c:923:40: warning: initialization from incompatible pointer type [-Wincompatible-pointer-types]
                 uint8_t * last_data8 = &last_data;
                                        ^
```

martgras, 2022-04-13:
That's just a compiler warning you can ignore.
The warning itself can only be fixed in the Arduino framework not in esphome

mmakaay, 2022-06-23:
Duplicate of #2177

closed 2022-06-23 by mmakaay