esphome/issues #3163: Install failing - CMake error - ESP-C3-32S dev kit

roesel, 2022-03-20:
### The problem

Unsuccessful install, CMake crashes.

### Which version of ESPHome has the issue?

2022.3.0

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2022.3.5

### What platform are you using?

ESP32-IDF

### Board

NodeMCU ESP-C3-32S-Kit

### Component causing the issue

CMake

### Example YAML snippet

```yaml
esphome:
  name: mhz

esp32:
  board: esp32-c3-devkitm-1
  framework:
    type: esp-idf

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:
  password: "99b7798d9ee9a4b37da99f4b0c94596e"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot in case wifi connection fails
  ap:
    ssid: "Mhz Fallback Hotspot"
    password: "wGkCRSlYdC6x"
```

### Anything in the logs that might be useful for us?

```txt
INFO Reading configuration /config/mhz.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing mhz (board: esp32-c3-devkitm-1; framework: espidf; platform: platformio/espressif32 @ 3.5.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32C3 160MHz, 320KB RAM, 4MB Flash
 - framework-espidf 3.40302.0 (4.3.2) 
 - tool-cmake 3.16.9 
 - tool-ninja 1.10.2 
 - toolchain-riscv32-esp 8.4.0+2021r2-patch2 
 - toolchain-xtensa-esp32s2 8.4.0+2021r2-patch2
Reading CMake configuration...
-- The C compiler identification is unknown
-- The CXX compiler identification is unknown
-- The ASM compiler identification is unknown
-- Found assembler: /config/.esphome/platformio/packages/toolchain-riscv32-esp/bin/riscv32-esp-elf-gcc
-- Check for working C compiler: /config/.esphome/platformio/packages/toolchain-riscv32-esp/bin/riscv32-esp-elf-gcc
-- Check for working C compiler: /config/.esphome/platformio/packages/toolchain-riscv32-esp/bin/riscv32-esp-elf-gcc -- broken
-- Configuring incomplete, errors occurred!
See also "/config/.esphome/build/mhz/.pioenvs/mhz/CMakeFiles/CMakeOutput.log".
See also "/config/.esphome/build/mhz/.pioenvs/mhz/CMakeFiles/CMakeError.log".

fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
CMake Error at /config/.esphome/platformio/packages/tool-cmake/share/cmake-3.16/Modules/CMakeTestCCompiler.cmake:60 (message):
  The C compiler

    "/config/.esphome/platformio/packages/toolchain-riscv32-esp/bin/riscv32-esp-elf-gcc"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: /config/.esphome/build/mhz/.pioenvs/mhz/CMakeFiles/CMakeTmp
    
    Run Build Command(s):/config/.esphome/platformio/packages/tool-ninja/ninja cmTC_f6b06 && [1/2] Building C object CMakeFiles/cmTC_f6b06.dir/testCCompiler.c.obj
    FAILED: CMakeFiles/cmTC_f6b06.dir/testCCompiler.c.obj 
    /config/.esphome/platformio/packages/toolchain-riscv32-esp/bin/riscv32-esp-elf-gcc   -march=rv32imc -o CMakeFiles/cmTC_f6b06.dir/testCCompiler.c.obj   -c testCCompiler.c
    /bin/sh: 1: /config/.esphome/platformio/packages/toolchain-riscv32-esp/bin/riscv32-esp-elf-gcc: not found
    ninja: build stopped: subcommand failed.
    
    

  

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  /config/.esphome/platformio/packages/framework-espidf/tools/cmake/project.cmake:311 (__project)
  CMakeLists.txt:3 (project)

========================== [FAILED] Took 9.65 seconds ==========================
```

### Additional information

This is my first time trying ESP Home, it's possible I misunderstood something. I am happy to answer any additional questions. The esphome instance is running in a privileged docker container on the same rpi4 as my HA instance, which is also dockerized.

SmarterHomeLife, 2022-03-31:
I'm getting the exact same error trying to compile on a RPi3 with the HA addon under HAOS.

Vanderbleek, 2022-04-06:
I have the same issue as above -- fails to compile on RPi3 with HA, trying to build a previously working file for the C3 devkit.

DanteBC, 2022-04-06:
Hey. It seems your issue is related to my. Check it out, there is a way to circumvent it there by stating an older framework version
https://github.com/esphome/issues/issues/3076

closed 2022-08-24 by github-actions[bot]

eingemaischt, 2022-09-02:
I do have the problem, too:

[compile.log](https://github.com/esphome/issues/files/9480674/compile.log)

martinschilliger, 2022-10-29:
Had the same problem, the same configuration that didn't work on the Raspberry Pi worked when I installed ESPHome on my mac. Maybe try on your desktop again?