esphome/issues #3165: 2022.3.0 breaks 3 hub one wire setup

szafran81, 2022-03-20:
### The problem

Hi,

I'm having problems with this config:

```
substitutions:
  devicename: czytnik-czujnikow-temperatury
  upper_devicename: "Czytnik czujnikow temperatury"
  upper_devicename_nospace: "Czytnik-czujnikow-temperatury"

esphome:
  name: ${devicename}

esp8266:
  board: esp01_1m

logger:
  level: DEBUG

# Enable Home Assistant API
api:
  reboot_timeout: 360s

ota:
  password: !secret esphome_otapass

wifi:
  ssid: !secret esphome_ssid
  password: !secret esphome_wifipass
  reboot_timeout: 5min
  fast_connect: true
  use_address: 192.168.1.218

  manual_ip:
    static_ip: 192.168.1.218
    gateway: 192.168.1.1
    subnet: 255.255.255.0
    dns1: 192.168.1.1

### PRG

dallas:
- id: "hub_1"
  pin: GPIO2
- id: "hub_2"
  pin: GPIO0
- id: "hub_3"
  pin: GPIO3

binary_sensor:
  - platform: status
    name: ${upper_devicename}
    id: sensor_status

text_sensor:
  - platform: template
    name: "${upper_devicename} - Uptime"
    id: uptime_human
    icon: mdi:clock-outline
  - platform: template
    name: "${upper_devicename} - Siła sygnału WiFi"
    id: wifi_percentage
    icon: mdi:wifi

sensor:
  - platform: uptime
    id: uptime_sensor
    update_interval: 5s
    internal: true
    on_raw_value:
      then:
        - text_sensor.template.publish:
            id: uptime_human
            state: !lambda |-
              int seconds = round(id(uptime_sensor).raw_state);
              int days = seconds / (24 * 3600);
              seconds = seconds % (24 * 3600);
              int hours = seconds / 3600;
              seconds = seconds % 3600;
              int minutes = seconds /  60;
              seconds = seconds % 60;
              return (
                (days ? String(days) + "d " : "") +
                (hours ? String(hours) + "h " : "") +
                (minutes ? String(minutes) + "m " : "") +
                (String(seconds) + "s")
              ).c_str();
  - platform: wifi_signal
    id: wifi_sensor
    update_interval: 5s
    internal: true
    on_raw_value:
      then:
        - text_sensor.template.publish:
            id: wifi_percentage
            state: !lambda |-
              int moc = round(id(wifi_sensor).raw_state);
              moc = 2 * (moc + 100);
              if (moc > 100) { moc = 100; };
              return ((moc ? String(moc) + "%" : "")).c_str();

#hub_1
  - platform: dallas
    address: 0xB7031571FBA6FF28
    name: "Balkon"
    id: ds18b20_temp_balkon
    dallas_id: hub_1
  - platform: dallas
    address: 0xC6000002E028D428
    name: "Serwer"
    id: ds18b20_temp_serwer
    dallas_id: hub_1
  - platform: dallas
    address: 0x1F000002E0429428
    name: "Duży pokój 1"
    id: ds18b20_temp_duzy_pokoj_1
    dallas_id: hub_1
  - platform: dallas
    address: 0xB1031572500BFF28
    name: "Duży pokój 2"
    id: ds18b20_temp_duzy_pokoj_2
    dallas_id: hub_1
  - platform: dallas
    address: 0x8C000002E0432728
    name: "Duży pokój TV"
    id: ds18b20_temp_duzy_pokoj_tv
    dallas_id: hub_1

#hub_2
  - platform: dallas
    address: 0xC8000002E00B8D28
    name: "UPSy"
    id: ds18b20_temp_upsy
    dallas_id: hub_2
  - platform: dallas
    address: 0x4B000002E0190428
    name: "Przedpokój 2"
    id: ds18b20_temp_przedpokoj_2
    dallas_id: hub_2
  - platform: dallas
    address: 0xB300000089C89D28
    name: "Kuchnia"
    id: ds18b20_temp_kuchnia
    dallas_id: hub_2
  - platform: dallas
    address: 0xEF000002E00E6928
    name: "Workspace"
    id: ds18b20_temp_workspace
    dallas_id: hub_2

#hub_3
  - platform: dallas
    address: 0x86000002E051C328
    name: "Klatka schodowa"
    id: ds18b20_temp_klatka_schodowa
    dallas_id: hub_3
  - platform: dallas
    address: 0x2F0000008949FE28
    name: "Przedpokój 1"
    id: ds18b20_temp_przedpokoj_1
    dallas_id: hub_3
  - platform: dallas
    address: 0xD403157248DFFF28
    name: "Mały pokój"
    id: ds18b20_temp_maly_pokoj
    dallas_id: hub_3
  - platform: dallas
    address: 0x5E031571FBE6FF28
    name: "Mały pokój TV"
    id: ds18b20_temp_maly_pokoj_tv
    dallas_id: hub_3

time:
  - platform: sntp
    id: sensor_time
    timezone: "Europe/Warsaw"
```

So... a basic simple 3 hub config with some temp sensors.
What was happening... The update to 2022.3.0 came and I did an update all on my esphome devices. Some time after that I've realized that some of my temperature sensors are not working properly. To be specific hub_1 from the config above was dead. Other 2 hubs worked without problems (that's why I thought it was a HW issue).
At first I thought that it was a HW problem and that one of the esp8266 GPIO pins died on me. But after exchanging it with 2 other ones I got the 2022.2.6 docker image installed instead of 2022.3.0 and after sending the recompiled bin everything started working again.

### Which version of ESPHome has the issue?

2022.3.0

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2022.3.5

### What platform are you using?

ESP8266

### Board

esp8266-01

### Component causing the issue

dallas sensor hub

nagyrobi, 2022-03-21:
Why do you need them in separate hubs? You could connect them all to the same GPIO. Some say even 100 sensors can be handled by the same bus, [I did tests](https://community.home-assistant.io/t/smart-heating-scheduler-for-home-assistant-extra-multi-zones-version/237966/111?u=robi) with 53 pieces of DS18B20 in parrallel, all of them were detected properly with ESPHome.

nagyrobi, 2022-03-21:
I have a few of them running on very long CAT5e cables (about 50m), all they needed was a higher pullup resistor, 10k did it.

nagyrobi, 2022-03-21:
Also see https://github.com/esphome/issues/issues/3153

szafran81, 2022-03-21:
Thank you for your answears.
Does it really matter why ? (not malicious, but a real question)
The software officially allows it.
It's not hacky in any way.
It worked with previus version of esphome.
And to be completelly honest this is how they were connected before when I was using owserver with 3 x OW-2-USB converters (don't remember what chips they had).
I didn't even consider if connecting all 3 cables to a single pin would work after I saw that the component used hubs.

nagyrobi, 2022-03-21:
I was just curious. Using multiple hubs will liekly increase memory consumption so unless there's no real reason to use them (exceeded sensor count limints, cable lengths etc) I wouldn't bother with that.

I agree it's not a proper 'fix' to the issue but might help you in restoring functionality until the root cause of the problem will be solved...

weinic, 2022-03-23:
I have the same problem.

Here are 2 hubs and that now also does not work since esphome version 2022.3.0

esphome:
  name: heizungstemperaturen

esp8266:
  board: esp_wroom_02

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:
  password: "20f98889dae952587376ddec8c21d6e9"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  fast_connect: on
#  manual_ip:
#    static_ip: 192.168.193.59
#    gateway: 192.168.193.1
#    subnet: 255.255.255.0
  

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Heizungstemperaturen"
    password: "VkM9P2TSvHm7"

captive_portal:

dallas:
- id: "dallas_hub_1"
  pin: GPIO12
  update_interval: 10s
- id: "dallas_hub_2"
  pin: GPIO14
  update_interval: 10s

sensor:
- platform: dallas
  address: 0xaf041700ab3aff28
  name: "Fußbodenheizung"
  dallas_id: dallas_hub_1
  
- platform: dallas
  address: 0x05041700b5c6ff28
  name: "Heizkörper"
  dallas_id: dallas_hub_1  
  
  
- platform: dallas
  address: 0xf90517009a44ff28
  name: "Solaranlage"
  dallas_id: dallas_hub_2 
  
- platform: dallas
  address: 0x5a0417009a29ff28
  name: "WW-Speicher"
  dallas_id: dallas_hub_2

- platform: adc
  pin: VCC
  id: "power_supply_level"
  name: "Versorgungsspannung"
  update_interval: 5min
  
button:
  - platform: restart
    name: "Neustart"

   
    
#[22:34:54][C][dallas.sensor:071]:   Pin: GPIO12
#[22:34:54][C][dallas.sensor:072]:   Update Interval: 60.0s
#[22:34:54][D][dallas.sensor:077]:   Found sensors:
#[22:34:54][D][dallas.sensor:079]:     0xaf041700ab3aff28
#[22:34:54][D][dallas.sensor:079]:     0x05041700b5c6ff28
#[22:34:54][C][dallas.sensor:070]: DallasComponent:
#[22:34:54][C][dallas.sensor:071]:   Pin: GPIO14
#[22:34:54][C][dallas.sensor:072]:   Update Interval: 60.0s
#[22:34:54][D][dallas.sensor:077]:   Found sensors:
#[22:34:54][D][dallas.sensor:079]:     0xf90517009a44ff28
#[22:34:54][D][dallas.sensor:079]:     0x5a0417009a29ff28

jesserockz, 2022-03-24:
There is a pr in 2022.3.1 (which I forgot to release yesterday) that perhaps fixes?

weinic, 2022-03-24:
I have tested 2022.3.1.
No it does not work. The error still exists.

szafran81, 2022-03-24:
Same here:

```
INFO Successfully connected to 192.168.1.218
[08:39:48][I][app:102]: ESPHome version 2022.3.1 compiled on Mar 24 2022, 08:39:06
[08:39:48][C][wifi:491]: WiFi:
(...)
[08:39:48][C][dallas.sensor:075]: DallasComponent:
[08:39:48][C][dallas.sensor:076]:   Pin: GPIO1
[08:39:48][C][dallas.sensor:077]:   Update Interval: 60.0s
[08:39:48][W][dallas.sensor:080]:   Found no sensors!
```

powercolorsky, 2022-03-24:
> […](#)

Same story((( still no sensors

szafran81, 2022-03-30:
Just tested 2022.3.2 - no changes -> no sensors on hub1

powercolorsky, 2022-03-30:
> Just tested 2022.3.2 - no changes -> no sensors on hub1

I found dirty solution. I add old image to docker, recompile firmware with 2022.2.6  put old firmware and upgrade to 2022.3.1 without upgrading firmware.. and got my sensors works

szafran81, 2022-03-30:
Swapping bin's is what I use after each test to flash the last working image version.
And it's not a solution - it's a (hopefully) temporary workaround.

powercolorsky, 2022-03-30:
> Swapping bin's is what I use after each test to flash the last working image version. And it's not a solution - it's a (hopefully) temporary workaround.

sure yes.. it's just temporary ..  My ESP gives information from important house part, so I need this information...

szafran81, 2022-03-30:
Same here... that's why I play the bin swapping game... otherwise I wouldn't bother with it.

edenhaus, 2022-04-18:
I have 2 hubs configured and with `2022.3.2` only one hub is working. When I revert back to `2022.2.6` both hubs work normaly.

> I have a few of them running on very long CAT5e cables (about 50m), all they needed was a higher pullup resistor, 10k did it.

In the next days I will check if this makes a different on my system, as the cable on the non working hub a approx 20m long

edenhaus, 2022-04-19:
After analyzing the changes between [`2022.2.6` and `2022.3.2`](https://github.com/esphome/esphome/compare/2022.2.6...2022.3.2), I found out that the biggest change on the dallas component was the PR [#3181](https://github.com/esphome/esphome/pull/3181), which improves the timing.

I quickly checked out the code on tag [2022.3.2](https://github.com/esphome/esphome/tree/2022.3.2) and reverted the commit of the above PR. See https://github.com/edenhaus/esphome/tree/analyse-dallas
I also pushed a docker image from this branch, which is available under `edenhaus/esphome:analyse-dallas`. With this image all my dallas sensors work correctly with the two hubs.

@szafran81 Can you please test with my image and give me feedback? So we can be sure, that I have identified the correct PR

szafran81, 2022-04-19:
@edenhaus
I'm sorry... still 1 of 3 hubs is down with the analyse-dallas docker image (GPIO0 hub shows no sensors)
```
[15:58:28][I][app:102]: ESPHome version 2022.3.2 compiled on Apr 19 2022, 15:57:41
[15:58:28][C][wifi:491]: WiFi:
[15:58:28][C][wifi:353]:   Local MAC: 5C:CF:7F:C0:1C:A3
[15:58:28][C][wifi:354]:   SSID: [redacted]
[15:58:28][C][wifi:355]:   IP Address: 192.168.1.218
[15:58:28][C][wifi:356]:   BSSID: [redacted]
[15:58:28][C][wifi:358]:   Hostname: 's-czytnik-dallas'
[15:58:28][C][wifi:360]:   Signal strength: -43 dB ▂▄▆█
[15:58:28][C][wifi:364]:   Channel: 6
[15:58:28][C][wifi:365]:   Subnet: 255.255.255.0
[15:58:28][C][wifi:366]:   Gateway: 192.168.1.1
[15:58:28][C][wifi:367]:   DNS1: 192.168.1.1
[15:58:28][C][wifi:368]:   DNS2: 0.0.0.0
[15:58:28][C][logger:233]: Logger:
[15:58:28][C][logger:234]:   Level: DEBUG
[15:58:28][C][logger:235]:   Log Baud Rate: 115200
[15:58:28][C][logger:236]:   Hardware UART: UART0
[15:58:28][C][template.text_sensor:021]: Template Sensor 'S - Czytnik Dallas - Uptime'
[15:58:28][C][template.text_sensor:021]:   Icon: 'mdi:clock-outline'
[15:58:28][C][template.text_sensor:021]: Template Sensor 'S - Czytnik Dallas - Siła sygnału WiFi'
[15:58:28][C][template.text_sensor:021]:   Icon: 'mdi:wifi'
[15:58:28][C][uptime.sensor:031]: Uptime Sensor 'uptime_sensor'
[15:58:28][C][uptime.sensor:031]:   State Class: 'total_increasing'
[15:58:28][C][uptime.sensor:031]:   Unit of Measurement: 's'
[15:58:28][C][uptime.sensor:031]:   Accuracy Decimals: 0
[15:58:28][C][uptime.sensor:031]:   Icon: 'mdi:timer-outline'
[15:58:28][C][dallas.sensor:070]: DallasComponent:
[15:58:28][C][dallas.sensor:071]:   Pin: GPIO1
[15:58:28][C][dallas.sensor:072]:   Update Interval: 60.0s
[15:58:28][D][dallas.sensor:077]:   Found sensors:
[15:58:28][D][dallas.sensor:079]:     0x1f000002e0429428
[15:58:28][D][dallas.sensor:079]:     0xc6000002e028d428
[15:58:28][D][dallas.sensor:079]:     0x8c000002e0432728
[15:58:28][D][dallas.sensor:079]:     0xb7031571fba6ff28
[15:58:28][D][dallas.sensor:079]:     0xb1031572500bff28
[15:58:28][C][dallas.sensor:084]:   Device 'DS18B20 - Balkon'
[15:58:28][C][dallas.sensor:084]:     Device Class: 'temperature'
[15:58:28][C][dallas.sensor:084]:     State Class: 'measurement'
[15:58:28][C][dallas.sensor:084]:     Unit of Measurement: '°C'
[15:58:28][C][dallas.sensor:084]:     Accuracy Decimals: 1
[15:58:28][C][dallas.sensor:092]:     Address: 0xb7031571fba6ff28
[15:58:28][C][dallas.sensor:093]:     Resolution: 12
[15:58:28][C][dallas.sensor:084]:   Device 'DS18B20 - Serwer'
[15:58:28][C][dallas.sensor:084]:     Device Class: 'temperature'
[15:58:28][C][dallas.sensor:084]:     State Class: 'measurement'
[15:58:28][C][dallas.sensor:084]:     Unit of Measurement: '°C'
[15:58:28][C][dallas.sensor:084]:     Accuracy Decimals: 1
[15:58:28][C][dallas.sensor:092]:     Address: 0xc6000002e028d428
[15:58:28][C][dallas.sensor:093]:     Resolution: 12
[15:58:28][C][dallas.sensor:084]:   Device 'DS18B20 - Duży pokój 1'
[15:58:28][C][dallas.sensor:084]:     Device Class: 'temperature'
[15:58:28][C][dallas.sensor:084]:     State Class: 'measurement'
[15:58:28][C][dallas.sensor:084]:     Unit of Measurement: '°C'
[15:58:28][C][dallas.sensor:084]:     Accuracy Decimals: 1
[15:58:28][C][dallas.sensor:092]:     Address: 0x1f000002e0429428
[15:58:28][C][dallas.sensor:093]:     Resolution: 12
[15:58:28][C][dallas.sensor:084]:   Device 'DS18B20 - Duży pokój 2'
[15:58:28][C][dallas.sensor:084]:     Device Class: 'temperature'
[15:58:29][C][dallas.sensor:084]:     State Class: 'measurement'
[15:58:29][C][dallas.sensor:084]:     Unit of Measurement: '°C'
[15:58:29][C][dallas.sensor:084]:     Accuracy Decimals: 1
[15:58:29][C][dallas.sensor:092]:     Address: 0xb1031572500bff28
[15:58:29][C][dallas.sensor:093]:     Resolution: 12
[15:58:29][C][dallas.sensor:084]:   Device 'DS18B20 - Duży pokój TV'
[15:58:29][C][dallas.sensor:084]:     Device Class: 'temperature'
[15:58:29][C][dallas.sensor:084]:     State Class: 'measurement'
[15:58:29][C][dallas.sensor:084]:     Unit of Measurement: '°C'
[... 235 lines cut ...]
[16:02:46][C][dallas.sensor:084]:     Unit of Measurement: '°C'
[16:02:46][C][dallas.sensor:084]:     Accuracy Decimals: 1
[16:02:46][C][dallas.sensor:092]:     Address: 0x5e031571fbe6ff28
[16:02:46][C][dallas.sensor:093]:     Resolution: 12
[16:02:46][C][status:034]: Status Binary Sensor 'S - Czytnik Dallas'
[16:02:46][C][status:034]:   Device Class: 'connectivity'
[16:02:46][C][mdns:084]: mDNS:
[16:02:46][C][mdns:085]:   Hostname: s-czytnik-dallas
[16:02:46][C][ota:085]: Over-The-Air Updates:
[16:02:46][C][ota:086]:   Address: 192.168.1.218:8266
[16:02:46][C][ota:089]:   Using Password.
[16:02:46][C][api:138]: API Server:
[16:02:46][C][api:139]:   Address: 192.168.1.218:6053
[16:02:46][C][api:143]:   Using noise encryption: NO
[16:02:46][C][wifi_signal.sensor:009]: WiFi Signal 'wifi_sensor'
[16:02:46][C][wifi_signal.sensor:009]:   Device Class: 'signal_strength'
[16:02:46][C][wifi_signal.sensor:009]:   State Class: 'measurement'
[16:02:46][C][wifi_signal.sensor:009]:   Unit of Measurement: 'dBm'
[16:02:46][C][wifi_signal.sensor:009]:   Accuracy Decimals: 0
[16:02:46][C][sntp:050]: SNTP Time:
[16:02:46][C][sntp:051]:   Server 1: '0.pool.ntp.org'
[16:02:46][C][sntp:052]:   Server 2: '1.pool.ntp.org'
[16:02:46][C][sntp:053]:   Server 3: '2.pool.ntp.org'
[16:02:46][C][sntp:054]:   Timezone: 'CET-1CEST,M3.5.0,M10.5.0/3'
```

szafran81, 2022-04-21:
Just tested the 2022.4.0 version and the result is the same... 1 hub down, 2 up.

edenhaus, 2022-04-24:
I finally found a solution for my setup.

After connecting the oscilloscope, I found out, that pulling up the signal via the resistor is to slow to read correct the signal. As with PR [#3181](https://github.com/esphome/esphome/pull/3181) the timing was improved (waittime for each bit is decreased), my 4.7K resistor was not anymore sufficient for sensor approx. 20-30m away. Using a 3.3K (had only this one at home) resistor instead and the sensors work again.

> I have a few of them running on very long CAT5e cables (about 50m), all they needed was a higher pullup resistor, 10k did it.

Are you sure that you have increased it? I needed to slower it to get a good signal again.

@szafran81 Can you maybe have the same problem? Are the sensor on the hub, which is not working further away as the other ones?

nagyrobi, 2022-04-24:
Yes, It's 10k in my case. Done in the old Tasmota days... It's long time since, on that particular node this issue didn't appear on ESPHome.

szafran81, 2022-04-26:
@edenhaus thanks for the info. I've just finished converting my last ceiling fan and lights to esphome. Will try the lower resistor theory tomorrow.
And IMHO bigger resistor (eg. 1k -> 10k) equals lower signal strength - so shorter cables.

szafran81, 2022-04-29:
Sorry for the delay... I've tested this a bit today... I went with 3k9 resistors.
Tested with v2022.3.2 and v2022.4.0 and it seems to be working fine so far.

szafran81, 2022-05-03:
After a couple of days of testing I think that this can be closed with conclusion to use a lower resistance pull up resistor (with testing for each case, as the DS18B20 datasheet calls for 5k pull up resistor).

closed 2022-05-03 by szafran81