esphome/issues #3169: esp8266 (NodeMCU V2 Amica, Wemos D1 mini): Can't connect to ESPHome API // Ping timed out!

luxing0815, 2022-03-22:
### The problem

Since update to v2022.3.0 all my esp8266 have the problem that they are losing their connection to Home Assistant. With v2022.2.0 I didn't have any problems.

This is independent of their intended use and the connected components. For example, I use the esp8266 to control LED strips, as well as for garden irrigation, where they switch a 4-way relay only.

After restarting the system and rebooting the esp's, it takes only a few minutes until they mutually lose their connection and pings get lost on both sides.

I checked the history. The issue occurred immediately after the update. Until then there were only isolated connection problems. Maybe 2-5 connection issues per day. Now it's several hundred a day!

I have already restarted my wifi. And there are no abnormalities with other devices in my network. All my esp8266 have static ip and their wifi signal strength is -60db to -70db.

It might be the same issue as reported in #3154. However, the error for me is not related to any specific sensors.

### Which version of ESPHome has the issue?

2022.3.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.3.5

### What platform are you using?

ESP8266

### Board

NodeMCU V2 Amica, Wemos D1 mini

### Component causing the issue

api

### Example YAML snippet

```yaml
esphome:
  name: kitchen-rgbw-led-strip1
  platform: ESP8266
  board: nodemcuv2

# Enable logging
logger:
  level: VERBOSE

# Enable Home Assistant API
api:

ota:
  password: !secret ota_password

wifi:
  ssid: !secret wlan_ssid
  password: !secret wlan_password

  # Optional manual IP
  manual_ip:
    static_ip: 192.168.1.82
    gateway: 192.168.1.1
    subnet: 255.255.255.0

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Kitchen Led Strip 1 Hotspot"
    password: !secret ap_password
[... 558 lines cut ...]
2022-03-22 08:40:51 INFO (MainThread) [aioesphomeapi.connection] kitchen-rgbw-led-strip1 @ 192.168.1.82: Ping timed out!
2022-03-22 08:40:51 INFO (MainThread) [aioesphomeapi.reconnect_logic] Disconnected from ESPHome API for kitchen-rgbw-led-strip1 @ 192.168.1.82
2022-03-22 08:40:51 INFO (MainThread) [aioesphomeapi.reconnect_logic] Successfully connected to kitchen-rgbw-led-strip1 @ 192.168.1.82
2022-03-22 08:42:06 INFO (MainThread) [aioesphomeapi.connection] kitchen-rgbw-led-strip1 @ 192.168.1.82: Ping timed out!
2022-03-22 08:42:06 INFO (MainThread) [aioesphomeapi.reconnect_logic] Disconnected from ESPHome API for kitchen-rgbw-led-strip1 @ 192.168.1.82
2022-03-22 08:42:06 INFO (MainThread) [aioesphomeapi.reconnect_logic] Successfully connected to kitchen-rgbw-led-strip1 @ 192.168.1.82
2022-03-22 08:42:51 INFO (MainThread) [aioesphomeapi.connection] kitchen-rgbw-led-strip1 @ 192.168.1.82: Ping timed out!
2022-03-22 08:42:51 INFO (MainThread) [aioesphomeapi.reconnect_logic] Disconnected from ESPHome API for kitchen-rgbw-led-strip1 @ 192.168.1.82
2022-03-22 08:42:52 INFO (MainThread) [aioesphomeapi.reconnect_logic] Successfully connected to kitchen-rgbw-led-strip1 @ 192.168.1.82
2022-03-22 08:43:32 INFO (MainThread) [aioesphomeapi.connection] kitchen-rgbw-led-strip1 @ 192.168.1.82: Ping timed out!
2022-03-22 08:43:32 INFO (MainThread) [aioesphomeapi.reconnect_logic] Disconnected from ESPHome API for kitchen-rgbw-led-strip1 @ 192.168.1.82
2022-03-22 08:44:02 WARNING (MainThread) [aioesphomeapi.reconnect_logic] Can't connect to ESPHome API for kitchen-rgbw-led-strip1 @ 192.168.1.82: Timeout while connecting to ('192.168.1.82', 6053)
2022-03-22 08:44:02 INFO (MainThread) [aioesphomeapi.reconnect_logic] Trying to reconnect to kitchen-rgbw-led-strip1 @ 192.168.1.82 in the background
2022-03-22 08:44:11 INFO (MainThread) [aioesphomeapi.reconnect_logic] Successfully connected to kitchen-rgbw-led-strip1 @ 192.168.1.82
2022-03-22 08:44:36 INFO (MainThread) [aioesphomeapi.connection] kitchen-rgbw-led-strip1 @ 192.168.1.82: Ping timed out!
2022-03-22 08:44:36 INFO (MainThread) [aioesphomeapi.reconnect_logic] Disconnected from ESPHome API for kitchen-rgbw-led-strip1 @ 192.168.1.82
2022-03-22 08:44:36 INFO (MainThread) [aioesphomeapi.reconnect_logic] Successfully connected to kitchen-rgbw-led-strip1 @ 192.168.1.82
2022-03-22 08:46:00 INFO (MainThread) [aioesphomeapi.connection] kitchen-rgbw-led-strip1 @ 192.168.1.82: Ping timed out!
2022-03-22 08:46:00 INFO (MainThread) [aioesphomeapi.reconnect_logic] Disconnected from ESPHome API for kitchen-rgbw-led-strip1 @ 192.168.1.82
2022-03-22 08:46:00 INFO (MainThread) [aioesphomeapi.reconnect_logic] Successfully connected to kitchen-rgbw-led-strip1 @ 192.168.1.82
2022-03-22 08:46:30 INFO (MainThread) [aioesphomeapi.connection] kitchen-rgbw-led-strip1 @ 192.168.1.82: Error while reading incoming messages: Error while reading data: [Errno 104] Connection reset by peer
2022-03-22 08:46:30 INFO (MainThread) [aioesphomeapi.reconnect_logic] Disconnected from ESPHome API for kitchen-rgbw-led-strip1 @ 192.168.1.82
2022-03-22 08:46:30 INFO (MainThread) [aioesphomeapi.connection] kitchen-rgbw-led-strip1 @ 192.168.1.82: Ping Failed: Error while reading data: [Errno 104] Connection reset by peer
2022-03-22 08:46:30 INFO (MainThread) [aioesphomeapi.reconnect_logic] Successfully connected to kitchen-rgbw-led-strip1 @ 192.168.1.82
2022-03-22 08:47:28 INFO (MainThread) [aioesphomeapi.reconnect_logic] Successfully connected to kitchen-rgbw-led-strip1 @ 192.168.1.82

luxing0815, 2022-03-23:
Apparently this is a known issue. A workaround can be found in #1237, which in turn refers to #1532.
If have added **power_save_mode: high** to the wifi section for all of esp8266. This really makes this difference. I can confirm what was observed by @CarlosGS in #issuecomment-718669014.

luxing0815, 2022-03-24:
If have added a binary ping sensor for each of my esp8266 with a scan interval of 60s. Now, with **power_save_mode: high**, whithin 24 hrs I have counted roughly 50 disconnects but only one warning ("Can't connect to ESPHome API")  per device. That's quite an improvement.
Whereby I have update my system to Home Assistant 2022.3.6 in the middle of this time period. But there was no noticable difference. The history graph looks pretty much the same before and after.
Fortunately, the remaining disconnections were not noticeable, as all esp8266 responded quickly and reliably to all my automation triggers. At least for now. 
I will continue to monitor my system...

luxing0815, 2022-03-25:
Update to Home Assistant 2022.3.7 and esphome 2022.3.1 did not bring any negative surprises. After a restart, the system runs very smoothly. Only once the error "Can't connect to ESPHome API" within a day. 
So, the workaround (power_save_mode: high) has solved my problem, which is why close this ticket now.

luxing0815, 2022-03-28:
Reopen because, unfortunately, not solved.

Within the past 72 hours the system has reported again dozens of "Can't connect to ESPHome API" warnings. I have added now a sensor to record the wifi signal strengh. Here are my current measurements:

Drip irrigation system (signal strength: -80db +/- 2db)
8x Can't connect to ESPHome API ...: Error connecting to
19x Can't connect to ESPHome API ...: Timeout while connecting to

LED Strip 1 (signal strength: -66db +/- 2db)
3x Can't connect to ESPHome API ...: Error connecting to
0x Can't connect to ESPHome API ...: Timeout while connecting

LED Strip 2 (signal strength: -70db +/- 2db)
17x Can't connect to ESPHome API ...: Error connecting to
20x Can't connect to ESPHome API ...: Timeout while connecting to

CarlosGS, 2022-03-28:
Hi! Seeing `"Cannot send message because of TCP buffer space"` makes me wonder if it is an issue with low RAM.
Could you try commenting out `captive_portal:`? I think this would free up some memory.
Also maybe remove `level: VERBOSE` from the logger component, it sometimes can cause problems. Good luck!

luxing0815, 2022-03-28:
> Hi! Seeing `"Cannot send message because of TCP buffer space"` makes me wonder if it is an issue with low RAM. Could you try commenting out `captive_portal:`? I think this would free up some memory. [...]

Thank you for the hint. I have used `level: VERBOSE` only for my root cause analysis. And you are right that this probably has caused the `Cannot send message because of TCP buffer space` error.
For field use, however, I am using the default log level (`level: WARN`). The memory usage is 40% RAM and 43% Flash. Commenting out `captive_portal` reduces the of usage each by ~2,5%. 
The disconnect of the esp8266 continues to take place.

luxing0815, 2022-03-31:
I evaluated the Wifi sensors over a longer period of time and found that the signal strength of all three esp8266 fluctuates at the same time (up to +/- 5 db). Then I scanned the wifi frequencies in different places in the house and found six wifi routers on "my" channel with a similarly high signal strength. Fiber to the Home was recently rolled out here and now all of our neighbors probably have the same router running on the same channel. After I switched all esp8266 to another, more powerful wifi access point on a different channel, I don't have a single disconnect anymore. The esp8266 respond very quickly. That's how it should be. I hereby close this ticket.

closed 2022-03-31 by luxing0815