esphome/issues #3175: ESPHome Add-On: CLI and web compile into different folders

Jpsy, 2022-03-24:
### The problem

I use two ways to compile my ESPHome YAML nodes:
1. through the web frontend of the ESPHome add-on for HA and
1. through CLI

**The problem:**
If I switch between both methods the compile goes through all files again. It does not seem to find any existing compile results and repeats all the work which takes a lot of time. I think this is because both methods store their compiled files in different folders.

I tried to compare the output of both compiles and upload runs. Here are the important lines from both versions.
They differ mainly in the leading /data/smoke-eg-flur/ of the web version. But the final "INFO Uploading" line differs only in a leading /data/.

Hint: My node YAML contains `build_path: "smoke-eg-flur"` for these examples.

Output of install OTA through **HA add-on web frontend**:
```
INFO Reading configuration /config/esphome/smoke-eg-flur.yaml...
...
INFO Deleting /data/smoke-eg-flur/.pioenvs
...
Compiling /data/smoke-eg-flur/.pioenvs/smoke-eg-flur/src/esphome/components/adc/adc_sensor.cpp.o
Compiling /data/smoke-eg-flur/.pioenvs/smoke-eg-flur/src/esphome/components/api/api_connection.cpp.o
Compiling /data/smoke-eg-flur/.pioenvs/smoke-eg-flur/src/esphome/components/api/api_frame_helper.cpp.o
...
Linking /data/smoke-eg-flur/.pioenvs/smoke-eg-flur/firmware.elf
RAM:   [====      ]  43.2% (used 35420 bytes from 81920 bytes)
Flash: [====      ]  44.7% (used 466485 bytes from 1044464 bytes)
Building /data/smoke-eg-flur/.pioenvs/smoke-eg-flur/firmware.bin
...
esp8266_copy_factory_bin(["/data/smoke-eg-flur/.pioenvs/smoke-eg-flur/firmware.bin"], ["/data/smoke-eg-flur/.pioenvs/smoke-eg-flur/firmware.elf"])
...
INFO Uploading /data/smoke-eg-flur/.pioenvs/smoke-eg-flur/firmware.bin (470640 bytes)
```

Output of compile and OTA upload through **CLI**:
```
INFO Reading configuration smoke-eg-flur.yaml...
...
INFO Deleting smoke-eg-flur/.pioenvs
...
Compiling .pioenvs/smoke-eg-flur/src/esphome/components/adc/adc_sensor.cpp.o
Compiling .pioenvs/smoke-eg-flur/src/esphome/components/api/api_connection.cpp.o
Compiling .pioenvs/smoke-eg-flur/src/esphome/components/api/api_frame_helper.cpp.o
...
Linking .pioenvs/smoke-eg-flur/firmware.elf
RAM:   [====      ]  43.2% (used 35420 bytes from 81920 bytes)
Flash: [====      ]  44.7% (used 466485 bytes from 1044464 bytes)
Building .pioenvs/smoke-eg-flur/firmware.bin
...
esp8266_copy_factory_bin([".pioenvs/smoke-eg-flur/firmware.bin"], [".pioenvs/smoke-eg-flur/firmware.elf"])
...
INFO Uploading smoke-eg-flur/.pioenvs/smoke-eg-flur/firmware.bin (470640 bytes)
```

### Which version of ESPHome has the issue?

2022.3.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.3.7

### What platform are you using?

ESP8266

### Board

Wemos D1-mini

### Example YAML snippet

```yaml
# Doesn't depend on concrete YAML content.
# Might be relevant that I use:
build_path: "smoke-eg-flur"
```

jesserockz, 2022-03-24:
What do you mean by CLI. On what machine?
The add-on stores the build files in the /data docker mount that is unique to that add-on.

Jpsy, 2022-03-24:
I use the CLI of the ESPHome add-on through SSH:

```
ssh hassio@homeassistant
docker docker exec -it addon_a0d7b954_esphome bash
cd config/esphome
esphome compile smoke-eg-flur.yaml
esphome upload smoke-eg-flur.yaml --device smoke-eg-flur
```

The CLI allows me to batch-compile groups of YAML nodes in a loop. I can select them through bash wildcard notation. This is not possible through the browser frontend.

jesserockz, 2022-03-24:
Ok, if you want to do that, `export ESPHOME_IS_HA_ADDON=true` first which tells it to use the same build folder in the `/data` directory.

Jpsy, 2022-03-24:
Oh, great! Thanks.
Will try that asap and report back.

randybb, 2022-03-25:
@jesserockz the same problem is with docker. Would be good to have this set in docker by default. I am using a script for compilation/upload firmware to my prod devices

nagyrobi, 2022-04-09:
Wow, didn't know this was possible.
All CLI commands are available?

Jpsy, 2022-04-09:
> All CLI commands are available?
Yes. This is the docker container. Your are talking to ESPHome itself.

nagyrobi, 2022-04-09:
Should be added to the docs!!!

Jpsy, 2022-04-09:
I agree. You go ahead? ;)

nagyrobi, 2022-04-09:
I need to test first in my environment. But if you already have the know-how, you can do it better than me.

Jpsy, 2022-05-23:
Hi @jesserockz,
I finally had a chance to test the `ESPHOME_IS_HA_ADDON ` setting that you mentioned above.
Setting this env var to true actually does the job: The CLI compile uses the same folder structure as the web compile.

But the two compiles still don't recognize each other. Doing a full compile through CLI does not shorten the compile through web. And vice versa. After I compiled one way, the other way will always do a full compile again.

To find the reason I have compared the log output of CLI vs. web compile.
The results are about 80% identical, but they differ in a few lib paths. It looks like there are different library repos in use with different latest versions of some dependencies. Here is a screenshot of a typical diff between a CLI log and a web log:

[image]

Do you have any idea what is causing this effect and how I can circumvent it?

Jpsy, 2022-05-23:
@nagyrobi:
FYI starting with version 2022.2.0 ESPHome has moved from the Community repo to the HA repo. See: [https://esphome.io/changelog/2022.2.0.html](https://esphome.io/changelog/2022.2.0.html). You have to switch to the new repo manually in HA. If you do so, this will also change the docker container ID of ESPHome. Here is the command list from above, updated to the new docker ID:
```
ssh hassio@homeassistant
docker docker exec -it addon_5c53de3b_esphome bash
cd config/esphome
esphome compile smoke-eg-flur.yaml
esphome upload smoke-eg-flur.yaml --device smoke-eg-flur
```

And here is a short bash script that will compile and upload all your ESPHome nodes that match a given name pattern. You can run it after you have CDed into `config/esphome`:
```
for config in smoke-*.yaml; do echo "----- COMPILE ----- $config -----"; if esphome compile $config; then echo "----- UPLOAD ----- $config -----"; esphome upload $config --device ${config%.*}.local; fi done
```

chatziko, 2022-08-31:
After lots of tries I finally managed to make a command line compilation that is identical to the one performed by the dashboard, and reuses any previously compiled files. For this we need:
- to set a few exports (not just `ESPHOME_IS_HA_ADDON`)
- run `esphome compile` with a full path to `/config/esphome/foo.yaml` (not run it from within `/config/esphome`).
  Otherwise [this check](https://github.com/esphome/esphome/blob/2650441013a587911d603e3e715e63ae7b808484/esphome/writer.py#L107-L108) will fail, causing the build files to be deleted.

The following works in my setup:
```
docker docker exec -it addon_5c53de3b_esphome bash

export ESPHOME_IS_HA_ADDON=true
pio_cache_base=/data/cache/platformio
export PLATFORMIO_PLATFORMS_DIR="${pio_cache_base}/platforms"
export PLATFORMIO_PACKAGES_DIR="${pio_cache_base}/packages"
export PLATFORMIO_CACHE_DIR="${pio_cache_base}/cache"
export PLATFORMIO_GLOBALLIB_DIR=/piolibs

esphome compile /config/esphome/my-node.yaml
```