esphome/issues #3176: This external component is broken in after 2022.03

vampywiz17, 2022-03-24:
### The problem

This external component is broken after 2022.03

https://github.com/vampywiz17/esphome-linp-doorbell-g03

### Which version of ESPHome has the issue?

2022.3.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.3.7

### What platform are you using?

ESP32

### Board

ESP32

### Component causing the issue

https://github.com/vampywiz17/esphome-linp-doorbell-g03

### Example YAML snippet

```yaml
esphome:
  name: csengo
  
  platform: ESP32
  board: esp32doit-devkit-v1
  platformio_options:
    platform: espressif32@3.2.0
    platform_packages:
       - framework-arduinoespressif32 @ https://github.com/pauln/arduino-esp32.git#solo-no-mac-crc/1.0.6
       
external_components:
  - source: github://vampywiz17/esphome-linp-doorbell-g03@feature/external_components
    components: [ linp_doorbell ]

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Csengo Fallback Hotspot"
    password: "bOBVza3z6Ial"

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: !secret api_encryption_key
  password: !secret api_password

# Enable over-the-air updates
ota:
  password: "acf468d37d254f5c3f95b74692a9ff7f"

sensor:
  - platform: linp_doorbell
    volume:
      name: "Doorbell volume"
    chime_playing:
      name: "Doorbell playing tune"
      id: doorbell_button
#    use_old_service_names: True

binary_sensor:
  - platform: template
    name: "Doorbell state"
    lambda: |-
      if (id(doorbell_button).state > 0) {
        return true;
      } else {
        return false;
      }
      
captive_portal:
```

### Anything in the logs that might be useful for us?

```txt
INFO Reading configuration /config/esphome/csengo.yaml...
ERROR Unable to load component linp_doorbell.sensor:
Traceback (most recent call last):
  File "/esphome/esphome/loader.py", line 162, in _lookup_module
    module = importlib.import_module(f"esphome.components.{domain}")
  File "/usr/lib/python3.9/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 790, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/config/esphome/.esphome/external_components/23fd15ff/components/linp_doorbell/sensor.py", line 20, in <module>
    cv.Optional(CONF_VOLUME): sensor.sensor_schema(
TypeError: sensor_schema() takes from 0 to 1 positional arguments but 3 were given
Failed config

sensor.linp_doorbell: [source /config/esphome/csengo.yaml:38]
  
  Platform not found: 'sensor.linp_doorbell'.
  platform: linp_doorbell
  volume: 
    name: Doorbell volume
  chime_playing: 
    name: Doorbell playing tune
    id: doorbell_button
```

### Additional information

It think this file need to change it:

https://github.com/vampywiz17/esphome-linp-doorbell-g03/blob/feature/external_components/components/linp_doorbell/sensor.py

But i dont know, how... 

and this problem connec tthis, i think:

https://github.com/esphome/esphome/pull/3216/commits/6e882c086d0139946f9c87117a12cdf453436480

jesserockz, 2022-03-24:
The arguments to sensor.sensor_schema need to be kwargs (keyword arguments) which means they need to have (key1=value1), not just (value1, value2)

See the linked PR for changes to internal components

vampywiz17, 2022-04-25:
It fixed.

closed 2022-04-25 by vampywiz17