esphome/issues #3177: ESP8266 Spamming "Increasing memory allocation"

rniemand, 2022-03-25:
### The problem

I updated all my ESP8266 devices this morning using "v2022.3.1" of ESPHome after seeing the following update: https://esphome.io/changelog/2022.3.0.html

3 of my devices are now spamming the following message over and over in the logs and as a result I cannot preform OTA updates to revert them back to an older build.

`[19:56:24][W][json:079]: Increasing memory allocation.
[19:56:24][W][json:079]: Increasing memory allocation.
[19:56:24][W][json:079]: Increasing memory allocation.
[19:56:24][W][json:079]: Increasing memory allocation.
[19:56:24][W][json:079]: Increasing memory allocation.
[19:56:24][W][json:079]: Increasing memory allocation.`

Is this a known issue, or something I may have caused?

### Which version of ESPHome has the issue?

2022.3.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.3.7

### What platform are you using?

ESP8266

### Board

Wemos D1 Mini

### Component causing the issue

JSON

### Example YAML snippet

```yaml
substitutions:
  device_name: "boys-esp8266"
  display_name: "Boys Room"
  
esphome:
  name: $device_name
  
# Temporary fix (https://esphome.io/components/light/fastled.html)
esp8266:
  board: d1_mini
  framework:
    version: 2.7.4

logger:

mqtt:
  broker: !secret mqtt_broker
  port: !secret mqtt_port
  username: !secret node_boysesp_mqtt_user
  password: !secret node_boysesp_mqtt_pass
  
ota:
  password: "..."

wifi:
  ssid: !secret wifi_ssid_24
  password: !secret wifi_pass_24
  
  manual_ip:
    static_ip: !secret node_boysesp_ip
    gateway: !secret wifi_gateway
    subnet: !secret wifi_subnet
    dns1: !secret wifi_dns1
    dns2: !secret wifi_dns2

  ap:
    ssid: "${display_name} Hotspot"
    password: !secret ap_password

captive_portal:

binary_sensor:
  - platform: gpio
    pin: D3
    name: "${display_name} Door"
    device_class: door

light:
  - platform: fastled_clockless
    chipset: WS2812B
    pin: D5
    id: 'boys_led'
    num_leds: 1
    rgb_order: GRB
    name: "${display_name} LED"

sensor:
  - platform: dht
    pin: D4
    temperature:
      name: "${display_name} Temperature"
      id: temperature_boys_room
    humidity:
      name: "${display_name} Humidity"
      id: humidity_boys_room
    update_interval: !secret collection_interval_dht
    
  - platform: adc
    pin: A0
    name: "${display_name} Brightness"
    id: boys_room_brightness
    update_interval: !secret collection_interval_ldr
    filters:
      - multiply: 3.3
      
  - platform: wifi_signal
    name: "${display_name} WiFi"
    update_interval: !secret collection_interval_wifi_signal
```

### Anything in the logs that might be useful for us?

```txt
[19:56:24][W][json:079]: Increasing memory allocation.
[19:56:24][W][json:079]: Increasing memory allocation.
```

### Additional information

I have included the entirety of my configuration file - this is about the same for the other 2 nodes presenting the same issue.

KrX3D, 2022-03-25:
Hello,
i got the same problem with 4 or 5 devices of mine.

but for me it works fine till i change a color or power off leds and than it starts spamming the "Increasing memory allocation." message and only powering off the device and on again helps

palengh, 2022-03-25:
Getting something similar on my _esp32dev_ board too.  I´m running ESPHome version 2022.3.1, by the way.

```
18:41:27 | [D] | [sensor:125] | 'Stue temperatur': Sending state 20.02000 °C with 2 decimals of accuracy
18:41:27 | [D] | [sensor:125] | 'Stue luftfuktighet': Sending state 44.79000 % with 2 decimals of accuracy
18:41:27 | [D] | [sensor:125] | 'Stue batterinivå': Sending state 83.00000 % with 0 decimals of accuracy
18:41:27 | [E] | [json:034] | Could not allocate memory for JSON document! Requested 5120 bytes, largest free heap block: 38872 bytes
18:41:27 | [D] | [sensor:125] | 'Stue batterispenning': Sending state 2.95400 V with 3 decimals of accuracy
18:41:27 | [E] | [json:034] | Could not allocate memory for JSON document! Requested 5120 bytes, largest free heap block: 38872 bytes
```

SH-Solutions, 2022-03-25:
Same problem here. All my ESP8266 devices loaded with ESPHome are stopped working.
As soon as the first request like toggle switch or scene has been sent to the device it starts spamming these error message.

In this state the device is no longer responding and also the action hasn't been executed.

user897943, 2022-03-26:
Is it just me or does it seem like the QC on this project is going down? I had to remove API from my configs (12 of them) and replaced with MQTT with discovery, because some SONOFFS were not communicating anymore. From HomeAssistant it kept telling me to add the api: on my esphome config but it was already there, and yes the password was correct.

And now, it's broken again after the latest update.

Serious pain in the ass to say the least, I'm really tired of having to open my appliances, remove the boards, solder the serial to USB, and flash manually to go back to a working state because code is being pushed with what appears to be lack of proper testing. This is not a criticism of anyone personally per se, but it's definitely criticism about the quality of the builds lately, it really needs to get better. I know, it's free, someone's giving their time, but I'd rather they didn't give their time to push broken code and break things every week. It's also not feasible to keep using the same old version forever because it is "stable", I find that excuse hilarious, if that's the solution then simply close the project on github and people can use the old "stable" forever, so I implore, please more QC before pushing out updates, thanks.

closed 2022-03-28 by jesserockz

theseanstewart, 2022-03-30:
@rniemand I had the same issue and didn't want to wait for esphome/esphome#3331 in 2022.3.2 so I added this to my config and it worked as expected:

```
external_components:
  - source: github://pr#3331
    components: ["json"]
```

SH-Solutions, 2022-03-30:
> @rniemand I had the same issue and didn't want to wait for [esphome/esphome#3331](https://github.com/esphome/esphome/pull/3331) in 2022.3.2 so I added this to my config and it worked as expected:
> 
> ```
> external_components:
>   - source: github://pr#3331
>     components: ["json"]
> ```

Good to know that this is possible. Will keep this in mind for next time.
But just for your info: It seems that the new version '2022.3.2' has been already released.

I received and installed it today morning and all my devices are working now again.

rniemand, 2022-03-30:
> @rniemand I had the same issue and didn't want to wait for [esphome/esphome#3331](https://github.com/esphome/esphome/pull/3331) in 2022.3.2 so I added this to my config and it worked as expected:
> 
> ```
> external_components:
>   - source: github://pr#3331
>     components: ["json"]
> ```

Thanks for the suggestion - I can confirm that adding in the following lines stops the spamming...

```
# https://github.com/esphome/issues/issues/3177
external_components:
  - source: github://pr#3331
    components: ["json"]
```

... however it seems that they are getting ignored.

I sent a command to turn on some `WS2812B` diodes, the intent is logged to the console, however they do not turn on.

[image]

There has been no other changes made to the boards `yaml` file (besides the above) and the hardware has not changed at all. I am a bit stumped - code for the LEDS:

```
light:
  - platform: fastled_clockless
    chipset: WS2812B
    pin: D4
    num_leds: 48
    rgb_order: GRB
    name: "${display_name}"
...
```
[image]

rniemand, 2022-03-30:
> Thanks for the suggestion - I can confirm that adding in the following lines stops the spamming...
>
> ```
> # https://github.com/esphome/issues/issues/3177
> external_components:
>   - source: [...]

So... I messed up.

Issue is fixed in `2022.3.2` - just make sure that you are NOT using the default HomeAssistant addon - you need to use the `https://github.com/esphome/home-assistant-addon` repository

segator, 2022-04-03:
I have shelly 2.5 upgraded to 2022.3.2 and the issue still appearing :(

andriej, 2022-04-20:
@jesserockz this shouldn't be closed yet, as still valid for some users?

mcmx, 2022-05-04:
I had the same issue, the device became unresponsive.

I had the old ESPHome addon, too bad I didn't know about that being deprecated, installed the new ESPHome addon, rebooted the device, and proceeded to install wirelessly before sending any command to it. Now it's working.

philsson, 2022-05-07:
> I had the same issue, the device became unresponsive.
> 
> I had the old ESPHome addon, too bad I didn't know about that being deprecated, installed the new ESPHome addon, rebooted the device, and proceeded to install wirelessly before sending any command to it. Now it's working.

Wow I had not realized this either. I was waiting and waiting for Home Assistant to suggest an upgrade. Now I see its even on version 2022.4. Lets hope this problem is gone now.

andriej, 2022-05-08:
Seems its not

philsson, 2022-05-09:
> Wow I had not realized this either. I was waiting and waiting for Home Assistant to suggest an upgrade. Now I see its even on version 2022.4. Lets hope this problem is gone now.

Tried and confirmed working for me on at least one device. Nice!

lucacri, 2022-06-15:
I still have the problem, I tried on 2022.5.1 and 2022.6-beta and it's happening all the time to all my ESP8266. No idea how to fix it so far!

mmakaay, 2022-06-15:
Can you share a minimal configuration that triggers the behavior for you?

Also, are you sure that the flashing process is working correctly? Since the log spamming could interfere with the flashing process. Doing a flash over serial would be the best way to rule out OTA flashing issues here.

lucacri, 2022-06-25:
Sorry for the late reply, it was definitely an incomplete flashing. I had to take each one of them out from their spots and flash them via USB.

All good now! Thank you for the help!