esphome/issues #3178: ESP32-C3 compile error with esp32_touch

arthurrump, 2022-03-25:
### The problem

I'm trying to use an ESP32-C3 with the esp32_touch integration, but compilation of the project fails with this error:

```
In file included from src/esphome/components/esp32_touch/esp32_touch.cpp:3:
src/esphome/components/esp32_touch/esp32_touch.h:10:10: fatal error: driver/touch_sensor.h: No such file or directory
 #include <driver/touch_sensor.h>
          ^~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
```

### Which version of ESPHome has the issue?

2022.3.1

### What type of installation are you using?

pip

### What platform are you using?

ESP32-IDF

### Board

esp32-c3-devkitm-1

### Component causing the issue

esp32_touch

### Example YAML snippet

```yaml
esphome:
  name: test

esp32:
  board: esp32-c3-devkitm-1
  variant: esp32c3
  framework:
    type: esp-idf

esp32_touch:
  setup_mode: true
  
binary_sensor:
  - platform: esp32_touch
    name: "Touch"
    pin: GPIO0
    threshold: 5
```

### Anything in the logs that might be useful for us?

```txt
$ esphome compile ./test.yaml
INFO Reading configuration ./test.yaml...
INFO Generating C++ source...
INFO Core config or version changed, cleaning build files...
INFO Compiling app...
************************************************************************************************************************
If you like PlatformIO, please:
- follow us on Twitter to stay up-to-date on the latest project news > https://twitter.com/PlatformIO_Org
- star it on GitHub > https://github.com/platformio/platformio
- try PlatformIO IDE for embedded development > https://platformio.org/platformio-ide
************************************************************************************************************************

Processing test (board: esp32-c3-devkitm-1; framework: espidf; platform: platformio/espressif32 @ 3.5.0)
------------------------------------------------------------------------------------------------------------------------
Platform Manager: Installing platformio/espressif32 @ 3.5.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Platform Manager: espressif32 @ 3.5.0 has been installed!
The platform 'platformio/espressif32 @ 3.5.0' has been successfully installed!
The rest of the packages will be installed later depending on your build environment.
Tool Manager: Installing espressif/toolchain-xtensa-esp32s2 @ 8.4.0+2021r2-patch2
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Tool Manager: toolchain-xtensa-esp32s2 @ 8.4.0+2021r2-patch2 has been installed!
Tool Manager: Installing espressif/toolchain-riscv32-esp @ 8.4.0+2021r2-patch2
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Tool Manager: toolchain-riscv32-esp @ 8.4.0+2021r2-patch2 has been installed!
Tool Manager: Installing platformio/framework-espidf @ ~3.40302.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Tool Manager: framework-espidf @ 3.40302.0 has been installed!
Tool Manager: Installing platformio/tool-esptoolpy @ ~1.30100.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Tool Manager: tool-esptoolpy @ 1.30100.210531 has been installed!
Tool Manager: Installing platformio/tool-cmake @ ~3.16.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Tool Manager: tool-cmake @ 3.16.4 has been installed!
Tool Manager: Installing platformio/tool-ninja @ ^1.7.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Tool Manager: tool-ninja @ 1.7.1 has been installed!
Tool Manager: Installing platformio/tool-scons @ ~4.40300.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Tool Manager: tool-scons @ 4.40300.1 has been installed!
HARDWARE: ESP32C3 160MHz, 320KB RAM, 4MB Flash
 - framework-espidf 3.40302.0 (4.3.2)
 - tool-cmake 3.16.4
 - tool-ninja 1.7.1
 - toolchain-riscv32-esp 8.4.0+2021r2-patch2
 - toolchain-xtensa-esp32s2 8.4.0+2021r2-patch2
Installing ESP-IDF's Python dependencies
Collecting future>=0.15.2
  Downloading future-0.18.2.tar.gz (829 kB)
Collecting pyparsing<2.4.0,>=2.0.3
  Downloading pyparsing-2.3.1-py2.py3-none-any.whl (61 kB)
Building wheels for collected packages: future
  Building wheel for future (setup.py): started
  Building wheel for future (setup.py): finished with status 'done'
  Created wheel for future: filename=future-0.18.2-py3-none-any.whl size=491058 sha256=ce7e2957e972c432aa49d899b778665ca59b066322c96b7e0cbcd6b96cf490ec
  Stored in directory: /home/arthur/.cache/pip/wheels/8e/70/28/3d6ccd6e315f65f245da085482a2e1c7d14b90b30f239e2cf4
Successfully built future
Installing collected packages: future, pyparsing
  Attempting uninstall: pyparsing
    Found existing installation: pyparsing 3.0.7
    Uninstalling pyparsing-3.0.7:
      Successfully uninstalled pyparsing-3.0.7
Successfully installed future-0.18.2 pyparsing-2.3.1
Reading CMake configuration...
Generating assembly for certificate bundle...
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
No dependencies
Compiling .pioenvs/test/src/esphome/components/api/api_connection.o
Compiling .pioenvs/test/src/esphome/components/api/api_frame_helper.o
Compiling .pioenvs/test/src/esphome/components/api/api_pb2.o
Compiling .pioenvs/test/src/esphome/components/api/api_pb2_service.o
Compiling .pioenvs/test/src/esphome/components/api/api_server.o
Compiling .pioenvs/test/src/esphome/components/api/list_entities.o
Compiling .pioenvs/test/src/esphome/components/api/proto.o
Compiling .pioenvs/test/src/esphome/components/api/subscribe_state.o
Compiling .pioenvs/test/src/esphome/components/api/user_services.o
Compiling .pioenvs/test/src/esphome/components/binary_sensor/automation.o
Compiling .pioenvs/test/src/esphome/components/binary_sensor/binary_sensor.o
Compiling .pioenvs/test/src/esphome/components/binary_sensor/filter.o
Compiling .pioenvs/test/src/esphome/components/esp32/core.o
Compiling .pioenvs/test/src/esphome/components/esp32/gpio_arduino.o
Compiling .pioenvs/test/src/esphome/components/esp32/gpio_idf.o
Compiling .pioenvs/test/src/esphome/components/esp32/preferences.o
Compiling .pioenvs/test/src/esphome/components/esp32_touch/esp32_touch.o
Compiling .pioenvs/test/src/esphome/components/logger/logger.o
In file included from src/esphome/components/esp32_touch/esp32_touch.cpp:3:
src/esphome/components/esp32_touch/esp32_touch.h:10:10: fatal error: driver/touch_sensor.h: No such file or directory
 #include <driver/touch_sensor.h>
          ^~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs/test/src/esphome/components/esp32_touch/esp32_touch.o] Error 1
============================================= [FAILED] Took 94.92 seconds =============================================
```

### Additional information

The board I have is labelled as a NodeMCU-Series ESP32-C3F-Kit 3252. It's not in the PlatformIO directories yet, so I picked the Espressif C3 devkit instead. I'm not sure this is relevant, as the compilation already fails before uploading to the board.

petergebruers, 2022-03-25:
ESP32 version "C3" does not have touch sensor according to this table:

https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/hw-reference/chip-series-comparison.html

arthurrump, 2022-03-25:
Ah, that makes sense then. Thanks!

closed 2022-03-25 by arthurrump