esphome/issues #3182: "WiFi Unknown connection status 0" in empty ESP32

anacruza, 2022-03-27:
### The problem

Recently I'm experiencing the same WiFi issue with 3 different ESP32 boards. The yaml file has only the base configuration, with no added components, just the plain suggested config.

I'm flashing using manual download, and just after the flashing is completed it tries to connect to the network, gets "Auth expired" and "Association Expired" messages to finally collapse repeating "WiFi Unknown connection status 0" ad eternum. 

I've checked the [#2742](https://github.com/esphome/issues/issues/2742) issue and even using the password without secrets it crashes. I also went trough [#455](https://github.com/esphome/issues/issues/455) issue and it seems it's not the problem either, because my config is almost blank.

Also I have a couple more ESP32 boards arround working in different automation projects with successful EspHome updates (i.e. sprinklers, alarm, etc.). 

I've ran out of new boards, so in the meanwhile any help would be preciated here.

### Which version of ESPHome has the issue?

2021.12.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

core-2022.3.7

### What platform are you using?

ESP32

### Component causing the issue

MAybe WiFi?

### Example YAML snippet

```yaml
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Prueba GPIO"
    password: !secret fallback_password
    
captive_portal:
logger:
api:
ota:

esphome:
  name: ${nombre}
  platform: ESP32
  board: esp-wrover-kit
  
substitutions:
  
  # Nombre del dispositivo en Home Assistant
  nombre: 'prueba-gpio'
```

### Anything in the logs that might be useful for us?

```txt
Using 'COM4' as serial port.
Writing at 0x0008c000... (100 %)Wrote 889008 bytes (509440 compressed) at 0x00010000 in 12.1 seconds (effective 588.2 kbit/s)...
Hash of data verified.

Leaving...
Hard Resetting...
Done! Flashing is complete!

Showing logs:
[21:11:45][I][logger:214]: Log initialized
[21:11:45][C][ota:458]: There have been 0 suspected unsuccessful boot attempts.
[21:11:45][D][esp32.preferences:114]: Saving preferences to flash...
[21:11:45][I][app:029]: Running through setup()...
[21:11:45][C][wifi:037]: Setting up WiFi...
[21:11:46][D][wifi:370]: Starting scan...
[21:11:47][D][wifi:385]: Found networks:
[21:11:47][I][wifi:429]: - 'VTR-6000486' (18:35:D1:B1:3F:D9) ▂▄▆█
[21:11:47][D][wifi:430]:     Channel: 11
[21:11:47][D][wifi:431]:     RSSI: -60 dB
[21:11:47][D][wifi:434]: - 'DIRECT-CBZEUSmsHo' (26:DB:C9:4B:53:5A) ▂▄▆█
[21:11:47][I][wifi:245]: WiFi Connecting to 'VTR-6000486'...
[21:11:49][W][wifi_esp32:495]: Event: Disconnected ssid='VTR-6000486' bssid=18:35:D1:B1:3F:D9 reason='Auth Expired'
[21:11:49][W][wifi:533]: Error while connecting to network.
[21:11:49][W][wifi:568]: Restarting WiFi adapter...
[21:11:49][I][wifi:245]: WiFi Connecting to 'VTR-6000486'...
[21:11:50][W][wifi_esp32:495]: Event: Disconnected ssid='VTR-6000486' bssid=18:35:D1:B1:3F:D9 reason='Association Expired'
[21:11:50][W][wifi:554]: WiFi Unknown connection status 0
[21:11:50][W][wifi:554]: WiFi Unknown connection status 0
[21:11:50][W][wifi:554]: WiFi Unknown connection status 0
[21:11:50][W][wifi:554]: WiFi Unknown connection status 0
[21:11:50][W][wifi:554]: WiFi Unknown connection status 0
[21:11:50][W][wifi:554]: WiFi Unknown connection status 0
[21:11:50][W][wifi:554]: WiFi Unknown connection status 0
[21:11:50][W][wifi:554]: WiFi Unknown connection status 0
[21:11:50][W][wifi:554]: WiFi Unknown connection status 0
[21:11:50][W][wifi:554]: WiFi Unknown connection status 0
[21:11:50][W][wifi:554]: WiFi Unknown connection status 0
[21:11:50][W][wifi:554]: WiFi Unknown connection status 0
[21:11:50][W][wifi:554]: WiFi Unknown connection status 0
[21:11:50][W][wifi:554]: WiFi Unknown connection status 0
[21:11:50][W][wifi:554]: WiFi Unknown connection status 0
[21:11:50][W][wifi:554]: WiFi Unknown connection status 0
[21:11:50][W][wifi:554]: WiFi Unknown connection status 0
[21:11:50][W][wifi:554]: WiFi Unknown connection status 0
[21:11:50][W][wifi:554]: WiFi Unknown connection status 0
[21:11:50][W][wifi:554]: WiFi Unknown connection status 0
[21:11:50][W][wifi:554]: WiFi Unknown connection status 0
[21:11:51][W][wifi:554]: WiFi Unknown connection status 0
```

donburch888, 2022-03-30:
Same here.  My first ESP32 project, simply connecting an ESP32-CAM. 

I am using an intel PC with Ubuntu 21.10
esphome version: 2022.3.2

Ran "esphome wizard ESP32-CAM.yaml" which resulted in:
```
esphome:
  name: esp32-cam

esp32:
  board: esp32cam
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: "api-password"

ota:
  password: "ota-password"

wifi:
  ssid: "erykah"
  password: "wifi-password"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esp32-Cam Fallback Hotspot"
    password: "RiYYkI0mWCkc"

captive_portal:
```

Running "esphome run ESP32-CAM.yaml", and the serial connection is filled with 
```
[W][wifi:557]: WiFi Unknown connection status 0
[W][wifi:557]: WiFi Unknown connection status 0
[W][wifi:557]: WiFi Unknown connection status 0
[W][wifi:557]: WiFi Unknown connection status 0
[W][wifi:557]: WiFi Unknown connection status 0
[W][wifi:557]: WiFi Unknown connection status 0
[W][wifi:557]: WiFi Unknown connection status 0
[W][wifi:557]: WiFi Unknown connection status 0
[W][wifi:557]: WiFi Unknown connection status 0
[W][wifi:557]: WiFi Unknown connection status 0
[W][wifi:557]: WiFi Unknown connection status 0
[W][wifi:557]: WiFi Unknown connection status 0
[W][wifi:557]: WiFi Unknown connection status 0
[W][wifi:530]: Timeout while connecting to WiFi.
[I][wifi:248]: WiFi Connecting to 'erykah'...
[W][wifi_esp32:495]: Event: Disconnected ssid='erykah' bssid=64:66:B3:ED:08:C4 reason='Association Expired'
[W][wifi:557]: WiFi Unknown connection status 0
[W][wifi:557]: WiFi Unknown connection status 0
[W][wifi:557]: WiFi Unknown connection status 0
[W][wifi:557]: WiFi Unknown connection status 0
[W][wifi:557]: WiFi Unknown connection status 0
[W][wifi:557]: WiFi Unknown connection status 0
[W][wifi:557]: WiFi Unknown connection status 0
[W][wifi:557]: WiFi Unknown connection status 0
[W][wifi:557]: WiFi Unknown connection status 0
[W][wifi:557]: WiFi Unknown connection status 0
```

SkateWarp, 2022-04-12:
I had this issue before adding `power_save_mode: none` in the WiFi component solved it for me.

ihavn, 2022-04-21:
I have tried adding `power_save_mode: none` but then `esp32_ble_tracker` is complaining.
Before updating to ESPHome to v2022.3.1 everything was working perfect. Now I get the "WiFi Unknown connection status 0" even on an empty ESP32 device.

donburch888, 2022-05-11:
Tried it again today ... esphome on my Ubuntu PC.   Same result.
Even installed Chrome browser so I can use the Home Assistant ESPHome integration to Install using "Plug into this computer". I believe this compiles on my RasPi 4 running HAOS, downloads to my desktop, but then uses https://web.esphome.io/?dashboard_install 

my yaml file is:
```
esphome:
  name: esp32-cam

esp32:
  board: esp32cam
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: ""

ota:
  password: ""

wifi:
  ssid: "erykah"
  password: "**redacted**"
  power_save_mode: none

  # Optional manual IP
  manual_ip:
    static_ip: 192.168.1.123
    gateway: 192.168.1.1
    subnet: 255.255.255.0

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esp32-Cam Fallback Hotspot"
    password: "RiYYkI0mWCkc"

captive_portal:

```

What else needs to be added to a minimal install to get Wi-Fi working ??

donburch888, 2022-05-11:
Looking back through the output from esphome wizard I find some messages which might be significant...
```
Library Manager: Installing Hash
Library Manager: Warning! Could not install dependency {'name': 'Hash', 'platforms': ['espressif8266']} for package 'ESPAsyncWebServer-esphome'
Library Manager: Installing ESP8266WiFi
Library Manager: Warning! Could not install dependency {'name': 'ESP8266WiFi', 'platforms': ['espressif8266']} for package 'ESPAsyncWebServer-esphome'
Dependency Graph
```
and later
```
Compiling .pioenvs/esp32-cam/FrameworkArduino/esp32-hal-touch.c.o
/home/don/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-spi.c: In function 'spiTransferBytesNL':
/home/don/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-spi.c:922:39: warning: initialization from incompatible pointer type [-Wincompatible-pointer-types]
                 uint8_t * last_out8 = &result[c_longs-1];
                                       ^
/home/don/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-spi.c:923:40: warning: initialization from incompatible pointer type [-Wincompatible-pointer-types]
                 uint8_t * last_data8 = &last_data;
                                        ^
Compiling .pioenvs/esp32-cam/FrameworkArduino/esp32-hal-uart.c.o
```

mmakaay, 2022-05-13:
I don't think those are really related to the issue.
I've got a device too now, that shows this behavior.
I have seen "Association Expired", but also this one:

```
[23:35:34][I][wifi:248]: WiFi Connecting to 'FOKKE'...
[23:35:37][W][wifi_esp32:495]: Event: Disconnected ssid='FOKKE' bssid=FC:EC:DA:8A:B6:16 reason='4-Way Handshake Timeout'
[23:35:37][W][wifi:557]: WiFi Unknown connection status 0
[23:35:37][W][wifi:557]: WiFi Unknown connection status 0
[23:35:37][W][wifi:557]: WiFi Unknown connection status 0
```

After a few flash operations, the device now connects. I don't think it's related to the flashing. Possibly the WiFi connection is a bit shabby currently (sitting downstairs with the device instead of upstairs, so the WiFi situation is different now).

donburch888, 2022-05-14:
> Possibly the WiFi connection is a bit shabby

That was it !  My phone works fine in my study, but it seems that the ESP01 and ESP32 aren't getting a good enough wi-fi signal. I have added a Wi-Fi extender and devices connect without any problem :-) 

Now to add to the YAML code to get them to do something ;-)

closed 2022-06-07 by jesserockz