esphome/issues #3185: Problem with Grow Fingerprint Reader and Wemos D1 Mini when setting a new password

flederelch, 2022-03-28:
### The problem

I am using a Wemos D1 Mini with a capacitive grow fingerprint reader (model R503). It was working fine using the different code snippets from the documentation.
To add more security to the setup I followed the documentation step by step for setting a new password. I copied the code from the documentation and used exact the same code for setting a new password. According to the log, the new password was successfully set. After updating the password value to the previously set password I always get a wrong password message. I also tried using the default password again.
In the meantime, I bought a second fingerprint reader and tried everything again. Unfortunately I have the same issue.

### Which version of ESPHome has the issue?

2022.3.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.3.7

### What platform are you using?

ESP8266

### Board

Wemos d1 Mini

### Component causing the issue

fingerprint_grow

### Example YAML snippet

```yaml
esphome:
  name: fingerprint1

esp8266:
  board: d1_mini

logger:

ota:
  password: ""

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  fast_connect: true
  manual_ip:
    static_ip: 192.168.1.202
    gateway: 192.168.1.1
    subnet: 255.255.255.0
    dns1: 192.168.1.2
    dns2: 192.168.1.1

captive_portal:

uart:
  rx_pin: D6
  tx_pin: D7
  baud_rate: 57600

fingerprint_grow:
  
  sensing_pin: D5
  
  password: 0x00000000
  new_password: 0x72AB96CD
  
  
  on_finger_scan_matched:
    - fingerprint_grow.aura_led_control:
        state: BREATHING
        speed: 255
        color: GREEN
        count: 1
    - text_sensor.template.publish:
        id: fingerprint_state
        state: !lambda 'return "Authorized finger " + to_string(finger_id) + ", confidence " + to_string(confidence);'
    # Pushing a tag_scanned event based on finger_id only if confidence is greater than 50
    - if:
        condition:
          lambda: 'return confidence > 50;'
        then:
          - homeassistant.tag_scanned: !lambda 'return "00-00-00-00-" + to_string(finger_id);'
    - delay: 10s
    - text_sensor.template.publish:
        id: fingerprint_state
        state: "Scan your finger" 
  on_finger_scan_unmatched:
    - fingerprint_grow.aura_led_control:
        state: FLASHING
        speed: 25
        color: RED
        count: 4
    - text_sensor.template.publish:
        id: fingerprint_state
        state: "Unauthorized finger"
    - delay: 5s
    - text_sensor.template.publish:
        id: fingerprint_state
        state: "Scan your finger" 
  on_enrollment_scan:
    - fingerprint_grow.aura_led_control:
        state: FLASHING
        speed: 25
        color: BLUE
        count: 2
    - text_sensor.template.publish:
        id: fingerprint_state
        state: !lambda 'return "Enrolling into slot " + to_string(finger_id) + ", scanned " + to_string(scan_num) + " time(s)";'
    - fingerprint_grow.aura_led_control:
        state: ALWAYS_ON
        speed: 0
        color: PURPLE
        count: 0
  on_enrollment_done:
    - fingerprint_grow.aura_led_control:
        state: BREATHING
        speed: 100
        color: BLUE
        count: 2
    - text_sensor.template.publish:
        id: fingerprint_state
        state: !lambda 'return "Enrolled into slot " + to_string(finger_id);'
  on_enrollment_failed:
    - fingerprint_grow.aura_led_control:
        state: FLASHING
        speed: 25
        color: RED
        count: 4
    - text_sensor.template.publish:
        id: fingerprint_state
        state: !lambda 'return "Failed to enroll into slot " + to_string(finger_id);'

api:
  
  password: !secret api_password
  encryption:
    key: !secret api_key
  
  services:
  - service: enroll
    variables:
      finger_id: int
      num_scans: int
    then:
      - fingerprint_grow.enroll:
          finger_id: !lambda 'return finger_id;'
          num_scans: !lambda 'return num_scans;'
      - text_sensor.template.publish:
          id: fingerprint_state
          state: "Place a finger on the reader"
      - fingerprint_grow.aura_led_control:
          state: ALWAYS_ON
          speed: 0
          color: PURPLE
          count: 0
  - service: cancel_enroll
    then:
      - fingerprint_grow.cancel_enroll:
  - service: delete
    variables:
      finger_id: int
    then:
      - fingerprint_grow.delete:
          finger_id: !lambda 'return finger_id;'
  - service: delete_all
    then:
      - fingerprint_grow.delete_all:

text_sensor:
  - platform: template
    name: "Fingerprint State"
    id: "fingerprint_state"

sensor:
  - platform: fingerprint_grow
    fingerprint_count:
      name: "Fingerprint Count"
    last_finger_id:
      name: "Fingerprint Last Finger ID"
    last_confidence:
      name: "Fingerprint Last Confidence"
    status:
      name: "Fingerprint Status"
    capacity:
      name: "Fingerprint Capacity"
    security_level:
      name: "Fingerprint Security Level"
```

### Anything in the logs that might be useful for us?

```txt
[21:48:04]{d[I][logger:214]: Log initialized
[21:48:04][C][ota:461]: There have been 0 suspected unsuccessful boot attempts.
[21:48:04][I][app:029]: Running through setup()...
[21:48:04][C][uart.arduino_esp8266:059]: Setting up UART bus...
[21:48:04][C][fingerprint_grow:052]: Setting up Grow Fingerprint Reader...
[21:48:04][D][fingerprint_grow:190]: Checking password
[21:48:04][D][fingerprint_grow:195]: Password verified
[21:48:04][D][fingerprint_grow:218]: Getting parameters
[21:48:04][D][fingerprint_grow:221]: Got parameters
[21:48:04][D][sensor:124]: 'Fingerprint Status': Sending state 4.00000  with 0 decimals of accuracy
[21:48:04][D][sensor:124]: 'Fingerprint Capacity': Sending state 200.00000  with 0 decimals of accuracy
[21:48:04][D][sensor:124]: 'Fingerprint Security Level': Sending state 3.00000  with 0 decimals of accuracy
[21:48:04][D][fingerprint_grow:242]: Getting fingerprint count
[21:48:04][D][fingerprint_grow:245]: Got fingerprint count
[21:48:04][D][sensor:124]: 'Fingerprint Count': Sending state 1.00000  with 0 decimals of accuracy
[21:48:04][C][wifi:037]: Setting up WiFi...
...
[21:56:19][I][logger:214]: Log initialized
[21:56:19][C][ota:461]: There have been 0 suspected unsuccessful boot attempts.
[21:56:19][I][app:029]: Running through setup()...
[21:56:19][C][uart.arduino_esp8266:059]: Setting up UART bus...
[21:56:19][C][fingerprint_grow:052]: Setting up Grow Fingerprint Reader...
[21:56:19][D][fingerprint_grow:190]: Checking password
[21:56:19][D][fingerprint_grow:195]: Password verified
[21:56:19][I][fingerprint_grow:205]: Setting new password: -806115653
[21:56:19][I][fingerprint_grow:209]: New password successfully set
[21:56:19][I][fingerprint_grow:210]: Define the new password in your configuration and reflash now
[21:56:19][W][fingerprint_grow:211]: !!!Forgetting the password will render your device unusable!!!
[21:56:19][C][wifi:037]: Setting up WiFi...
...
[21:58:31][I][logger:214]: Log initialized
[21:58:31][C][ota:461]: There have been 0 suspected unsuccessful boot attempts.
[21:58:31][I][app:029]: Running through setup()...
[21:58:31][C][uart.arduino_esp8266:059]: Setting up UART bus...
[21:58:31][C][fingerprint_grow:052]: Setting up Grow Fingerprint Reader...
[21:58:31][D][fingerprint_grow:190]: Checking password
[21:58:31][E][fingerprint_grow:198]: Wrong password
[21:58:31][E][component:112]: Component fingerprint_grow was marked as failed.
[21:58:31][C][wifi:037]: Setting up WiFi...
...
[22:06:39][I][logger:214]: Log initialized
[22:06:39][C][ota:461]: There have been 0 suspected unsuccessful boot attempts.
[22:06:39][I][app:029]: Running through setup()...
[22:06:39][C][uart.arduino_esp8266:059]: Setting up UART bus...
[22:06:39][C][fingerprint_grow:052]: Setting up Grow Fingerprint Reader...
[22:06:39][D][fingerprint_grow:190]: Checking password
[22:06:39][E][fingerprint_grow:198]: Wrong password
[22:06:39][E][component:112]: Component fingerprint_grow was marked as failed.
[22:06:39][C][wifi:037]: Setting up WiFi...
```

### Additional information

Full log file: [grow-fingerprint-log.txt](https://github.com/esphome/issues/files/8360275/grow-fingerprint-log.txt)

leranp, 2022-08-03:
same here, someone need to fix it, a lot of people will destroyed there device like me :-(

ssieb, 2022-08-17:
There's code at https://community.home-assistant.io/t/cannot-communicate-with-grow-r503-fingerprint-sensor-after-setting-password/432628/5 to find the current password.

closed 2022-08-17 by ssieb

nagyrobi, 2022-09-01:
Wonder how could we recover the bricked sensors now.

leranp, 2022-09-01:
> Wonder how could we recover the bricked sensors now.

You need to flash the device with different code that will brute force the sensor to find the password, look the that last comment

nagyrobi, 2022-09-01:
Ah great.