esphome/issues #3187: Could not allocate memory for JSON document!

sajikur, 2022-03-28:
### The problem

Hi, someone could help me for this error?

_[19:04:29][E][json:034]: Could not allocate memory for JSON document! Requested 5120 bytes, largest free heap block: 38872 bytes_
[19:04:29][D][sensor:125]: 'RSSI sensore 12': Sending state -48.00000 dB with 0 decimals of accuracy
_[19:04:29][E][json:034]: Could not allocate memory for JSON document! Requested 5120 bytes, largest free heap block: 38872 bytes_
[19:04:30][D][esp32_ble_tracker:723]: Found device 00:7C:2D:F4:D6:2F RSSI=-92
[19:04:30][D][esp32_ble_tracker:744]:   Address Type: PUBLIC
[19:04:31][D][xiaomi_ble:314]: Got Xiaomi HHCCJCY01 (C4:7C:8D:67:38:6B):
[19:04:31][D][xiaomi_ble:326]:   Conductivity: 1674µS/cm
[19:04:31][D][sensor:125]: 'Fertilizzazione sensore 5': Sending state 1674.00000 µS/cm with 0 decimals of accuracy
INFO 192.168.0.151: Ping timed out!
INFO Disconnected from ESPHome API for 192.168.0.151
WARNING Disconnected from API
INFO 192.168.0.151: Error while reading incoming messages: Error while reading data: 0 bytes read on a total of 1 expected bytes
WARNING Can't connect to ESPHome API for 192.168.0.151: Error while reading data: 0 bytes read on a total of 1 expected bytes
INFO Trying to reconnect to 192.168.0.151 in the background
INFO 192.168.0.151: Error while reading incoming messages: Error while reading data: [Errno 104] Connection reset by peer
INFO Successfully connected to 192.168.0.151
[19:05:38][D][sensor:125]: 'Segnale WIFI ESP piante': Sending state -81.00000 dBm with 0 decimals of accuracy
[19:06:38][D][sensor:125]: 'Segnale WIFI ESP piante': Sending state -81.00000 dBm with 0 decimals of accuracy
[19:07:38][D][sensor:125]: 'Segnale WIFI ESP piante': Sending state -80.00000 dBm with 0 decimals of accuracy
[19:08:38][D][sensor:125]: 'Segnale WIFI ESP piante': Sending state -82.00000 dBm with 0 decimals of accuracy
[19:08:49][I][ota:105]: Boot seems successful, resetting boot loop counter.

My esp32 mini D1 goes offline and reconnect many times in a day. I'm on the latest update, Wifi signal is strong but rssi goes between -50 to -87. Any ideas to solve? THX

### Which version of ESPHome has the issue?

Current version: 2022.3.1

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Board

mini d1 with esternal antenna

sajikur, 2022-03-28:
> Exactly the same problem on a Sonoff Dual R3:
>
> On Sonoff Basic I had to go back to 2022.2.5 due to a problem with JSON
> too. A lot of lines in log like these:
> [18:14:17][W][json:079]: Increasing memory allocation.
> [18:14:17][W][json:079]: Increasing memory allocation.
> [18:14:17][W][json:079]: Increasing memory allocation.
> [18:14:17][W][json:079]: Increasing memory allocation.
> [18:14:17][W][json:079]: Increasing memory allocation.
> [18:14:17][W][json:079]: Increasing memory allocation.
> [18:14:17][W][json:079]: Increasing memory allocation.
> [18:14:17][W][json:079]: Increasing memory allocation.
> [18:14:18][W][json:079]: Increasing memory allocation.
> [18:14:18][W][json:079]: Increasing memory allocation.
> [18:14:18][W][json:079]: Increasing memory allocation.
> [18:14:18][W][json:079]: Increasing memory allocation.
> [18:14:18][W][json:079]: Increasing memory allocation.
> [18:14:18][W][json:079]: Increasing memory allocation.
> [18:14:18][W][json:079]: Increasing memory allocation.
> [18:14:18][W][json:079]: Increasing memory allocation.
> [18:14:18][W][json:079]: Increasing memory allocation.
> This occurs when I switch the relay through Home Assistant.

I've the same issue on esphome esp32 d1 mini. Bug on last update?

mivano, 2022-03-28:
Same issue here with NSPanel, boots up and starts throwing below error. 

07][W][json:079]: Increasing memory allocation.

Will then reboot again and again. Started  couple of days ago with the v2022.3.1 release.

mivano, 2022-03-28:
I do see this PR https://github.com/esphome/esphome/pull/3289 as part of the latest release. I m not clever enough to see if it is indeed related but maybe @jesserock knows?

sajikur, 2022-03-29:
I don't know. Hope in a short time solution ;)

sajikur, 2022-03-29:
compiling with your workaround hide the log.
with only "logger: " I see a new warning

/data/cache/platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-spi.c: In function 'spiTransferBytesNL':
/data/cache/platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-spi.c:922:39: warning: initialization from incompatible pointer type [-Wincompatible-pointer-types]
                 uint8_t * last_out8 = &result[c_longs-1];
                                       ^
/data/cache/platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-spi.c:923:40: warning: initialization from incompatible pointer type [-Wincompatible-pointer-types]
                 uint8_t * last_data8 = &last_data;

sajikur, 2022-03-30:
last update, with change of the version, solve "could not allocate..."  problem. Remain last problem and this 

"pio system prune"

what is that and how can I solve?

THANX

mivano, 2022-03-30:
The 2022.3.2 fixed the issue for me (sonoff NSpanel). No longer crashing and normal log output. Thanks.

sanderlv, 2022-03-31:
Landed here due to I have this issue now on a  just update ESP32: 2022.3.2
```
<html><body>
<!--StartFragment-->

11:31:26 | [D] | [atc_mithermometer:121] | Battery Level: 50 %
-- | -- | -- | --
11:31:26 | [D] | [atc_mithermometer:124] | Battery Voltage: 2.679 V
11:31:26 | [D] | [sensor:125] | 'Temperature': Sending state 19.10000 °C with 1 decimals of accuracy
11:31:26 | [D] | [sensor:125] | 'Humidity': Sending state 51.00000 % with 0 decimals of accuracy
11:31:26 | [E] | [json:034] | Could not allocate memory for JSON document! Requested 5120 bytes, largest free heap block: 38388 bytes
11:31:26 | [D] | [sensor:125] | 'Battery-Level': Sending state 50.00000 % with 0 decimals of accuracy
11:31:26 | [E] | [json:034] | Could not allocate memory for JSON document! Requested 5120 bytes, largest free heap block: 38388 bytes
11:31:26 | [D] | [sensor:125] | 'Battery-Voltage': Sending state 2.67900 V with 3 decimals of accuracy
11:31:26 | [E] | [json:034] | Could not allocate memory for JSON document! Requested 5120 bytes, largest free heap block: 38388 bytes
11:31:27 | [D] | [esp32_ble_tracker:723] | Found device E7:74:2E:CC:0C:9F RSSI=-83
11:31:27 | [D] | [esp32_ble_tracker:744] | Address Type: RANDOM
11:31:32 | [I] | [ota:105] | Boot seems successful, resetting boot loop counter.
11:31:32 | [D] | [esp32.preferences:114] | Saving preferences to flash...
11:31:33 | [D] | [esp32_ble_tracker:217] | Starting scan...
11:31:33 | [D] | [binary_sensor:036] | 'Flower Bluetooth': Sending state OFF

<!--EndFragment-->
</body>
</html>
```

sajikur, 2022-03-31:
> Landed here due to I have this issue now on a just update

same here

sajikur, 2022-04-02:
it seems that last update doesn't solve. Reflashed boards but in the logs there are same errors
any ideas to solve?
Thanx

sajikur, 2022-04-07:
Hi, no ideas to solve?

statte, 2022-04-07:
One of the possible solutions in my case would be to disable the API component and use MQTT , but I've yet to test this(!).  I Don't know how relevant it would be in your case.

sajikur, 2022-04-07:
I don't usa MQTT and the error persists from the penultimate version and there is no way around it. I hope everything will work again with the next update

paulius2k, 2022-04-07:
I am getting the same error:
```
[E][json:034]: Could not allocate memory for JSON document! Requested 5120 bytes, largest free heap block: 37660 bytes
```

on this board:
```
esp32:
  board: nodemcu-32s
  framework:
    type: arduino
```

briodan, 2022-04-07:
same error with ESPHome 2022.3.2 on HA 2022.4

not seeing any restarts, i'm also seeing it intermittently when reading my airthings BLE sensors. when i say intermittently sometimes it reads fine but most time it errors out on at least one of the sensors its reading, most of the times the error comes up on the last sensor value read (tvoc) as in the log sample below.

with two boards:
```
esp32:
  board: nodemcu-32s
  framework:
    type: arduino
```
and 
```
esp32:
  board: esp32dev
  framework:
    type: arduino
```

here is the code running on both boards:
```
esphome:
  name: keeyees

esp32:
  board: nodemcu-32s
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:
  password: "XXXXXXXXXXXXX"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

sensor:
  - platform: airthings_wave_mini
    ble_client_id: airthingsmini1
    update_interval: 5min # default
    temperature:
      name: "B1 WaveMini Temperature"
    pressure:
      name: "B1 WaveMini Pressure"
    humidity:
      name: "B1 WaveMini Humidity"
    tvoc:
      name: "B1 WaveMini VOC"

ble_client:
  - mac_address: F4:60:77:75:19:91
    id: airthingsmini1
    
esp32_ble_tracker:

web_server:
  port: 80
```
log snipet
```
[12:07:25][D][ble_client:047]: Found device at MAC address [F4:60:77:75:19:91]
[12:07:25][I][ble_client:085]: Attempting BLE connection to f4:60:77:75:19:91
[12:07:39][I][airthings_wave_mini:015]: Connected successfully!
[12:07:43][I][ble_client:161]: Service UUID: 0x1800
[12:07:43][I][ble_client:162]:   start_handle: 0x1  end_handle: 0x7
[12:07:43][I][ble_client:347]:  characteristic 0x2A00, handle 0x3, properties 0x2
[12:07:43][I][ble_client:347]:  characteristic 0x2A01, handle 0x5, properties 0x2
[12:07:43][I][ble_client:347]:  characteristic 0x2A04, handle 0x7, properties 0x2
[12:07:43][I][ble_client:161]: Service UUID: 0x1801
[12:07:43][I][ble_client:162]:   start_handle: 0x8  end_handle: 0xb
[12:07:43][I][ble_client:347]:  characteristic 0x2A05, handle 0xa, properties 0x20
[12:07:43][I][ble_client:161]: Service UUID: B42E3882-ADE7-11E4-89D3-123B93F75CBA
[12:07:43][I][ble_client:162]:   start_handle: 0xc  end_handle: 0x18
[12:07:43][I][ble_client:347]:  characteristic B42E3B98-ADE7-11E4-89D3-123B93F75CBA, handle 0xe, properties 0x2
[12:07:43][I][ble_client:347]:  characteristic B42E3EF4-ADE7-11E4-89D3-123B93F75CBA, handle 0x12, properties 0x2c
[12:07:43][I][ble_client:347]:  characteristic B42E41C4-ADE7-11E4-89D3-123B93F75CBA, handle 0x16, properties 0x10
[12:07:43][I][ble_client:161]: Service UUID: F000FFC0-0451-4000-B000-000000000000
[12:07:43][I][ble_client:162]:   start_handle: 0x19  end_handle: 0x25
[12:07:43][I][ble_client:347]:  characteristic F000FFC1-0451-4000-B000-000000000000, handle 0x1b, properties 0x1c
[12:07:43][I][ble_client:347]:  characteristic F000FFC2-0451-4000-B000-000000000000, handle 0x1f, properties 0x1c
[12:07:43][I][ble_client:347]:  characteristic F000FFC5-0451-4000-B000-000000000000, handle 0x23, properties 0x14
[12:07:43][I][ble_client:161]: Service UUID: 0x180A
[12:07:43][I][ble_client:162]:   start_handle: 0x26  end_handle: 0xffff
[12:07:43][I][ble_client:347]:  characteristic 0x2A23, handle 0x28, properties 0x2
[12:07:43][I][ble_client:347]:  characteristic 0x2A24, handle 0x2a, properties 0x2
[12:07:43][I][ble_client:347]:  characteristic 0x2A25, handle 0x2c, properties 0x2
[12:07:43][I][ble_client:347]:  characteristic 0x2A26, handle 0x2e, properties 0x2
[12:07:43][I][ble_client:347]:  characteristic 0x2A27, handle 0x30, properties 0x2
[12:07:43][I][ble_client:347]:  characteristic 0x2A29, handle 0x32, properties 0x2
[12:07:43][D][esp32_ble_tracker:217]: Starting scan...
[12:07:43][D][sensor:125]: 'B1 WaveMini Humidity': Sending state 31.67000 % with 2 decimals of accuracy
[12:07:43][D][sensor:125]: 'B1 WaveMini Pressure': Sending state 896.23999 hPa with 2 decimals of accuracy
[12:07:43][D][sensor:125]: 'B1 WaveMini Temperature': Sending state 20.23001 °C with 2 decimals of accuracy
[12:07:43][D][sensor:125]: 'B1 WaveMini VOC': Sending state 72.00000 ppb with 0 decimals of accuracy
[12:07:43][E][json:034]: Could not allocate memory for JSON document! Requested 5120 bytes, largest free heap block: 38872 bytes
[12:07:43][I][ble_client:075]: [f4:60:77:75:19:91] Disabling BLE client.
[12:07:43][W][airthings_wave_mini:021]: Disconnected!
```

statte, 2022-04-07:
Best if I provide some more additional info and to point out that I'm using ESP8266 not ESP32

```
esphome:
  platform: ESP8266
  board: d1_mini
```

And HomeAssistant 

```
Version	core-2022.3.8
Installation Type	Home Assistant Container
```

running config

```
substitutions:
  device_name: slimmelezer-1sagXXXXXXXXXXXX
  device_description: "DIY P1 module to read your smart meter"
     
esphome:
  name: ${device_name}
  comment: "${device_description}"
  platform: ESP8266
  esp8266_restore_from_flash: true
  board: d1_mini
  name_add_mac_suffix: false
  project:
    name: zuidwijk.slimmelezer
    version: "1.0"
  on_boot:
    then:
      - if:
          condition:
            lambda: return id(has_key);
          then:
            - lambda: |-
                std::string key(id(stored_decryption_key), 32);
                id(dsmr_instance).set_decryption_key(key);
          else:
            - logger.log:
                level: info
                format: "Not using decryption key. If you need to set a key use Home Assistant service 'ESPHome:  ${device_name}_set_dsmr_key'"
 
wifi:
  networks:
    ssid: !secret wifi_ssid
    password: !secret wifi_password
  domain: ".somedomain.here"
 
  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "${device_name}-AP"
    password: XXXXXXXXXX
    ap_timeout: 15s
 
captive_portal:
 
# Enable logging
logger:
  baud_rate: 0
 
# Enable Home Assistant API
api:
  services:
    service: set_dsmr_key
    variables:
      private_key: string
[... 146 lines cut ...]
[21:55:41][E][json:033]: Could not allocate memory for JSON document! Requested 5120 bytes, largest free heap block: 1144 bytes
[21:55:41][D][sensor:124]: 'devicenamehere Power Produced': Sending state 0.00000 kW with 3 decimals of accuracy
[21:55:41][E][json:033]: Could not allocate memory for JSON document! Requested 5120 bytes, largest free heap block: 1144 bytes
[21:55:41][D][sensor:124]: 'devicenamehere Voltage Phase 1': Sending state 229.50000 V with 1 decimals of accuracy
[21:55:41][E][json:033]: Could not allocate memory for JSON document! Requested 5120 bytes, largest free heap block: 1144 bytes
[21:55:41][D][sensor:124]: 'devicenamehere Voltage Phase 2': Sending state 0.00000 V with 1 decimals of accuracy
[21:55:41][E][json:033]: Could not allocate memory for JSON document! Requested 5120 bytes, largest free heap block: 1144 bytes
[21:55:41][D][sensor:124]: 'devicenamehere Voltage Phase 3': Sending state 227.39999 V with 1 decimals of accuracy
[21:55:41][E][json:033]: Could not allocate memory for JSON document! Requested 5120 bytes, largest free heap block: 1144 bytes
[21:55:41][D][sensor:124]: 'devicenamehere Current Phase 1': Sending state 1.91000 A with 1 decimals of accuracy
[21:55:41][E][json:033]: Could not allocate memory for JSON document! Requested 5120 bytes, largest free heap block: 1144 bytes
[21:55:41][D][sensor:124]: 'devicenamehere Current Phase 2': Sending state 0.68000 A with 1 decimals of accuracy
[21:55:41][E][json:033]: Could not allocate memory for JSON document! Requested 5120 bytes, largest free heap block: 1144 bytes
[21:55:41][D][sensor:124]: 'devicenamehere Current Phase 3': Sending state 2.14000 A with 1 decimals of accuracy
[21:55:41][E][json:033]: Could not allocate memory for JSON document! Requested 5120 bytes, largest free heap block: 1144 bytes
[21:55:41][D][sensor:124]: 'devicenamehere Power Consumed Phase 1': Sending state 0.34400 kW with 3 decimals of accuracy
[21:55:41][E][json:033]: Could not allocate memory for JSON document! Requested 5120 bytes, largest free heap block: 1144 bytes
[21:55:41][D][sensor:124]: 'devicenamehere Power Consumed Phase 2': Sending state 0.00000 kW with 3 decimals of accuracy
[21:55:41][E][json:033]: Could not allocate memory for JSON document! Requested 5120 bytes, largest free heap block: 1144 bytes
[21:55:41][D][sensor:124]: 'devicenamehere Power Consumed Phase 3': Sending state 0.09700 kW with 3 decimals of accuracy
[21:55:41][E][json:033]: Could not allocate memory for JSON document! Requested 5120 bytes, largest free heap block: 1144 bytes

```

Then reboot happens and works again for like 1 to 2 minutes

sajikur, 2022-04-08:
Same errors here with esp32, esp32 mini d1, esp8266

DJJo14, 2022-04-13:
I also have this problem on my ESP32 with home assistant 2022.4.1 and esphome 2022.3.2
Looks like there is a fix in the make: esphome/esphome#3377 but this is only for the esp32 not the esp8266

sajikur, 2022-04-13:
No way for me. Ask for downgrade or update. My esp32 has many sensors 
miflora and temp/hum to control so workaround doesn't solve

dz0ny, 2022-04-13:
> No way for me. Ask for downgrade or update. My esp32 has many sensors miflora and temp/hum to control so workaround doesn't solve In data 13 aprile 2022 09:26:16 DJJo14 ***@***.***> ha scritto:
> […](#)

The fix is for both families of devices. If you want to test it out, create **components** directory and then copy just the folder https://github.com/dz0ny/esphome/tree/fix/json-malloc/esphome/components/json.

Structure:
```
one.yaml
two.yaml
components
  json
```

sajikur, 2022-04-13:
Thank you but I prefer wait for update. I'm Not sure to be able to do it in 
the right mode.

jesserockz, 2022-04-13:
> Thank you but I prefer wait for update

@sajikur If you cannot test the fix and verify that it actually works, how do I know to merge the PR?

sajikur, 2022-04-13:
I'm not so skilled like you to handle code

jesserockz, 2022-04-13:
The YAML was given to you above:

```yaml
external_components:
  - source: github://pr#3384
    components: [ json ]
logger:
  level: VERBOSE
```

You dont need to know code, you just need to add that to your device yaml and you can test the PR code.

statte, 2022-04-13:
I've tested on my setup on ESP8266 with the DSMR 

```
23:00:23 | [V] | [api.connection:893] | Cannot send message because of TCP buffer space
23:00:23 | [V] | [sensor:074] | 'namehere Energy Consumed Tariff 2': Received new state 10143.069336
23:00:23 | [D] | [sensor:124] | 'namehere Energy Consumed Tariff 2': Sending state 10143.06934 kWh with 3 decimals of accuracy
23:00:23 | [V] | [json:040] | Size after shrink 124 bytes
23:00:23 | [V] | [api.connection:893] | Cannot send message because of TCP buffer space
23:00:23 | [V] | [dsmr:105] | Start reading data from P1 port
23:00:23 | [V] | [dsmr:142] | Header of telegram found
23:00:23 | [V] | [dsmr:176] | Footer of telegram found
23:00:23 | [V] | [dsmr:257] | Trying to parse telegram
23:00:23 | [V] | [dsmr:118] | Stop reading data from P1 port
23:00:23 | [V] | [sensor:074] | 'namehere Energy Consumed Tariff 1': Received new state 34.446999
23:00:23 | [D] | [sensor:124] | 'namehere Energy Consumed Tariff 1': Sending state 34.44700 kWh with 3 decimals of accuracy
23:00:23 | [V] | [json:040] | Size after shrink 120 bytes
23:00:23 | [V] | [api.connection:893] | Cannot send message because of TCP buffer space
23:00:23 | [V] | [api.connection:893] | Cannot send message because of TCP buffer space
23:00:24 | [V] | [sensor:074] | 'namehere Energy Consumed Tariff 2': Received new state 10143.069336
23:00:24 | [D] | [sensor:124] | 'namehere Energy Consumed Tariff 2': Sending state 10143.06934 kWh with 3 decimals of accuracy
23:00:24 | [V] | [json:040] | Size after shrink 124 bytes
23:00:24 | [V] | [api.connection:893] | Cannot send message because of TCP buffer space
23:00:24 | [V] | [api.connection:893] | Cannot send message because of TCP buffer space
23:00:24 | [V] | [sensor:074] | 'namehere Energy Produced Tariff 1': Received new state 0.032000
23:00:24 | [V] | [api.connection:893] | Cannot send message because of TCP buffer space
23:00:24 | [V] | [dsmr:142] | Header of telegram found
23:00:24 | [V] | [dsmr:176] | Footer of telegram found
23:00:24 | [V] | [dsmr:257] | Trying to parse telegram
23:00:24 | [V] | [dsmr:118] | Stop reading data from P1 port
```

So I am getting a different error message, but the reboot still happens.

dz0ny, 2022-04-13:
Thanks, @statte. In your case, I think reboot happens because the node runs out of memory. The "Cannot send message because of TCP buffer space" is saying "hey I'm supposed to write something, but my buffer is empty". The memory error would be dumped to your UART when in happens, so you don't see it here. You can maybe try removing some non-essential service(OTA, captive portal) to see if that gives you more room.

statte, 2022-04-13:
Oh ok,

Sorry I hijacked this thread.

At least the JSON errors are gone.

Thanks

dz0ny, 2022-04-13:
And another hack would be to have shorter names ( Power Consumed Phase  > PCP) as that also consumes memory.

sajikur, 2022-04-14:
> The YAML was given to you above:
> 
> ```yaml
> external_components:
>   - source: github://pr#3384
>     components: [ json ]
> logger:
>   level: VERBOSE
> ```
> 
> You dont need to know code, you just need to add that to your device yaml and you can test the PR code.

<html>
<body>

06:57:32 | [V] | [json:040] | Size after shrink 140 bytes
-- | -- | -- | --
06:57:32 | [V] | [api.connection:893] | Cannot send message because of TCP buffer space
06:57:32 | [V] | [json:040] | Size after shrink 144 bytes
06:57:32 | [V] | [api.connection:893] | Cannot send message because of TCP buffer space
06:57:32 | [V] | [json:040] | Size after shrink 156 bytes
06:57:32 | [V] | [api.connection:893] | Cannot send message because of TCP buffer space
06:57:32 | [V] | [json:040] | Size after shrink 164 bytes
06:57:32 | [V] | [api.connection:893] | Cannot send message because of TCP buffer space
06:57:32 | [V] | [api.connection:893] | Cannot send message because of TCP buffer space
06:57:32 | [V] | [json:040] | Size after shrink 128 bytes
06:57:32 | [V] | [api.connection:893] | Cannot send message because of TCP buffer space
06:57:32 | [V] | [json:040] | Size after shrink 144 bytes
06:57:32 | [V] | [api.connection:893] | Cannot send message because of TCP buffer space
06:57:32 | [V] | [json:040] | Size after shrink 156 bytes
06:57:32 | [V] | [api.connection:893] | Cannot send message because of TCP buffer space
06:57:32 | [V] | [json:040] | Size after shrink 164 bytes
06:57:32 | [V] | [api.connection:893] | Cannot send message because of TCP buffer space
06:57:32 | [V] | [api.connection:893] | Cannot send message because of TCP buffer space
06:57:32 | [V] | [api.connection:893] | Cannot send message because of TCP buffer space
06:57:32 | [V] | [api.connection:893] | Cannot send message because of TCP buffer space
06:57:32 | [V] | [json:040] | Size after shrink 144 bytes
06:57:32 | [V] | [api.connection:893] | Cannot send message because of TCP buffer space
06:57:32 | [V] | [json:040] | Size after shrink 148 bytes
06:57:32 | [V] | [api.connection:893] | Cannot send message because of TCP buffer space
06:57:32 | [V] | [json:040] | Size after shrink 164 bytes
06:57:32 | [V] | [api.connection:893] | Cannot send message because of TCP buffer space
06:57:32 | [V] | [json:040] | Size after shrink 128 bytes
06:57:32 | [V] | [json:040] | Size after shrink 148 bytes
06:57:32 | [V] | [api.connection:893] | Cannot send message because of TCP buffer space
06:57:32 | [V] | [json:040] | Size after shrink 144 bytes
06:57:32 | [V] | [api.connection:893] | Cannot send message because of TCP buffer space
06:57:32 | [V] | [api.connection:893] | Cannot send message because of TCP buffer space
06:57:32 | [V] | [api.connection:893] | Cannot send message because of TCP buffer space
06:57:32 | [V] | [api.connection:893] | Cannot send message because of TCP buffer space
06:57:32 | [V] | [json:040] | Size after shrink 164 bytes

</body>
</html>

statte, 2022-04-14:
> 06:57:32	[V]	[api.connection:893]	Cannot send message because of TCP buffer space

The original poster is now also mentioning this error in the logs, so maybe i'm not hijacking afterall ?

I've tried to disable all non essential components(captive portal, web, OTA , ...), changed the names as suggested,
It takes a lot longer for the issue (reboot) to occur , but it still happens instead of 1-2 minutes now its +-10 minutes

dz0ny, 2022-04-14:
It's a larger issue for sure, bottom line is not enough memory to construct new objects in memory. The JSON became exposed because @jesserockz added tracing for it. The **api** module which uses **protobuffers** had tracing from the beginning.

Now in your case the **dsmr** can also be leaking since the issue becomes more severe with time or just overflows when trying to parse data from UART.

statte, 2022-04-14:
@dz0ny thank you for clarifying!

sajikur, 2022-04-16:
But I do not understand.  Wouldn't it be easier to go back to the latest working version?

jesserock, 2022-04-16:
I'm still on this list. OK!

jesserock, 2022-04-16:
Then lets burry every one with messages.

jesserock, 2022-04-16:
Like, I'm thinking of what I want for lunch. Maybe salad? Does anyone want
the YAML file I'll use to create my salad?

jesserock, 2022-04-16:
No? I'm wondering if I can do this with Python.

jesserock, 2022-04-16:
I mean, you can pretty much do anything in python amirite?

jesserock, 2022-04-16:
Or Ruby. Go. You know, so many options, maybe Perl.

jesserock, 2022-04-16:
Maybe bash shell. What script should I use to pelt this with so many
messages you all shut the fuck up?

jesserock, 2022-04-16:
I'm open to suggestions.

sanderlv, 2022-04-16:
> I'm open to suggestions.
> 
> 
> > Maybe bash shell. What script should I use to pelt this with so many
> > messages you all shut the fuck up?
> >
> >
> >> Or Ruby. Go. You know, so many options, maybe Perl.
> >>
> >>
> >>> I mean, you can pretty much do anything in python amirite?
> >>>
> >>>
> >>>
> >>>> No? I'm wondering if I can do this with Python.
> >>>>
> >>>>
> >>>>> Like, I'm thinking of what I want for lunch. Maybe salad? Does anyone
> >>>>> want the YAML file I'll use to create my salad?
> >>>>>
> >>>>>
> >>>>>> Then lets burry every one with messages.
> >>>>>>
> >>>>>>
> >>>>>>> I'm still on this list. OK!
> >>>>>>>
> >>>>>>>
> >>>>>>>> But I do not understand. Wouldn't it be easier to go back to the
> >>>>>>>> latest working version?

Maybe help to solve this issue? Since you are knowledgeable to help out? Thank you! Jesse you Rock!

jesserock, 2022-04-16:
Maybe pay me. I don't work for free.

mmakaay, 2022-04-19:
Note that this is not gool ol' @jesserockz but somebody else who is acting quite peculiar.
The Real Jesse™ has been warned so he can report the abuse.

tdvantine, 2022-04-19:
I reported the account but GitHub came back and said they wouldn't do anything that I should just block the account and let the admins here know 🤦‍♂️

ssieb, 2022-04-19:
@jesserock, someone tagged you by accident.  There's a big "Unsubscribe" button you can click to stop getting messages.

jesserockz, 2022-04-19:
I have cleaned up the comments here. Please keep on topic from now.

sajikur, 2022-04-21:
last version of esphome fix the error in log for me. Thanx for your work

Rockument, 2022-05-20:
Multiple Mi Floras with latest esphome have this error and entities not show all of it
ESP32

`10:41:29 | [E] | [json:034] | Could not allocate memory for JSON document! Requested 5120 bytes, largest free heap block: 37700 bytes`

sajikur, 2022-05-20:
My esp32 lan it's OK. No errors in log. Esp32 wifi has memory problems to read miflora

dz0ny, 2022-05-20:
> Multiple Mi Floras with latest esphome have this error and entities not show all of it ESP32
> 
> `10:41:29 | [E] | [json:034] | Could not allocate memory for JSON document! Requested 5120 bytes, largest free heap block: 37700 bytes`

Requested 5120 bytes is from version released in March, the latest release no longer uses static allocation of documents, so you wouldn’t get that error. https://github.com/esphome/esphome/releases/tag/2022.5.0

dz0ny, 2022-05-20:
@jesserockz There should probably a doc that outlines what is this error and how to fix it.

Tuckie, 2022-10-06:
I just had this issue on an esp32 running the bedjet component.

jesserock, 2022-10-11:
Man I would really like to get off this noisy ass list. Especially since I
was put on it by accident and the topic isn't really anything I touch.

mmakaay, 2022-10-11:
Yeah, that is a pain. 
Maybe try: https://docs.github.com/en/account-and-profile/setting-up-and-managing-your-personal-account-on-github/managing-access-to-your-personal-repositories/removing-yourself-from-a-collaborators-repository

trrevvorr, 2022-11-20:
FWIW I got this message ("Could not allocate memory for JSON document! Requested 0 bytes") when the API responded with a non-200 status code. In my case I had to switch from http to https for the API and then everything worked as expected but I would imagine you may get the same error with any other non-200 status code. This was using an ESP32.

sheminasalam, 2022-12-26:
i am still getting this issue in my sonoff basic with the latest esphome. any solution?

Pro-Control, 2023-02-28:
[17:38:27][E][json:039]: Could not allocate memory for JSON document! Requested 488 bytes, largest free heap block: 488 bytes
[17:38:27][E][json:039]: Could not allocate memory for JSON document! Requested 488 bytes, largest free heap block: 488 bytes

andrewjswan, 2023-06-19:
ESP8266 - ESPHome 2023.5.5
```
[20:02:33][E][json:048]: Could not allocate memory for JSON document! Overflowed largest free heap block: 120 bytes
[20:02:41][E][json:048]: Could not allocate memory for JSON document! Overflowed largest free heap block: 56 bytes
[20:05:53][E][json:048]: Could not allocate memory for JSON document! Overflowed largest free heap block: 8 bytes
[20:07:40][E][json:048]: Could not allocate memory for JSON document! Overflowed largest free heap block: 40 bytes
[20:07:47][E][json:048]: Could not allocate memory for JSON document! Overflowed largest free heap block: 8 bytes
[20:08:03][E][json:048]: Could not allocate memory for JSON document! Overflowed largest free heap block: 40 bytes
[20:08:04][E][json:048]: Could not allocate memory for JSON document! Overflowed largest free heap block: 40 bytes
[20:08:18][E][json:048]: Could not allocate memory for JSON document! Overflowed largest free heap block: 8 bytes
[20:08:19][E][json:048]: Could not allocate memory for JSON document! Overflowed largest free heap block: 40 bytes
[20:08:30][E][json:048]: Could not allocate memory for JSON document! Overflowed largest free heap block: 40 bytes
[20:08:35][E][json:048]: Could not allocate memory for JSON document! Overflowed largest free heap block: 40 bytes
[20:10:27][E][json:048]: Could not allocate memory for JSON document! Overflowed largest free heap block: 40 bytes
[20:10:32][E][json:048]: Could not allocate memory for JSON document! Overflowed largest free heap block: 40 bytes
```

autox86, 2023-08-15:
And here as well with my serial output:
```
[00:12:07][E][json:039]: Could not allocate memory for JSON document! Requested 232 bytes, largest free heap block: 232 bytes
[00:12:07][D][cover:076]: 'ShiftTest12v Esstisch' - Setting
[00:12:07][D][cover:080]:   Command: STOP
[00:12:07]
[00:12:07]User exception (panic/abort/assert)
[00:12:07]--------------- CUT HERE FOR EXCEPTION DECODER ---------------
[00:12:07]
[00:12:07]Unhandled C++ exception: OOM
[00:12:07]
[00:12:07]>>>stack>>>
WARNING Found stack trace! Trying to decode it
[00:12:07]
[00:12:07]ctx: cont
[00:12:07]sp: 3ffffd10 end: 3fffffd0 offset: 0000
[00:12:07]3ffffd10:  3ffffd30 000001ff 00000000 0000002a
[00:12:07]3ffffd20:  000000fe 00000000 00000000 00000000
[00:12:07]3ffffd30:  00000000 00000000 00000000 40100d8c
```

DEBUG info:
```
[00:10:43][C][debug:056]: Debug component:
[00:10:43][D][debug:068]: ESPHome version 2023.7.1
[00:10:43][D][debug:072]: Free Heap Size: 17024 bytes
[00:10:43][D][debug:100]: Flash Chip: Size=4096kB Speed=40MHz Mode=DOUT
[00:10:43][D][debug:256]: Chip ID: 0x00FF67CB
[00:10:43][D][debug:257]: SDK Version: 2.2.2-dev(38a443e)
[00:10:43][D][debug:258]: Core Version: 3.0.2
[00:10:43][D][debug:259]: Boot Version=31 Mode=1
[00:10:43][D][debug:260]: CPU Frequency: 80
[00:10:43][D][debug:261]: Flash Chip ID=0x0016405E
```

Device keeps restarting :(
No use possible