esphome/issues #3190: SNTP's now() returns values in powers of 2 instead of the real timestamp

melyux, 2022-03-29:
### The problem

I'm using the SNTP platform to get the time whenever a button is pressed and put that time into a Template Sensor. I'm not using the Home Assistant API. The logs show the SNTP time is grabbed correctly every 15 minutes.

The value put into the Template Sensor by the `sensor.template.publish` action using `!lambda 'return id(sntp_time).now().timestamp;'` is indeed a timestamp... but it's off. Like this, as you can see in the log:

- At 4:49:34 PM, the Template Sensor is set to 1648590592.00000 → 4:49:52 PM (in the future!)
- At 4:49:41 PM, the Template Sensor is AGAIN set to 1648590592.00000 → 4:49:52 PM (in the future!)
- At 4:49:50 PM, the Template Sensor is ONCE AGAIN set to 1648590592.00000 → 4:49:52 PM (in the future!) 

A few minutes later, the published timestamp finally changes

- At 4:54:34 PM, we get 1648590848.00000 → 4:54:08 PM (in the past)
- At 5:01:54 PM, we get 1648591360.00000 → 5:02:40 PM (in the future!)

Interestingly, if you subtract these epoch times, you get these differences:
* 1648591360 – 1648590848 = 512
* 1648590848 – 1648590592 = 256

So 2^9 and 2^8. Very strange.

### Which version of ESPHome has the issue?

2022.2.4

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

N/A

### What platform are you using?

ESP8266

### Board

D1 Mini

### Component causing the issue

sntp

### Example YAML snippet

```yaml
time:
- platform: sntp
  id: sntp_time
  timezone: "America/Los_Angeles"

sensor:
- platform: template
  name: "Last Press"
  device_class: timestamp
  id: last_press
  update_interval: 12h

binary_sensor:
- platform: gpio
  pin:
    number: D2
    inverted: true
  name: "Button"
  filters:
  - delayed_on: 50ms
  - delayed_off: 5000ms
  on_press:
    then:
    - if:
        condition:
          switch.is_off: chime_mute
        then:
        - switch.turn_on: chime
        - delay: 250ms
        - switch.turn_off: chime
    - sensor.template.publish:
        id: last_press
        state: !lambda 'return id(sntp_time).now().timestamp;'
```

### Anything in the logs that might be useful for us?

```txt
[16:47:35][D][sntp:074]: Synchronized time: 2022-03-29 16:47:35
[16:49:34][D][binary_sensor:036]: 'Button': Sending state ON
[16:49:34][D][sensor:124]: 'Last Press': Sending state 1648590592.00000  with 1 decimals of accuracy
[16:49:39][D][binary_sensor:036]: 'Button': Sending state OFF
[16:49:41][D][binary_sensor:036]: 'Button': Sending state ON
[16:49:41][D][sensor:124]: 'Last Press': Sending state 1648590592.00000  with 1 decimals of accuracy
[16:49:47][D][binary_sensor:036]: 'Button': Sending state OFF
[16:49:50][D][binary_sensor:036]: 'Button': Sending state ON
[16:49:50][D][sensor:124]: 'Last Press': Sending state 1648590592.00000  with 1 decimals of accuracy
[16:49:55][D][binary_sensor:036]: 'Button': Sending state OFF
[16:54:34][D][binary_sensor:036]: 'Button': Sending state ON
[16:54:34][D][sensor:124]: 'Last Press': Sending state 1648590848.00000  with 1 decimals of accuracy
[16:54:39][D][binary_sensor:036]: 'Button': Sending state OFF
[17:01:54][D][binary_sensor:036]: 'Button': Sending state ON
[17:01:54][D][sensor:124]: 'Last Press': Sending state 1648591360.00000  with 1 decimals of accuracy
[17:01:59][D][binary_sensor:036]: 'Button': Sending state OFF
```

melyux, 2022-07-28:
I don't think it should mark as stale issues that have received no replies. Only issues where the last poster is a contributor (meaning a response was expected from the reporter but they failed to further collaborate).

ssieb, 2022-11-27:
The problem is that you're trying to represent a 32-bit integer using a float.  But a float does not have enough resolution for that, so you get rounding.  There's nothing wrong with the timestamp.

closed 2022-11-27 by ssieb

melyux, 2022-11-27:
@ssieb Okay, how do I not use a float?

ssieb, 2022-11-27:
GitHub issues are not for support requests, it would be better to ask on the esphome discord server.
Why are you trying to put it in a sensor?  What is your goal?

melyux, 2022-11-27:
Support request? I meant, what part of my configuration indicated to you that I was using a float?

I'm using the sensor to record the last time a button was pressed. This sensor acts to send a retained message to the MQTT server with that timestamp.

ssieb, 2022-11-27:
A sensor stores a float value.  I suggest sending an mqtt message directly with the timestamp if that's what you're trying to do.
"support request".   This isn't a bug or problem with esphome.  It's something you're trying to do and need help with.

melyux, 2022-12-28:
This was the preferred way of publishing a timestamp that I thought was implied through some documentation, probably, so whether it was a "support request" or a bug was something that I didn't know enough to know at the time.

I'm trying to get the MQTT publishing directly to work as suggested, but MQTT publish only takes strings and `.timestamp_now()` or `.now()` have `time_t`. Trying to figure that out.

There are earlier issues about this that I just found, so this isn't a support request but a problem (or more like an ambiguity) in ESPHome. See #1099 and #2674. This is essentially a duplicate of those

ssieb, 2022-12-28:
`to_string()`