esphome/issues #3193: ESP8266 only accessible/visible when in adoption mode.

UV-PWRD, 2022-03-31:
### The problem

New to al this so forgive my oversights. 

This is a Nodemcu V3 board and I'm trying to get it going with a Dallas temp sensor. 

I can make the board adoptable via the ESP Home Web Portal and have it connect to my home WiFi at which point I can browse to the web interface on the device and the device is visible in the ESP Home add on within Home Assistant. 

If I attempt to flash ANYTHING to the device, including the very same config that's on it after adoption, from that point forward the device will response to ping only but I cannot access the web interface and the device is no longer visible in Home Assistant. 

I have tried flashing from the Nodemcu flash tool, ESP Home command line (Win 10 x64) and manually selecting the BIN and doing it OTA or from the ESP Home Web Portal all without success. 

I've tried custom binaries created by the ESP Home command line wizard as well as copy/paste the config from the yaml that contains the code used when the device is adoptable also. 

I've tried selecting Nodemcuv2 and also the generic esp01_1m boards.

I've also tried using a different wireless network (home vs office to rule out the Wifi/router) and I've also intentionally set the WiFi details wrong to force it to local AP mode, which I can connect to. 

Any help appreciated! 

### Which version of ESPHome has the issue?

2022.3.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.2.6

### What platform are you using?

ESP8266

### Board

modemcu

### Component causing the issue

Unsure

### Example YAML snippet

```yaml
esphome:
  name: hotwatertank

esp8266:
  board: nodemcuv2

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: ""

ota:
  password: ""

wifi:
  ssid: "REDACTED"
  password: "REDACTED"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Hotwatertank Fallback Hotspot"
    password: "REDCATED"

captive_portal:

dallas:
  - pin: GPIO4
```

dougiteixeira, 2022-03-31:
> ```yaml
> # Enable Home Assistant API
> api:
>   password: ""
> ```

> ```yaml
> ota:
>   password: ""
> ```

Have you tried removing `password` from `ota` and `api` to see if it works?

I'm not sure if it's really blank in your code or just removed the content to post.

nagyrobi, 2022-03-31:
What do you mean by adoption mode?

UV-PWRD, 2022-04-01:
> > ```yaml
> > # Enable Home Assistant API
> > api:
> >   password: ""
> > ```
> 
> > ```yaml
> > ota:
> >   password: ""
> > ```
> 
> Have you tried removing `password` from `ota` and `api` to see if it works?
> 
> I'm not sure if it's really blank in your code or just removed the content to post.

Yes have tried that also, that config is as produced by the ESP Home Web Portal

UV-PWRD, 2022-04-01:
> What do you mean by adoption mode?
> 
> Also web server config is missing, no wonders you can't access it:
> 
> ```
> web_server:
>   port: 80
> ```

This as pictured. Once that step is done the device is visible on the network, in ESP Home within HA and I can access the web server for the board. 

If that server config is missing then it's missing from the base config as installed by the process pictured in my attachment. 

Yet I can still access the web server after adoption has been enabled? 

[image]

UV-PWRD, 2022-04-01:
> web_server:
>   port: 80

Adding this to the config caused the device to loop after applying it and would not connect to the wifi either, I added it at the bottom of the config but if there's a different spot let me know. 

This is an exceprt of what ESP Home Command Line utility was showing during the process.

[18:59:24]--------------- CUT HERE FOR EXCEPTION DECODER ---------------
[18:59:24]
[18:59:24]Exception (28):
←[33mWARNING Exception type: Access to invalid address: LOAD (wild pointer?)←[0m
[18:59:24]epc1=0x40222991 epc2=0x00000000 epc3=0x00000000 excvaddr=0x0c500000 depc=0x00000000
←[33mWARNING Decoded 0x40222991: esphome::progmem_read_byte(unsigned char const*)←[0m

After removing your suggested code it went back to the previous behaviour.

UV-PWRD, 2022-04-01:
> Yes have tried that also, that config is as produced by the ESP Home Web Portal

I've just realised something, we might be talking 2 different things. 

Let me clarify. 

After making the device adoptable, via the link on the ESP Home Web Portal as shown above, the device has a web interface that can be accessed once it connects to the wifi, pictured here. It just shows the log and allows OTA updates. 

[image]

Also, the device is visible in ESP Home within HA. Pictured. 

[image]

After I've uploaded ANYTHING else to the device, even using the same config that appears to be on it after it's "adoptable" I can no longer see the web front end of the device as shown below and I can no longer see the device in HA. 

However the device is still on the network, responds to ping by IP and via Hostname, which has changed confirming the firmware was uploaded. 

Does that help clarify?

nagyrobi, 2022-04-01:
Very strange. I don't see any "ADOPT" option anywhere in my environment.
I upload my first binary to the device with ESPHome flasher via serial port. After that, I use Install > Wirelessly.

UV-PWRD, 2022-04-01:
I'm unable to upload configs from HA directly because my HA is 32b and my RPi4 is 64bit so I have library issues.

I read it wouldn't matter as I can simply create/upload from the web portal, OTA or ESP Home command line, all of which appear to "work" even in so far as when i configure the dallas section I can see the sensor and receive an address from it. 

But I cannot see them in HA after that or access the Web Interface for the node itself

UV-PWRD, 2022-04-01:
ok I have worked it out happy days

[image]

nagyrobi, 2022-04-01:
don't forget to close issue

UV-PWRD, 2022-04-01:
I will after I've written a more detailed post about the outcome/steps taken for others to follow. 

That said, the end result is more to do with my lack of understanding but there is also still the same original issues. 

I'll clarify later when I'm on PC and can screen shot easily etc.

UV-PWRD, 2022-04-03:
OK so I'm back to explain a bit more. 

So it seems that all the code I had been flashing on was actually "working" but I was wrong about what the next steps should be and what I should expect to see after. 

I was expecting to see the device appear in the ESP Home Dashboard within HA and be adoptable, like I showed above. I assumed I would see my new device with the right hostname and "Discovered" next to it. 

What I found was, after flashing the device I could then go into integrations and manually add a device by IP/Hostname by selecting add integration and choosing ESP Home, you are then given this window.

[image]

As you can see in the background I was then able to add my device (after adding the address for the sensor itself so it had an entity to show me).

That gave me what I was after, in that the device is working, the sensor is sensing etc.

However, the device does not show up in the ESP Home Dashboard and is also not accessible on its own web interface still. 

Should the device be visible in the ESP Home Dashboard? All I see on the dashboard itself is this.

[image]

closed 2022-08-17 by github-actions[bot]