esphome/issues #3198: Gosund SP1 plug stopped working after 2022.3.0 update

flapedriza, 2022-04-03:
### The problem

I flashed some Gosund SP1 smart plugs with ESPHome using tuyaconvert when the current version was 2022.2.0, everything was working perfectly with all 4 of them.

Some days ago I saw a new update disclaimer on the ESPHome Home assistant integration and I wirelessly updated one of the plugs.

Now the plug stoppped working and in the router page I see that it connects for some seconds and then disconnects, the plug itself goes on and off every minute or so too. When it's connected to the network I cannot ping it.

I read the "My node keeps reconnecting randomly" section in the FAQ but I couldn't find anything there since I didn't change anything in the configuration, just updated the plug.

I tried pressing the plug button for some seconds to see if it gets into wifi setup mode so I can flash it again with tuyaconvert, but I had no luck with that either.

### Which version of ESPHome has the issue?

2022.3.0

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP8266

### Board

Gosund SP1

### Example YAML snippet

```yaml
substitutions:
  plug_name: gosund1
  friendly_name: Gosund 1
  current_res: "0.001193"
  voltage_div: "942"
  upd_interval: 3s

esphome:
  name: ${plug_name}

esp8266:
  board: esp8285

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  use_address: 192.168.2.31

logger:

api:

ota:

web_server:
  port: 80
  auth:
    username: !secret web_server_username
    password: !secret web_server_password

binary_sensor:
  - platform: gpio
    pin:
      number: GPIO3
      inverted: true
    name: "${friendly_name} Power Button"
    on_press:
      - switch.toggle: ${plug_name}_relay

switch:
  - platform: gpio
    id: ${plug_name}_led
    pin: GPIO1
    inverted: true
  
  - platform: gpio
    id: ${plug_name}_led_red
    pin: GPIO13
    inverted: true

  - platform: gpio
    name: "${friendly_name}"
    pin: GPIO14
    id: ${plug_name}_relay
    on_turn_on:
      - switch.turn_on: ${plug_name}_led
    on_turn_off:
      - switch.turn_off: ${plug_name}_led

sensor:
  - platform: hlw8012
    sel_pin:
      number: GPIO12
      inverted: true
    cf_pin: GPIO04
    cf1_pin: GPIO05
    current_resistor: ${current_res}
    voltage_divider: ${voltage_div}
    current:
      name: "${friendly_name} current"
      unit_of_measurement: A
      accuracy_decimals: 3
    voltage:
      name: "${friendly_name} Voltage"
      unit_of_measurement: V
      accuracy_decimals: 1
    power:
      name: "${friendly_name} Wattage"
      unit_of_measurement: W
      accuracy_decimals: 1
      id: ${plug_name}_wattage
      filters:
        - calibrate_linear:
          - 0 -> 1.65
          - 26.6 -> 15.01
          - 1674.5 -> 870.7
          - 3250.5 -> 1685
        - lambda: if (x < (2 + 1.65)) return 0; else return (x - 1.65);
    change_mode_every: 8
    update_interval: ${upd_interval}
  - platform: total_daily_energy
    name: "${friendly_name} Total Daily Energy"
    power_id: ${plug_name}_wattage
    filters:
      - multiply: 0.001
    unit_of_measurement: kWh

time:
  - platform: sntp
    id: ${plug_name}_time
```

Spirituss, 2022-05-18:
Faced the same issue with Blitzwolf SHP6 Pro and  Gosund SP111 - on Esphome 2021.11 they worked very stable for months, but after update to 2022.4 the situation became catastrophic. Here is the last restart sensor history (early morning, May, 18 update):
[image]

ssieb, 2022-08-02:
Since you're using the uart pins, you need to disable the serial logging.

scy, 2022-11-11:
> Since you're using the uart pins, you need to disable the serial logging.

Well, as far as I can see, this is true for the Gosund SP1, which is using GPIO1 (TX) as its blue LED and GPIO3 (RX) as its button pin. However, this shouldn’t really interfere with anything, should it? Maybe the LED will flicker when logs are written, but as long as you don’t manage to push the button fast and accurate enough to issue a reboot command, why should the device shut down or reboot?

Also, the Gosund SP111 (as well as the Nous A1, which I was seeing this issue on before I downgraded) is using neither GPIO1 nor GPIO3 for anything and is affected by this problem, too.

I’ve updated one of my A1s to 2022.10.2, let’s see whether something magically solved this issue in the mean time. I’ll leave the rest of my A1s at 2021.12.3, which is the last stable version.

scy, 2022-11-13:
Nope, using 2022.10.2 doesn’t solve the issue, the devices are still rebooting on irregular intervals.

Disabling serial logging by setting

```yaml
logger:
  baud_rate: 0
```

makes no difference either. It has to be something else.

I’m thinking about `git bisect`ing the issue, but since it can take up to a two-digit number of hours before a device reboots, this would probably take days or weeks to complete.

whighway, 2023-08-13:
I had the same problem (esphome 2023.7.1), device went offline every time trying to switch voltage to current (which fails). removing the `total_daily_energy` sensor "fixed" it