esphome/issues #3202: ESP32C3 E1.31 Light Support - Compiler Error

jedney, 2022-04-06:
### The problem

I have a working ESP32C3 (Arduino) build of ESPHome with a strip of 100x WS2812B lights attached to GPIO5. These are working fine from the WebUI and from Home Assistant. I wish to add E1.31 support so I can also use Hyperion NG, so I've added:

```
e131:
  method: unicast
```
Note: Using multicast doesn't make any difference.

And against the working light profile I've added:
```
effects:
  - e131:
    universe: 1
    channels: RGB
```

I now get a compiler error (see below).

### Which version of ESPHome has the issue?

2022.3.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.3.8

### What platform are you using?

ESP32

### Board

M5Stamp-C3

### Component causing the issue

E131

### Example YAML snippet

```yaml
esphome:
  name: m5-red
  platformio_options:
    platform_packages:
      - framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#2.0.2
    platform:
      - https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream

esp32:
  board: esp32-c3-devkitm-1
  framework:
    type: arduino

api:

ota:
  password: {REMOVED}
  
logger:
  level: WARN

wifi:
  ssid: !secret wifi_ssid
[... 182 lines cut ...]
   universe = htons(sbuff->universe);
   ^~~~~~~~
   utimes
src/esphome/components/e131/e131_packet.cpp:133:3: error: 'packet' was not declared in this scope
   packet.count = htons(sbuff->property_value_count);
   ^~~~~~
src/esphome/components/e131/e131_packet.cpp:133:3: note: suggested alternative: 'packet_'
   packet.count = htons(sbuff->property_value_count);
   ^~~~~~
   packet_
src/esphome/components/e131/e131.cpp: In member function 'virtual void esphome::e131::E131Component::loop()':
src/esphome/components/e131/e131.cpp:54:43: error: no matching function for call to 'esphome::e131::E131Component::packet_(std::vector<unsigned char>&, int&, esphome::e131::E131Packet&)'
     if (!packet_(payload, universe, packet)) {
                                           ^
In file included from src/esphome/components/e131/e131.cpp:3:
src/esphome/components/e131/e131.h:42:8: note: candidate: 'bool esphome::e131::E131Component::packet_(int)'
   bool packet_(const std::vector<uint8_t> &data, int &universe, E131Packet &packet);
        ^~~~~~~
src/esphome/components/e131/e131.h:42:8: note:   candidate expects 1 argument, 3 provided
*** [/data/m5-red/.pioenvs/m5-red/src/esphome/components/e131/e131_packet.cpp.o] Error 1
Compiling /data/m5-red/.pioenvs/m5-red/src/main.cpp.o
*** [/data/m5-red/.pioenvs/m5-red/src/esphome/components/e131/e131_addressable_light_effect.cpp.o] Error 1
*** [/data/m5-red/.pioenvs/m5-red/src/esphome/components/e131/e131.cpp.o] Error 1
========================== [FAILED] Took 4.14 seconds ==========================
```

jedney, 2022-08-05:
This still needs reviewing and fixing.

ssieb, 2022-12-17:
I wonder why this didn't get auto-closed by the merge.

closed 2022-12-17 by ssieb