esphome/issues #3203: Compile error with custom sensor component

Scoff123, 2022-04-06:
### The problem

I am no longer able to compile my code for my custom sensor component in ESPHome. This is after updating to ESPHome addon in Home Assistant OS 2022.3.1 and also subsequently to 2022.3..

I get error messages during the compile stage, which relates to my tof_vl53l1x.h file. 

### Which version of ESPHome has the issue?

2022.3.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.3.8

### What platform are you using?

ESP8266

### Board

Lolin D1 Mini Pro

### Component causing the issue

VL53L1X

### Example YAML snippet

```yaml
esphome:
  name: oiltank
  includes:
    - tof_vl53l1x.h
  libraries:
    - "Wire"
    - "VL53L1x"

esp8266:
  board: d1_mini_pro

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "myssid"
    password: "mypassword"

captive_portal:

web_server:
  port: 80
```

### Anything in the logs that might be useful for us?

```txt
When trying to compile to install to the board:

In file included from src/main.cpp:26:
src/tof_vl53l1x.h:7:63: error: expected class-name before '{' token
    7 | class MyCustomSensor : public PollingComponent, public Sensor {
      |                                                               ^
src/tof_vl53l1x.h: In member function 'virtual void MyCustomSensor::update()':
src/tof_vl53l1x.h:34:7: error: 'publish_state' was not declared in this scope
   34 |       publish_state(mm);
      |       ^~~~~~~~~~~~~
*** [/data/oiltank/.pioenvs/oiltank/src/main.cpp.o] Error 1
========================= [FAILED] Took 142.91 seconds =========================
```

### Additional information

My tof_vl53l1x.h file:

`#include "esphome.h"
#include <Wire.h>
#include <VL53L1X.h>

VL53L1X tof_sensor;

class MyCustomSensor : public PollingComponent, public Sensor {
 public:
  // constructor
  MyCustomSensor() : PollingComponent(5000) {} // polling every 5s

  void setup() override {
    // This will be called by App.setup()
    Wire.begin();
    Wire.setClock(400000); // use 400 kHz I2C

    tof_sensor.setTimeout(500);
    tof_sensor.setAddress(0x29);
    if (!tof_sensor.init()) {
      ESP_LOGE("VL53L1X custom sensor", "Failed to detect and initialize sensor!");
      while (1);
    }

    tof_sensor.setDistanceMode(VL53L1X::Short);
    tof_sensor.setMeasurementTimingBudget(500000);

    tof_sensor.startContinuous(50);
  }

  void update() override {
    uint16_t mm = tof_sensor.read();
    
    if (!tof_sensor.timeoutOccurred()) {
      publish_state(mm);
    } else {
      ESP_LOGE("VL53L1X custom sensor", "Timeout during read().");
    }
  }
};`

gordonthree, 2022-08-03:
Any luck with this? Running into the same issue, seems publish_state needs to come from an include somewhere, but not sure?

Scoff123, 2022-08-03:
I no longer use my custom component but an alternative native supported platform instead. However, before switching a couple of months ago, I did try recompiling with the custom component and it compiled fine and installed on my esp board with no problems. Therefore I believe one of the includes that I was referencing must have either received an update which addressed this issue, or an ESPHome update addressed the issue.

G-Pereira, 2022-09-12:
@Scoff123 is there any reason to keep this issue open in that case?

Scoff123, 2022-09-12:
Only my poor memory - will close it now

closed 2022-09-12 by Scoff123