esphome/issues #3204: Graph doesn't show trend

ilmisu, 2022-04-06:
### The problem

Hi All.
I've a graph component that should show a trend of a sensor that is fetched using the homeassistant platform.
I could not figure out how to show the trend.
the only think shown is the X axis, nothing more.
could somebody help me understanding why I've got this issue? I've followed all the guidelines reportend onto display component help page on ESPHome website.

### Which version of ESPHome has the issue?

2022.3.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.3.8

### What platform are you using?

ESP32

### Component causing the issue

display graph

### Example YAML snippet

```yaml
sensor:
  - platform: homeassistant
    entity_id: sensor.temperatura_esterna
    id: outside_temperature

graph:
  - id: trend_esterna
    height: 50
    width: 86
    sensor: outside_temperature
    duration: 12h
display:
  - platform: ssd1306_i2c
    model: "SSD1306 128x64"
    reset_pin: GPIO12
    address: 0x3C
    id: my_display
    pages:
      - id: tempex
        lambda: |-
          it.printf(64, 0, id(font3), TextAlign::TOP_CENTER, "Esterna");
          if (id(outside_temperature).has_state()) {
            it.printf(0, 60, id(font3), TextAlign::BASELINE_LEFT , "%.1f°", id(outside_temperature).state);
          }
          it.graph(43, 54 , id(trend_esterna));
```

### Additional information

from which DB esp takes datas to be shown in trends?

matthew73210, 2022-07-31:
Hello, have you managed to sort this out? I currently have the same issue.

ilmisu, 2022-07-31:
Unfortunately not, still same issue.

jesserockz, 2022-07-31:
ESPHome only keeps state updates in memory from the last time it rebooted. It does not fetch the history from anywhere else.

matthew73210, 2022-07-31:
Kinda defeats the purpose of having a display that fetches data from home assistant. I think it would be a good idea though, there's demande for it. My setup is an e-paper which updates at the end of the day, would be nice to have a graph and not upload and display an image of the graph.

closed 2022-12-17 by github-actions[bot]

ssieb, 2022-12-17:
Yes, it would be nice, but it's not currently possible.  It requires extending the api and everything between the device and HA needs to support it.  You could file a feature request.