esphome/issues #3207: 4MB device only utilizing 1835008 bytes.

pashdown, 2022-04-08:
### The problem

[image]

The board is recognized as  having 4MB of flash, why is it reporting that it has only 1835008? I can see a problem trying to upload new firmware if the size of the firmware.bin is larger than the free space (1106608 > (1835008 - 1106498)).  Attempting to OTA updates always fail, the board needs to be pulled from service in order to USB update and this can be inconvenient for production.

### Which version of ESPHome has the issue?

2022.3.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.3.6

### What platform are you using?

ESP32

### Board

wt32-eth01

### Component causing the issue

OTA upload

nagyrobi, 2022-04-09:
Did you try to put boot ESPHome in Safe Mode before doing OTA?

pashdown, 2022-04-09:
> Did you try to put boot ESPHome in Safe Mode before doing OTA?

I will try that, but it requires taking them out of production to USB flash this feature.  It still doesn't address why the 4MB flash appears to be ignored.

nagyrobi, 2022-04-09:
It's another thing indeed.
If you use memory intensive stuff like BLE, OTA is likely to crash... Safe Mode helps with that...

akomelj, 2022-04-10:
The default `platformio` partition table for 4 MB ESP32 devices:

```
nvs,      data, nvs,     0x009000, 0x005000,
otadata,  data, ota,     0x00e000, 0x002000,
app0,     app,  ota_0,   0x010000, 0x1C0000,
app1,     app,  ota_1,   0x1D0000, 0x1C0000,
eeprom,   data, 0x99,    0x390000, 0x001000,
spiffs,   data, spiffs,  0x391000, 0x00F000
```

- `nvs` - non-volatile storage (key=value storage): 20 kB
- `otadata` - pointer to active application image (`app0` or `app1`): 8 kB
- `app0` - first application image partition: 1792 kB
- `app1` - second application image partition: 1792 kB
- `eeprom` - non-volatile storage, non structured: 4 kB
- `spiffs` - embedded filesystem: 60 kB
- TOTAL: 3764224 bytes

You could get rid of eeprom and spiffs partitions (nvs is used by ESPHome for storing states; otadata is used by the bootloader) and squeeze a few more kBs out of the device, but you will never be able to use 4 MB for your app when using OTA. You cannot perform in-place application image updates - i.e. OTA image overwriting the image of a running application. Program instructions are loaded and executed from flash and overwriting a running program with a new image would most probably crash your app - basically, you're overwriting a code that is currently executing with a new one.

You can, of course, use a custom partition table (see `platformio` docs) for a larger app image, but you cannot use OTA in this case.

pashdown, 2022-04-13:
> You can, of course, use a custom partition table (see `platformio` docs) for a larger app image, but you cannot use OTA in this case.

Thanks for this summary.  I should add that Safe Mode doesn't fix this.

pashdown, 2022-04-13:
Does anyone have an ESP32 module recommendation that has 8MB or more flash with ethernet that works with esphome?

closed 2022-04-13 by pashdown