esphome/issues #3211: ESP32-C3-DevKitC-02 Boot Loops?

efaden, 2022-04-10:
### The problem

Getting the following loop.... 

```
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x38c
load:0x403ce000,len:0x6a8
load:0x403d0000,len:0x236c
entry 0x403ce000
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x403d0f88
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x38c
load:0x403ce000,len:0x6a8
load:0x403d0000,len:0x236c
entry 0x403ce000
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x403d0f88
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x38c
load:0x403ce000,len:0x6a8
load:0x403d0000,len:0x236c
entry 0x403ce000
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x403d0f88
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x38c
load:0x403ce000,len:0x6a8
load:0x403d0000,len:0x236c
entry 0x403ce000
Terminal disconnected: NetworkError: The device has been lost.
```

### Which version of ESPHome has the issue?

2022.3.2

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Board

ESP32-C3-DevKitC-02

### Example YAML snippet

```yaml
substitutions:
  name: testc3-test
  friendly_name: Test C3
  device_description: Test C3 DevKit.
  reboot_timeout: 14d
  update_interval: 10min
  ip_address: 10.0.10.102
esphome:
  platformio_options:
    board_build.flash_mode: dio
    platform_packages:
    - framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#2.0.2
    platform:
    - https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream
  name: esph-testc3-test
  comment: Test C3 DevKit.
  build_path: .esphome/build/esph-testc3-test
  includes: []
  libraries: []
  name_add_mac_suffix: false
esp32:
  board: esp32-c3-devkitm-1
  framework:
    version: 1.0.6
    source: ~3.10006.0
    platform_version: platformio/espressif32 @ 3.5.0
    type: arduino
  variant: ESP32C3
external_components:
- source:
    url: https://github.com/myhomeiot/esphome-components.git
    type: git
  refresh: 1d
  components: all
esp32_ble_tracker:
  scan_parameters:
    interval: 320ms
    window: 120ms
    duration: 60s
    active: true
ble_gateway:
  devices:
  - mac_address: A4:C1:38:50:69:6C
  - mac_address: 49:42:08:00:5B:09
  - mac_address: '49:42:07:00:29:24'
  - mac_address: 49:42:07:00:2E:00
  - mac_address: 49:42:07:00:27:C1
  on_ble_advertise:
  - then:
    - homeassistant.service:
        service: ble_monitor.parse_data
        data:
          packet: !lambda |-
            return packet;
          gateway_id: testc3-test
        data_template: {}
        variables: {}
sensor:
- platform: uptime
  name: testc3-test Uptime Sensor
  id: uptime_sensor
  update_interval: 60s
  on_raw_value:
  - then:
    - text_sensor.template.publish:
        id: uptime_human
        state: !lambda |-
          int seconds = round(id(uptime_sensor).raw_state);
          int days = seconds / (24 * 3600);
          seconds = seconds % (24 * 3600);
          int hours = seconds / 3600;
          seconds = seconds % 3600;
          int minutes = seconds /  60;
          seconds = seconds % 60;
          return (
            (days ? to_string(days) + "d " : "") +
            (hours ? to_string(hours) + "h " : "") +
            (minutes ? to_string(minutes) + "m " : "") +
            (to_string(seconds) + "s")
          ).c_str();
  disabled_by_default: false
  force_update: false
  unit_of_measurement: s
  icon: mdi:timer-outline
  accuracy_decimals: 0
  state_class: total_increasing
  entity_category: diagnostic
- platform: wifi_signal
  name: testc3-test WiFi Signal
  update_interval: 10min
  disabled_by_default: false
  force_update: false
  unit_of_measurement: dBm
  accuracy_decimals: 0
  device_class: signal_strength
  state_class: measurement
  entity_category: diagnostic
binary_sensor:
- platform: status
  name: testc3-test Status
  id: system_status
  disabled_by_default: false
  entity_category: diagnostic
  device_class: connectivity
text_sensor:
- platform: template
  name: testc3-test Uptime Human Readable
  id: uptime_human
  icon: mdi:clock-start
  disabled_by_default: false
  update_interval: 60s
- platform: wifi_info
  ip_address:
    name: testc3-test IP Address
    disabled_by_default: false
    entity_category: diagnostic
    update_interval: 1s
  ssid:
    name: testc3-test SSID
    disabled_by_default: false
    entity_category: diagnostic
    update_interval: 1s
  bssid:
    name: testc3-test BSSID
    disabled_by_default: false
    entity_category: diagnostic
    update_interval: 1s
  mac_address:
    name: testc3-test Connected MAC
    disabled_by_default: false
    entity_category: diagnostic
- platform: version
  name: testc3-test ESPHome Version
  disabled_by_default: false
  icon: mdi:new-box
  entity_category: diagnostic
  hide_timestamp: false
switch:
- platform: restart
  name: testc3-test Restart
  disabled_by_default: false
  icon: mdi:restart
  entity_category: config
ota:
  password: !secret 'ota_password'
  safe_mode: true
  port: 3232
  reboot_timeout: 5min
  num_attempts: 10
time:
- platform: homeassistant
  id: homeassistant_time
  timezone: EST5EDT,M3.2.0,M11.1.0
  update_interval: 15min
- platform: sntp
  id: sntp_time
  timezone: EST5EDT,M3.2.0,M11.1.0
  update_interval: 15min
  servers:
  - 0.pool.ntp.org
  - 1.pool.ntp.org
  - 2.pool.ntp.org
wifi:
  reboot_timeout: 14d
  ap:
    ssid: !secret 'wifi_ssid'
    password: !secret 'wifi_password'
    ap_timeout: 1min
  manual_ip:
    static_ip: 10.0.10.102
    gateway: 10.0.10.1
    subnet: 255.255.255.0
    dns1: 0.0.0.0
    dns2: 0.0.0.0
  domain: .local
  power_save_mode: LIGHT
  fast_connect: false
  networks:
  - ssid: !secret 'wifi_ssid'
    password: !secret 'wifi_password'
    priority: 0.0
  use_address: 10.0.10.102
captive_portal: {}
api:
  reboot_timeout: 14d
  password: !secret 'api_password'
  port: 6053
web_server:
  auth:
    username: admin
    password: !secret 'wifi_password'
  local: true
  port: 80
  version: 2
  include_internal: false
  ota: true
  css_url: ''
  js_url: https://oi.esphome.io/v2/www.js
logger:
  level: INFO
  baud_rate: 115200
  tx_buffer_size: 512
  deassert_rts_dtr: false
  hardware_uart: UART0
  logs: {}
light:
- platform: neopixelbus
  type: GRB
  variant: ws2812
  pin: 8
  num_leds: 1
  name: NeoPixel Light
  disabled_by_default: false
  restore_mode: RESTORE_DEFAULT_OFF
  gamma_correct: 2.8
  default_transition_length: 1s
  flash_transition_length: 0s
  invert: false
  method:
    channel: 1
    type: esp32_rmt
```

cg6, 2022-04-12:
"framework-arduinoespressif32"

Last I knew only the espressif IDF was supported on the C3 - not the Arduino framework.

Can you provide more detail?
Does initial setup of the device work?

efaden, 2022-04-12:
It's "supported" if you you pull the latest.

Interestingly it works if I upload it directly from the ESPHome computer via USB.  It doesn't work, however, through the web.  The firmware also has other issues though.  The BLE and WIFI conflict and eventually the board becomes unresponsive to the web.

efaden, 2022-04-12:
See: https://github.com/esphome/issues/issues/2946

closed 2022-08-25 by github-actions[bot]