esphome/issues #3212: Setting global in display lambda causes hang/restart

timn, 2022-04-10:
### The problem

**Context**
I have a [setup that drives an ili9341 display](https://github.com/timn/esphome-busydisp). It has a global variable `busy_state` set by a template switch. A `state_processed` global variable makes sure that the display is only updated when needed (busy_state has changed). The reason is that the lambda draws images which takes time. Doing this every time is less efficient and makes the monochromatic background light fading judder.

**Issue**
Setting the `state_processed` global variable in the display lambda makes the process hang, the task watchdog for the loopTask kicks in and reboots the board.

I could trace that the issue did not occur with 2021.11.4 (the one I used to develop the feature). But in 2021.12.0b1 it broke and it shows the described behavior. Every following version up to the current release suffers from this regression.

### Which version of ESPHome has the issue?

2021.12.0 and later

### What type of installation are you using?

Docker

### What platform are you using?

ESP32

### Board

AZDelivery NodeMCU w/ ESP32-WROOM-32.

### Component causing the issue

ili9341, but more likely something in core broke

### Example YAML snippet

```yaml
esp32:
  framework:
    type: arduino
  board: nodemcu-32s

esphome:
  name: busydisp-office

wifi:
  networks:
    - ssid: !secret wifi_ssid
      password: !secret wifi_password

# Enable logging
logger:

# Enable Home Assistant API
api:
  id: api_server
  encryption:
    key: !secret api_noise_key

ota:
  password: !secret ota_password

globals:
  - id: busy_state
    type: bool
    restore_value: no
    initial_value: "false"
  - id: state_processed
    type: bool
    restore_value: no
    initial_value: "false"

switch:
  - platform: template
    name: "Busy"
    id: busy_switch
    lambda: |-
      return id(busy_state);
    turn_on_action:
      - globals.set:
          id: busy_state
          value: "true"
      - globals.set:
          id: state_processed
          value: "false"
    turn_off_action:
      - globals.set:
          id: busy_state
          value: "false"
      - globals.set:
          id: state_processed
          value: "false"

spi:
  clk_pin: GPIO18
  mosi_pin: GPIO23
  miso_pin: GPIO19

display:
  - platform: ili9341
    model: "TFT 2.4"
    cs_pin: GPIO5
    dc_pin: GPIO4
    reset_pin: GPIO22
    auto_clear_enabled: false
    rotation: 90
    lambda: |-
      if (!id(state_processed)) {
        it.fill(Color::WHITE);
        id(state_processed) = true;
      }
```

### Anything in the logs that might be useful for us?

```txt
19:48:14][D][api.connection:771]: Home Assistant 2022.4.1 (x.x.x.x): Connected successfully
[19:48:21]E (13149) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
[19:48:21]E (13149) task_wdt:  - loopTask (CPU 1)
[19:48:21]E (13149) task_wdt: Tasks currently running:
[19:48:21]E (13149) task_wdt: CPU 0: IDLE0
[19:48:22]E (13149) task_wdt: CPU 1: loopTask
[19:48:22]E (13149) task_wdt: Aborting.
[19:48:22]abort() was called at PC 0x4014dec8 on core 0
[19:48:22]
[19:48:22]ELF file SHA256: 0000000000000000
[19:48:22]
[19:48:22]Backtrace: 0x40088c18:0x3ffbf870 0x40088e95:0x3ffbf890 0x4014dec8:0x3ffbf8b0 0x4008732d:0x3ffbf8d0 0x4016a91f:0x3ffbc100 0x4014f7d3:0x3ffbc120 0x4008b655:0x3ffbc140 0x40089ea6:0x3ffbc160
WARNING Found stack trace! Trying to decode it
WARNING Decoded 0x40088c18: invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:715
WARNING Decoded 0x40088e95: abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:715
WARNING Decoded 0x4014dec8: task_wdt_isr at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/task_wdt.c:252
WARNING Decoded 0x4008732d: _xt_lowint1 at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/xtensa_vectors.S:1154
WARNING Decoded 0x4016a91f: esp_pm_impl_waiti at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/pm_esp32.c:492
WARNING Decoded 0x4014f7d3: esp_vApplicationIdleHook at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/freertos_hooks.c:108
WARNING Decoded 0x4008b655: prvIdleTask at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c:3507
WARNING Decoded 0x40089ea6: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)
[19:48:22]
[19:48:22]Rebooting...
[19:48:22]ets Jun  8 2016 00:22:57
[19:48:22]
[19:48:22]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[19:48:22]configsip: 0, SPIWP:0xee
[19:48:22]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[19:48:22]mode:DIO, clock div:2
[19:48:22]load:0x3fff0018,len:4
[19:48:22]load:0x3fff001c,len:1044
[19:48:22]load:0x40078000,len:10124
[19:48:22]load:0x40080400,len:5828
[19:48:22]entry 0x400806a8
[19:48:22][E][esp32-hal-misc.c:111] disableCore1WDT(): Failed to remove Core 1 IDLE task from WDT
```

### Additional information

It seems that this causes a hang when setting the RTC value for the global variable, my suspicion is some kind of deadlock. But that's where I had to leave that rabbit hole.

timn, 2022-04-10:
I did chase a bit more down the rabbit hole and ran a bisect from 2021.11.4 to 2021.12.0b1. This shows that [5410617 ](https://github.com/esphome/esphome/commit/54106179a11ec9ff4586f73f8246507165ba9390) (esphome/esphome#2846) is causing the issue, which from the sound of it seems plausible. But I don't know enough details to see why.

@oxan does this ring a bell for you why what I'm trying to do is problematic with the changed watchdog?

timn, 2022-04-19:
@jesserockz do you have some idea how this can be solved (you reviewed that PR)?

timn, 2022-07-03:
@oxan @jesserockz do you have some idea how to fix this?

ssieb, 2022-11-03:
Can you try again with the latest release?

timn, 2023-03-01:
Tried with 2023.2.4 and the issue appears to be fixed (works as expected).

closed 2023-03-01 by timn