esphome/issues #3215: Module TCA9548A(.cpp)

jmBrosch, 2022-04-12:
### The problem

The TCA9548A does not select channels properly. 

Reason: tca9548a.cpp - line44:   '**auto err = this->write_register(0x70, &channel_val, 1);**' needs to be replaced with: '**auto err = this->write(&channel_val, 1);**'

The TCA9548A does not have a register to address. The channel_mask is written to the to the chip address (default 0x70).
Function '**write_register(0x70...)**' in '**switch_to_channel(channel)**' causes  a write command _0x70 TX 0x70 channel_mask_ 
 (2 bytes) but it should be _0x70 TX channel_mask_ (1 byte).
Best regards
  Juergen

### Which version of ESPHome has the issue?

V2022.3.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

any

### What platform are you using?

ESP32

### Board

ESP32 D1 Mini

### Component causing the issue

TCA9548A I2C Multiplexer

### Example YAML snippet

```yaml
tca9548a:
  - address: 0x70
    id: multiplex0
    i2c_id: i2c0
    channels:
      - bus_id: multiplex0channel0
        channel: 0
      - bus_id: multiplex0channel1
        channel: 1
.
.
.
```

### Additional information

The TCA9548A does not have a register to address but only to write the channel mask to the chip address (default 0x70).
As it is right now they 'switch_to_channel (channel)' function causes  a write command _0x70 TX 0x70 (channel_mask)_ instead of _0x70 TX (channel_mask)_.

jmBrosch, 2022-04-15:
> 

Sorry I am travelling for a week and cannot verify it on the target before I return. As soon as I am back I will test it.

jmBrosch, 2022-04-24:
Hi, 
it throws errors when compiling with the external component (while it compiles without external component) - see attachment.
Regards
  Juergen
[config.yaml.txt](https://github.com/esphome/issues/files/8549328/config.yaml.txt)
[image]

martgras, 2022-04-24:
Hm these errors are in code areas I didn't touch. 
I'm on the dev branch and your yaml compiles for me. 
Did you try ```esphome clean <yaml>``` after switching to the external component?

jmBrosch, 2022-04-25:
Hi,
diving a bit deeper the problem seems to be that the tca9548aChannel class defines:
**_i2c::ErrorCode writev(uint8_t address, i2c::WriteBuffer *buffers, size_t cnt, bool stop) override;_**
with 4 parameters while the base-class "I2CBus" defines it with 3 parameters:
**_virtual ErrorCode writev(uint8_t address, WriteBuffer *buffers, size_t cnt) = 0;_**
Removing **"bool stop"** from the tca9548aChannel class function definition and the function call in line 19:
**_...return parent_->bus_->writev(address, buffers, cnt, stop);_**
should fix this.

jmBrosch, 2022-04-25:
With the files attached for the TCA9548a component it was compiling and running on my target without problems
[tca9548a.zip](https://github.com/esphome/issues/files/8551797/tca9548a.zip)

martgras, 2022-04-25:
Ah ok
https://github.com/esphome/esphome/pull/3370 changed the signature of writev and since your are on 2022.03 you don’t have that change yet.
So the change works – going to submit it

jmBrosch, 2022-04-25:
ok, thanks for your support.

closed 2022-05-10 by jesserockz