esphome/issues #3216: i2c bus boot problem on nodemcu-32s

xuxiaohe0308, 2022-04-13:
### The problem

Hi
I am using ESP32 (nodemcu-32s) and TSL2561 through i2c, and trying to use MQTT to publish sensor data.
When I flash my ESP32 it seems to be stuck in some kind of dead loop (which is shown in the log below). Some important messages are like 
` Performing I2C bus recovery [21:31:02]E (10192) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:`
I think it is i2c's problem because when I delete i2c and TSL2561 configuration in the yaml file it can boot correctly. 
Also I try another BME280 sensor and the problem still exists.
Can you solve my problem?

### Which version of ESPHome has the issue?

2022.3.2

### What type of installation are you using?

pip

### What platform are you using?

ESP32

### Board

nodemcu-32s

### Component causing the issue

i2c

### Example YAML snippet

```yaml
esphome:
  name: esp32-1

esp32:
  board: nodemcu-32s

#Enable logging
logger:

#Enable Home Assistant API
# api:

#ota:
#  password: "****************************"

wifi:
  ssid: ****
  password: ********

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esp32-1 Fallback Hotspot"
    password: "cNBtFYNhEpnm"

#captive_portal:

i2c:
    sda: GPIO21
    scl: GPIO22
    scan: true
    id: bus_a
  
sensor:
#  - platform: bme280
#    temperature:
#      name: "BME280 Temperature"
#      oversampling: 16x
[... 262 lines cut ...]
[21:31:07]E (10192) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
[21:31:07]E (10192) task_wdt:  - loopTask (CPU 1)
[21:31:07]E (10192) task_wdt: Tasks currently running:
[21:31:07]E (10192) task_wdt: CPU 0: IDLE0
[21:31:07]E (10192) task_wdt: CPU 1: IDLE1
[21:31:07]E (10192) task_wdt: Aborting.
[21:31:07]abort() was called at PC 0x4014360c on core 0
[21:31:07]
[21:31:07]ELF file SHA256: 0000000000000000
[21:31:07]
[21:31:07]Backtrace: 0x40088efc:0x3ffbf870 0x40089179:0x3ffbf890 0x4014360c:0x3ffbf8b0 0x40087785:0x3ffbf8d0 0x4015901f:0x3ffbc160 0x40144f17:0x3ffbc180 0x4008b949:0x3ffbc1a0 0x4008a18a:0x3ffbc1c0
WARNING Found stack trace! Trying to decode it
WARNING Decoded 0x40088efc: invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:715
WARNING Decoded 0x40089179: abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:715
WARNING Decoded 0x4014360c: task_wdt_isr at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/task_wdt.c:252
WARNING Decoded 0x40087785: _xt_lowint1 at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/xtensa_vectors.S:1154
WARNING Decoded 0x4015901f: esp_pm_impl_waiti at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/pm_esp32.c:492
WARNING Decoded 0x40144f17: esp_vApplicationIdleHook at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/freertos_hooks.c:108
WARNING Decoded 0x4008b949: prvIdleTask at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c:3507
WARNING Decoded 0x4008a18a: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)
[21:31:08]
[21:31:08]Rebooting...
...
...
```

velaar, 2022-08-08:
Just reproduced this exact problem. Did you have a chance to test if it's i2c pin-dependent?

branch14, 2022-08-14:
I have the same issue with TSL2591.

It does not seem to be pin related.

Checked with:

```
i2c:
  sda: 16
  scl: 17
```

If I remove either SDA or SCL connection to  TSL2591 the device boots fine with:

```
[19:06:22][C][i2c.arduino:038]: I2C Bus:
[19:06:22][C][i2c.arduino:039]:   SDA Pin: GPIO16
[19:06:22][C][i2c.arduino:040]:   SCL Pin: GPIO17
[19:06:22][C][i2c.arduino:041]:   Frequency: 50000 Hz
[19:06:22][C][i2c.arduino:044]:   Recovery: bus successfully recovered
```

If TSL2591 is fully connected the device crashes here:

```
[19:05:40][I][i2c.arduino:161]: Performing I2C bus recovery
[19:05:45]E (10230) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
```

I also tried to connect a CCS811. But the device is not found on the i2c bus.

My controller is a "Goouuu-ESP32" (ESP-WROOM-32, nodemcu-32). Power via USB does suffice, right? Just checking every assumption.

dmpolukhin, 2022-08-20:
I had similar issue and resolved it by checking sensor address and wiring. I tried to use power supply from GPIO pins but it looks like esphome initializes them after I2C and if anything is wrong with I2C it enters in reboot loop.

xuxiaohe0308, 2022-08-20:
Thanks for everyone. I solved this problem by using shorter wires, so maybe this is a wire quality problem. But it's weird that I use the same wires on i2c on RaspberryPi and it works normally.

velaar, 2022-08-20:
Thank you! I’ve traced the problem to a faulty SCD30 sensor. Replacing the sensor fixed the problem.  We might want to document this somewhere.