esphome/issues #3219: Exceeding an unreported resource limit on ESP32 - Boot fails

mr-sneezy, 2022-04-14:
### The problem

In brief. 
ESPhome flashed ESP32 board fails to boot after loading code. New code size is an increment of previously used code blocks that does boot and run on the target board fine.
ESPhome validates this size code OK. 
No warnings about resource limits is given.
Note: The code uses a LOT of Template Sensors and Text Sensors.

In detail.
I have a LARGE ESPhome program that creates an 8 zone water system controller with internal scheduler (>1200 lines) that seems to work fine when reducing the code size by 2 zones to 6 zones [(based on this project)](https://community.home-assistant.io/t/esphome-diy-irrigation-controller-with-internal-scheduler/171844). 
Other than adding 2 extra repeat sets of code blocks to the 6 zone code there is no changes (other than correct naming of the extra zone entities etc).
I have no low level coding knowledge to add much insight. There is a some rapidly repeated serial log that may indicate a cause to someone. 
Looking for an explanation of cause or a work around to be able to complete the 8 zone controller...

### Which version of ESPHome has the issue?

Current version: 2022.3.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

core-2022.4.1

### What platform are you using?

ESP32

### Board

KC868-A8 ESP32 relay board

### Component causing the issue

Unknown

### Anything in the logs that might be useful for us?

```txt
Compiler messages.

INFO Generating C++ source...
INFO Compiling app...
Processing irrigation-controller (board: nodemcu-32s; framework: arduino; platform: platformio/espressif32 @ 3.5.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- <WiFi> 1.0
|-- <ESPmDNS> 1.0
|-- <Update> 1.0
|-- <Wire> 1.0.1
Compiling /data/irrigation-controller/.pioenvs/irrigation-controller/src/main.cpp.o
Linking /data/irrigation-controller/.pioenvs/irrigation-controller/firmware.elf
RAM:   [=         ]  13.1% (used 42896 bytes from 327680 bytes)
Flash: [=====     ]  51.5% (used 945026 bytes from 1835008 bytes)
Building /data/irrigation-controller/.pioenvs/irrigation-controller/firmware.bin
esp32_create_combined_bin(["/data/irrigation-controller/.pioenvs/irrigation-controller/firmware.bin"], ["/data/irrigation-controller/.pioenvs/irrigation-controller/firmware.elf"])
Wrote 0xf6bf0 bytes to file /data/irrigation-controller/.pioenvs/irrigation-controller/firmware-factory.bin, ready to flash to offset 0x0
========================= [SUCCESS] Took 59.75 seconds =========================
INFO Successfully compiled program.
INFO Resolving IP address of irrigation-controller.local
INFO  -> 192.168.0.27
INFO Uploading /data/irrigation-controller/.pioenvs/irrigation-controller/firmware.bin (945136 bytes)
Uploading: [============================================================] 100% Done...

INFO Waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from irrigation-controller.local using esphome API
WARNING Can't connect to ESPHome API for irrigation-controller.local: Error resolving IP address: [Errno -2] Name or service not known
INFO Trying to reconnect to irrigation-controller.local in the background
```
Repetitive boot error messages in this code size.
```
Hash of data verified.

Leaving...
Hard Resetting...
Hard resetting via RTS pin...
Done! Flashing is complete!

Showing logs:
[19:11:58][I][logger:214]: Log initialized
[19:11:58][C][ota:461]: There have been 0 suspected unsuccessful boot attempts.
[19:11:58][D][esp32.preferences:114]: Saving preferences to flash...
[19:11:58][I][app:029]: Running through setup()...
[19:11:58][I][i2c.arduino:161]: Performing I2C bus recovery
[19:11:58][C][pcf8574:010]: Setting up PCF8574...
[19:11:58][C][pcf8574:010]: Setting up PCF8574...
[19:11:58][C][switch.gpio:011]: Setting up GPIO Switch 'irrigation_channel_1'...
[19:11:58][D][switch:017]: 'irrigation_channel_1' Turning OFF.
[19:11:58]Guru Meditation Error: Core  1 panic'ed (Unhandled debug exception)
[19:11:58]Debug exception reason: Stack canary watchpoint triggered (loopTask) 
[19:11:58]Core 1 register dump:
[19:11:58]PC      : 0x4008cfb3  PS      : 0x00060736  A0      : 0x4008c9ae  A1      : 0x3ffafff0  
[19:11:58]A2      : 0x3ffb0060  A3      : 0x3ff7b44c  A4      : 0x00000001  A5      : 0x00000000  
[19:11:58]A6      : 0xffffffff  A7      : 0x00000004  A8      : 0x00000003  A9      : 0x4008ca9c  
[19:11:58]A10     : 0x3ffb00c0  A11     : 0x00000000  A12     : 0x4008772c  A13     : 0x00003000  
[19:11:58]A14     : 0x00000002  A15     : 0x3ffb02e4  SAR     : 0x0000001d  EXCCAUSE: 0x00000001  
[19:11:58]EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffe  
[19:11:58]
[19:11:58]ELF file SHA256: 0000000000000000
[19:11:58]
[19:11:58]Backtrace: 0x4008cfb3:0x3ffafff0 0x4008c9ab:0x00000000
[19:11:58]
[19:11:58]Rebooting...
[19:11:59][I][logger:214]: Log initialized
[19:11:59][C][ota:461]: There have been 1 suspected unsuccessful boot attempts.
[19:11:59][D][esp32.preferences:114]: Saving preferences to flash...
[19:11:59][I][app:029]: Running through setup()...
[19:11:59]Guru Meditation Error: Core  1 panic'ed (Unhandled debug exception)
[19:11:59]Debug exception reason: Stack canary watchpoint triggered (loopTask) 
[19:11:59]Core 1 register dump:
[19:11:59]PC      : 0x40087711  PS      : 0x00060336  A0      : 0x00060330  A1      : 0x3ffb0070  
[19:11:59]A2      : 0x3f401c95  A3      : 0x3f401c91  A4      : 0x000000ff  A5      : 0x0000ff00  
[19:11:59]A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x00000000  A9      : 0x00000001  
[19:11:59]A10     : 0x3ffb830c  A11     : 0x3f403b7e  A12     : 0x3ffb830d  A13     : 0x000000a1  
[19:11:59]A14     : 0x3f401dd1  A15     : 0x00000000  SAR     : 0x00000004  EXCCAUSE: 0x00000001  
[19:11:59]EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffe  
[19:11:59]
[19:11:59]ELF file SHA256: 0000000000000000
[19:11:59]
[19:11:59]Backtrace: 0x40087711:0x3ffb0070 0x0006032d:0x3ffb0140 0x40114f02:0x3ffb0450 0x40114f3e:0x3ffb04e0 0x400d85b1:0x3ffb0520 0x400d85e5:0x3ffb0550 0x400d8619:0x3ffb05a0 0x400d87d9:0x3ffb05c0 0x400e0afd:0x3ffb05e0 0x400e0b2a:0x3ffb0610 0x400d826c:0x3ffb0670 0x400d846d:0x3ffb0690 0x400d84e6:0x3ffb06b0 0x4016a6d5:0x3ffb06d0 0x4016a7a1:0x3ffb06f0 0x400dff51:0x3ffb0710 0x400ed6ee:0x3ffb0760 0x400f285a:0x3ffb1fb0 0x4008a37e:0x3ffb1fd0
[19:11:59]
[19:11:59]Rebooting...
```
```

### Additional information

Main project YAML is available here in a 7 zone version size (where the booting trouble begins).
https://github.com/mr-sneezy/ESP32_irrigation/blob/3ba699ad0f9847e6214d9d915221efa256cf1795/irrigation.yaml

martgras, 2022-04-14:
You are running out of stack space in loop task
Please try if these settings make a difference for you

```yaml
esphome:
   ...
    build_flags: -DCONFIG_ARDUINO_LOOP_STACK_SIZE=16384  
    platform_packages:
      - framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git
    platform:
      - https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream  
```

mr-sneezy, 2022-04-22:
Thanks for the tip. I found some other threads too once I had you code block.

I've hit this issue now though with these additions so still not 'out of the woods' with it yet.

```
INFO Generating C++ source...
INFO Compiling app...
Processing irrigation-controller (board: nodemcu-32s; framework: arduino; platform: https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream)
--------------------------------------------------------------------------------
Configuring from remote
* Adding toolchain toolchain-riscv32-esp with version 8.4.0+2021r2
* Adding toolchain toolchain-xtensa-esp32 with version 8.4.0+2021r2
* Adding toolchain toolchain-xtensa-esp32s2 with version 8.4.0+2021r2
Tool Manager: Installing espressif/toolchain-xtensa-esp32 @ 8.4.0+2021r2
Error: Could not find the package with 'espressif/toolchain-xtensa-esp32 @ 8.4.0+2021r2' requirements for your system 'linux_aarch64'
```

mr-sneezy, 2022-04-23:
> You are running out of stack space in loop task Please try if these settings make a difference for you
>
> ```yaml
> esphome:
>    ...
>     build_flags: -DCONFIG_ARDUINO_LOOP_STACK_SIZE=16384 [...]

mr-sneezy, 2022-04-23:
I used a work around for the issue with ESPhome installing the required packages on HASSOS and RPi.
I installed python and ESPhome on my Windows 10 PC and compiled from that using this modified header for esphome: and esp32:

`# 8 Zone Version
substitutions:
  devicename: irrigation-controller

esphome:
  name: $devicename
  platformio_options:
    build_flags: 
      - -DCONFIG_ARDUINO_LOOP_STACK_SIZE=16384
  includes:
    - irrigation.h

esp32:
  board: nodemcu-32s
  variant: esp32
  framework:
    type: arduino
    version: 2.0.2
    source: https://github.com/espressif/arduino-esp32.git#2.0.2
    platform_version: https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream`

closed 2022-09-15 by github-actions[bot]