esphome/issues #3220: D1 mini fails OTA in current configuration

cbhiii, 2022-04-14:
### The problem

While testing an ESP8266 D1 mini with a ST7735 display I noticed that I was unable to perform wireless updates.

This problem is reproducible in my current configuration.

Please let me know what other detail you need. I'll do my best to provide.

### Which version of ESPHome has the issue?

2022.3.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.4.3

### What platform are you using?

ESP8266

### Board

d1 mini (esp8266mod)

### Component causing the issue

ota

### Example YAML snippet

```yaml
Config file used on the d1 mini

esphome:
  name: test-bed
  platform: ESP8266
  board: d1_mini

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:
  password: 

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Test-Bed Fallback Hotspot"
    password: 

captive_portal:

color:
  - id: my_red
    red: 100%
    green: 0%
    blue: 0%
  - id: my_yellow
    red: 100%
    green: 100%
[... 504 lines cut ...]
Dependency Graph
|-- <ESPAsyncTCP-esphome> 1.2.3
|-- <ESPAsyncWebServer-esphome> 2.1.0
|   |-- <ESPAsyncTCP-esphome> 1.2.3
|   |-- <Hash> 1.0
|   |-- <ESP8266WiFi> 1.0
|-- <DNSServer> 1.1.1
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS> 1.2
|-- <SPI> 1.0
Compiling /data/test-bed/.pioenvs/test-bed/src/main.cpp.o
Linking /data/test-bed/.pioenvs/test-bed/firmware.elf
RAM:   [=====     ]  48.1% (used 39396 bytes from 81920 bytes)
Flash: [=====     ]  49.5% (used 517345 bytes from 1044464 bytes)
Building /data/test-bed/.pioenvs/test-bed/firmware.bin
esp8266_copy_factory_bin(["/data/test-bed/.pioenvs/test-bed/firmware.bin"], ["/data/test-bed/.pioenvs/test-bed/firmware.elf"])
========================= [SUCCESS] Took 12.38 seconds =========================
INFO Successfully compiled program.
INFO Resolving IP address of test-bed.local
INFO  -> 192.168.32.238
INFO Uploading /data/test-bed/.pioenvs/test-bed/firmware.bin (521504 bytes)
INFO Compressed to 366299 bytes
Uploading: [=========                                                   ] 15% 
ERROR Error sending data: timed out
```

sumirati, 2022-04-27:
Hi,

Have you retried with 2022.4.0?
I had a crash with 2022.03.1 and D1 mini, but different sensors - and different exception

After upgrading to the newer version, it worked.

LorenzThomas1961, 2022-07-15:
I have the same issue with D1-Mini and also with NodeMCU as soon as I include the ST7735 as described. For me it looks like a stack overflow when the ST7735 is used. When the ST7735 is excluded everything is working normally. I'm running on HA version 2022.7.5 from today, But this problem seems to be already minimum one week old. My Log-files look exactly the same! After including the ST7735, no (stable) WiFi-connection possible.