esphome/issues #3224: Address already in use

MelvinSnijders, 2022-04-16:
### The problem

I'm getting a address already in use error, but I'm not really sure where it comes from.

### Which version of ESPHome has the issue?

v2022.3.1

### What type of installation are you using?

Docker

### What platform are you using?

ESP8266

### Board

nodemcu

### Example YAML snippet

```yaml
esphome:
  name: gate

esp8266:
  board: esp01_1m

# Enable logging
logger:

ota:
  password: "password"
  
api:
  
switch:
  - platform: gpio
    pin: 13
    id: relay
    name: "Deel hek"
    icon: "mdi:gate"
    on_turn_on:
    - delay: 500ms
    - switch.turn_off: relay
    
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  use_address: 10.0.0.4

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Gate Fallback Hotspot"
    password: "password"

captive_portal:
```

### Anything in the logs that might be useful for us?

```txt
INFO Reading configuration /config/gate.yml...
INFO Starting log output from 10.0.0.4 using esphome API
Traceback (most recent call last):
  File "/usr/local/bin/esphome", line 33, in <module>
    sys.exit(load_entry_point('esphome', 'console_scripts', 'esphome')())
  File "/esphome/esphome/__main__.py", line 825, in main
    return run_esphome(sys.argv)
  File "/esphome/esphome/__main__.py", line 812, in run_esphome
    rc = POST_CONFIG_ACTIONS[args.command](args, config)
  File "/esphome/esphome/__main__.py", line 362, in command_logs
    return show_logs(config, args, port)
  File "/esphome/esphome/__main__.py", line 284, in show_logs
    return run_logs(config, port)
  File "/esphome/esphome/components/api/client.py", line 71, in run_logs
    asyncio.run(async_run_logs(config, address))
  File "/usr/lib/python3.9/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/usr/lib/python3.9/asyncio/base_events.py", line 642, in run_until_complete
    return future.result()
  File "/esphome/esphome/components/api/client.py", line 53, in async_run_logs
    zc = zeroconf.Zeroconf()
  File "/usr/local/lib/python3.9/dist-packages/zeroconf/_core.py", line 448, in __init__
    listen_socket, respond_sockets = create_sockets(interfaces, unicast, ip_version, apple_p2p=apple_p2p)
  File "/usr/local/lib/python3.9/dist-packages/zeroconf/_utils/net.py", line 358, in create_sockets
    listen_socket = new_socket(ip_version=ip_version, apple_p2p=apple_p2p, bind_addr=('',))
  File "/usr/local/lib/python3.9/dist-packages/zeroconf/_utils/net.py", line 248, in new_socket
    s.bind(bind_tup)
OSError: [Errno 98] Address already in use
```

netzbus, 2022-04-27:
I have the same issue. Did you find some solution?

MelvinSnijders, 2022-04-27:
No I didn't find a solution. Ended up switching from TrueNAS to just running Ubuntu with ZFS and docker containers.

usernein, 2022-08-22:
I'm facing the same issue. It happens when i try to read the logs wirelessly. The other ways works fine

board: esp01_1m
v2022.6.3

[image]

ssieb, 2022-08-26:
This is not an esphome issue.  It's a configuration problem on your system.  Something else is holding the mdns port.

closed 2022-08-26 by ssieb