esphome/issues #3225: Nextion display not working unless logger is VERY_VERBOSE on ESP8266

s00500, 2022-04-17:
### The problem

I am using the NodeMCU v2 devboard and trying to set text values on my nextion display. The issue is it does not work at all (no change on display), unless I set the logger to log level Verbose. If I do this it works flawlessly

I have tried pins D6/D7 and also D1/2 with baud_rate: 0
Both configs behave the same...

This is really strange, any ideas why this could be happening ?

### Which version of ESPHome has the issue?

Version: 2022.3.2

### What type of installation are you using?

pip

### What platform are you using?

ESP8266

### Board

NodeMCUv2

### Component causing the issue

nextion

### Example YAML snippet

```yaml
esphome:
  name: testdisplay
  platform: ESP8266
  board: nodemcu

wifi:
  ssid: "------"
  password: "------"
  fast_connect: true

# Enable logging
logger:
  baud_rate: 0 # Disable UART logging (pins GPIO1/3 are used for Nextion communication)
#  level: VERY_VERBOSE

# Example configuration entry
uart:
  id: uart_2
  rx_pin: D1
  tx_pin: D2
  baud_rate: 9600

display:
  - platform: nextion
    id: nextion1
    uart_id: uart_2
    update_interval: 2s
    lambda: |-
      it.set_component_text("g0", "ESPHOME!");
#      it.set_component_text("qr0", "ESPHOME!");
binary_sensor:
  - platform: nextion
    nextion_id: nextion1
    page_id: 0
    component_id: 2
    id: button1
    on_press:
      then:
        - lambda: |-
            ESP_LOGD("custom", "This is a custom debug message");
            id(nextion1).set_component_text("t0", "ESPHOME!");
```

kbx81, 2022-07-26:
would love to hear if the above-linked PR fixes this issue for you! 😄

s00500, 2022-07-27:
Cool, I can test it on the weekend I guess, thanks

s00500, 2022-07-30:
@kbx81 I can confirm this is working correctly now, thanks for the workaround =D

s00500, 2022-12-15:
Has this been merged now ? it is otherwise NOT completed, just needs a merge

closed 2026-01-22 by kbx81

s00500, 2026-01-22:
Awesome!!! So sick to see how this project is moving! Also awesome work on the .1 release ! (My old shelly 1 is now super stable again!)