esphome/issues #3231: exception occur by Setting up BME680 via BSEC...

heizie, 2022-04-20:
### The problem

I've built up a terminal which based on Wemos D1 Mini. It is connected with a BME 680 sensor, an ssd1306 OLED, two led for showing doors' states and a button (with a 10 kOhm physical pull up sensor).

Both BME 680 and ssd 1306 OLED are using the same pin for i2c connetction.

everything seems fine in a couple of days. but it will crash suddenly and the state-led blick fast (about 4-5 times per second)
I connect it to my PC and download the logs (pls see followings)

error log:
[logs.txt](https://github.com/esphome/issues/files/8519780/logs.txt)

more details:
https://heizie.github.io/posts/table_terminal/

### Which version of ESPHome has the issue?

2022.3.1

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP8266

### Board

Wemos D1 Mini

### Component causing the issue

BME 680 or i2c

### Example YAML snippet

```yaml
esphome:
  name: terminal-bedroom

esp8266:
  board: esp01_1m

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

font:
  - file: "Ubahn.ttf"
    id: normal_font
    size: 24

globals:
  - id: flag_dis
    type: int
    restore_value: no
    initial_value: '0'
    
# animation:
#   - file: "face.gif"
#     id: display_animation
#     resize: 128x64
#     type: BINARY

binary_sensor:
[... 489 lines cut ...]
tail 4
chksum 0xcc
load 0x3fff20b8, len 40, room 4 
tail 4
chksum 0xc9
csum 0xc9
v0007d3c0
~ld
[I][logger:214]: Log initialized
[C][status_led:014]: Setting up Status LED...
[C][ota:461]: There have been 1 suspected unsuccessful boot attempts.
[I][app:029]: Running through setup()...
[I][i2c.arduino:161]: Performing I2C bus recovery
[D][binary_sensor:034]: 'display button': Sending initial state ON
[C][light:036]: Setting up light 'bathroom state'...
[D][light:035]: 'bathroom state' Setting:
[D][light:040]:   Color mode: 
[C][light:036]: Setting up light 'toilet state'...
[D][light:035]: 'toilet state' Setting:
[D][light:040]:   Color mode: 
[C][bme680_bsec.sensor:016]: Setting up BME680 via BSEC...

User exception (panic/abort/assert)
.....sth similar....
```

trvrnrth, 2022-04-20:
Running that much on an ESP8266 is probaby pushing it and things seem to have been a bit worse with the more recent arduino version bump. I've not had the time to delve into things properly, but a bit of initial investigation seemed to suggest that publishing all the sensor states in one app loop iteration is tieing things up and not yielding to other tasks such as setting up wifi so that could be connected. You might have some luck [rolling back the arduino version](https://github.com/esphome/issues/issues/3083#issuecomment-1048799421) but I make no promises.

heizie, 2022-04-20:
> Running that much on an ESP8266 is probaby pushing it and things seem to have been a bit worse with the more recent arduino version bump. I've not had the time to delve into things properly, but a [...]

let me have a try, I'll test it in these days. thx

closed 2022-04-25 by heizie