esphome/issues #3232: `esp32_ble_tracker` not reconnect after IP change

javicalle, 2022-04-21:
### The problem

Not sure if this can be an ESPHome issue or HA.

Recently after a router reset my ESPHome device is no longer accessible.
Checking the traces I see that something is trying to connect to the IP 192.168.0.103, but after checking the router, the device has a new IP assigned (192.168.0.106).
From the ESPHome web UI I can connect to the device and logs show me data from device but in HA the device and entities are unavailable.

Since the connection is made by DHCP, it would be expected that the device reconnects correctly when the IP changes.

### Which version of ESPHome has the issue?

2022.3.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

core-2022.3.8

### What platform are you using?

ESP32

### Board

nodemcu-32s

### Component causing the issue

esp32_ble_tracker

### Example YAML snippet

```yaml
esphome:
  name: ble-tracker
  platform: ESP32
  board: nodemcu-32s

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

# Enable Home Assistant API
api:
  password: !secret api_password

ota:
  password: !secret ota_password

esp32_ble_tracker:

binary_sensor:
  # Presence based on MAC address
  - platform: ble_presence
    mac_address: AA:BB:CC:DD:EE:FF
    name: "miband"
```

### Anything in the logs that might be useful for us?

```txt
2022-04-21 17:39:33 WARNING (MainThread) [aioesphomeapi.reconnect_logic] Can't connect to ESPHome API for ble-tracker @ 192.168.0.103: Error connecting to ('192.168.0.103', 6053): [Errno 111] Connect call failed ('192.168.0.103', 6053)
```

### Additional information

The current configuration of the `core.config_entries` shows the following configuration:
```
             {
                 "entry_id": "2242c488ed2539b19f3c8e6c6c3540d7",
                 "Version 1,
                 "domain": "esphome",
                 "title": "ble-tracker",
                 "data": {
                     "host": "192.168.0.103",
                     "port": 6053,
                     "password": "my_easy_password"
                 },
                 "options": {},
                 "pref_disable_new_entities": false,
                 "pref_disable_polling": false,
                 "source": "user",
                 "unique_id": "ble-tracker",
                 "disabled_by": null
             },
```

javicalle, 2022-04-21:
After updating the IP in `core.config_entries` and restarting HA (doesn't work just reloading the integration) the device is available again in HA.

javicalle, 2022-06-30:
Maybe fixed with:
* https://github.com/home-assistant/core/pull/74206

ssieb, 2022-11-03:
The device doesn't connect to HA.  It's HA that connects to the device.  If you added the device by IP address, there's no way for HA to find the new address.  If you added it by mdns address, then it should work.

javicalle, 2022-11-03:
That makes sense 🤦🏻‍♂️ 

I have done a test and removing and adding it by the mdns address change the register to:
```json
    "config": {
      "entry_id": "4c5752ea7bffb529356c91cb3bee39b1",
      "version": 1,
      "domain": "esphome",
      "title": "ble-tracker",
      "data": {
        "host": "ble-tracker.local",
        "port": 6053,
        "password": "**REDACTED**",
        "noise_psk": ""
      },
```

So the next time the router changes the IP it shouldn't be a problem 👍🏻 

Thank you so much.

closed 2022-11-03 by javicalle