esphome/issues #3234: esp8266EX (e.g. Sonoff basic) OTA will result in timeout

volkerrichert, 2022-04-22:
### The problem

Hi!

After updating to 2022.2.x, i realized one updated device (sonoff basic) doesn't came back to live. I thought it was my fault and leave it as it is, because it was installed behind a mirror. After upgrading to 2022.3.2 I tried to bring it back to live.

After flashing, replacing the device and debugging a while a found a few issues.
- the esp doesn't find the name of my network during scan. The names are mostly empty. After enabling `fast-connect` it reconnect to my network.
- but updating via OTA still doesn't work in most of the times. It starts and then slows down massively until it times out. 
- Wifi connection is up and running during OTA.
- The device is "online" and ping is working with 3-8ms turnaround time without any lost
- Signal strength is at -44 dB up to -34 dB
- LOG (verbose via serial) doesn't show anything than the upload progress
- progress increases a bit after(!) esphome times out. It seems to work on some buffers
- after 1 min after esphome times out esp times out as well.
- in the logs you can see upload was "ok" until around 78% (big increase of percentage) and then slows down.
- the 89.4 & 90.5% war printed after timeout.
- in most of the times it will reach 15-20%
- LED is blinking slow even wifi works and esphome is connected

LOGs
```
␛[0;37m[V][api:114]: Removing connection to ESPHome Logs 2022.3.2 (192.168.120.136)␛[0m
␛[0;36m[D][ota:139]: Starting OTA Update from 192.168.120.136...␛[0m
␛[0;37m[V][ota:170]: OTA features is 0x01␛[0m
␛[0;37m[V][ota:253]: OTA size is 320746 bytes␛[0m
sleep disable
[begin] roundedSize:       0x0004F000 (323584)
[begin] updateEndAddress:  0x000FB000 (1028096)
[begin] currentSketchSize: 0x00072000 (466944)
[begin] _startAddress:     0x000AC000 (704512)
[begin] _currentAddress:   0x000AC000 (704512)
[begin] _size:             0x0004E4EA (320746)
␛[0;37m[V][ota:270]: Update: Binary MD5 is 23f8def74bd01184ff55ceda80dd440d␛[0m
␛[0;36m[D][ota:308]: OTA in progress: 0.3%␛[0m
␛[0;36m[D][ota:308]: OTA in progress: 16.9%␛[0m
␛[0;36m[D][ota:308]: OTA in progress: 32.1%␛[0m
␛[0;36m[D][ota:308]: OTA in progress: 47.5%␛[0m
␛[0;36m[D][ota:308]: OTA in progress: 64.0%␛[0m
␛[0;36m[D][ota:308]: OTA in progress: 78.2%␛[0m
␛[0;36m[D][ota:308]: OTA in progress: 82.0%␛[0m
␛[0;36m[D][ota:308]: OTA in progress: 83.0%␛[0m
␛[0;36m[D][ota:308]: OTA in progress: 84.3%␛[0m
␛[0;36m[D][ota:308]: OTA in progress: 85.4%␛[0m
␛[0;36m[D][ota:308]: OTA in progress: 86.8%␛[0m
␛[0;36m[D][ota:308]: OTA in progress: 88.1%␛[0m
␛[0;36m[D][ota:308]: OTA in progress: 89.4%␛[0m
␛[0;36m[D][ota:308]: OTA in progress: 90.5%␛[0m
␛[0;33m[W][ota:293]: Remote end closed connection␛[0m
premature end: res:0, pos:286720/320746
␛[0;37m[V][component:199]: Component ota took a long time for an operation (138.47 s).␛[0m
␛[0;37m[V][component:200]: Components should block for at most 20-30ms.␛[0m
```

ping during upload:

```
volker@DESKTOP-UPL0S6K:~$ ping 192.168.120.60
PING 192.168.120.60 (192.168.120.60) 56(84) bytes of data.
64 bytes from 192.168.120.60: icmp_seq=1 ttl=253 time=3.79 ms
64 bytes from 192.168.120.60: icmp_seq=2 ttl=253 time=4.46 ms
64 bytes from 192.168.120.60: icmp_seq=3 ttl=253 time=4.81 ms
64 bytes from 192.168.120.60: icmp_seq=4 ttl=253 time=3.56 ms
64 bytes from 192.168.120.60: icmp_seq=5 ttl=253 time=3.31 ms
64 bytes from 192.168.120.60: icmp_seq=6 ttl=253 time=3.79 ms
64 bytes from 192.168.120.60: icmp_seq=7 ttl=253 time=8.76 ms
64 bytes from 192.168.120.60: icmp_seq=8 ttl=253 time=6.77 ms
64 bytes from 192.168.120.60: icmp_seq=9 ttl=253 time=3.39 ms
64 bytes from 192.168.120.60: icmp_seq=10 ttl=253 time=4.17 ms
64 bytes from 192.168.120.60: icmp_seq=11 ttl=253 time=3.04 ms
64 bytes from 192.168.120.60: icmp_seq=12 ttl=253 time=5.88 ms
64 bytes from 192.168.120.60: icmp_seq=13 ttl=253 time=3.90 ms
64 bytes from 192.168.120.60: icmp_seq=14 ttl=253 time=7.16 ms
64 bytes from 192.168.120.60: icmp_seq=15 ttl=253 time=3.68 ms
64 bytes from 192.168.120.60: icmp_seq=16 ttl=253 time=3.81 ms
64 bytes from 192.168.120.60: icmp_seq=17 ttl=253 time=4.22 ms
64 bytes from 192.168.120.60: icmp_seq=18 ttl=253 time=4.20 ms
64 bytes from 192.168.120.60: icmp_seq=19 ttl=253 time=3.85 ms
64 bytes from 192.168.120.60: icmp_seq=20 ttl=253 time=4.11 ms
64 bytes from 192.168.120.60: icmp_seq=21 ttl=253 time=14.7 ms
64 bytes from 192.168.120.60: icmp_seq=22 ttl=253 time=19.9 ms
64 bytes from 192.168.120.60: icmp_seq=23 ttl=253 time=65.1 ms
64 bytes from 192.168.120.60: icmp_seq=24 ttl=253 time=119 ms
.....
--- 192.168.120.60 ping statistics ---
85 packets transmitted, 85 received, 0% packet loss, time 84142ms
rtt min/avg/max/mdev = 2.397/8.184/119.548/16.455 ms
```

I already know issue #455, but I'm not sure it matches. 

Any hint / idea?

Volker

### Which version of ESPHome has the issue?

20222.3.2

### What type of installation are you using?

Docker

### What platform are you using?

ESP8266

### Board

sonoff basic and other

### Component causing the issue

wifi

volkerrichert, 2022-04-22:
Uploadvia Web seems to work. But i haven't enabled it on my normal devices.

yanvigdev, 2022-05-26:
Hello,
I have in my possession two types of sonoff basic:
- **Sonoff basic v1.1**
[image]

- **Sonoff basic R2 v1**
[image]

Applying EspHome configuration to **Sonoff basic v1.1** was very difficult. No wifi network found. So I tested many different configurations. Now it is obvious that the problem comes from the **mqtt** configuration and the **status_led** on GPIO13. Once these two configurations are disabled, the wifi of the sonoff works normally.

For the **Sonoff basic R2 v1**, no configuration problem. The configuration of the EspHome documentation works fully.

Here are my configurations:
For **Sonoff basic v1.1**

```yaml
substitutions:
  devicename: sonoff-3
  comment: bureau sous table

esphome:
  name: $devicename
  comment: $comment

esp8266:
  board: sonoff_basic

binary_sensor:
  - platform: gpio
    pin:
      number: GPIO0
      mode:
        input: true
        pullup: true
      inverted: true
    name: "Sonoff Basic Button"
    on_press:
      - switch.toggle: relay

switch:
  - platform: gpio
    name: "Sonoff Basic Relay"
    pin: GPIO12
    id: relay

ota:
  password: !secret 'ota_password'

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  domain: .home
  ap:
    ssid: ${devicename}_ap
    password: !secret wifi_ap_password

logger:
  level: DEBUG

api:

```

For **Sonoff basic R2 v1**

```yaml
substitutions:
  devicename: sonoff-7
  comment: bureau sous table

esphome:
  name: $devicename
  comment: $comment

esp8266:
  board: sonoff_basic

mqtt:
  broker: broker.home
  topic_prefix: home/$devicename

binary_sensor:
  - platform: gpio
    pin:
      number: GPIO0
      mode:
        input: true
        pullup: true
      inverted: true
    name: "Sonoff Basic Button"
    on_press:
      - switch.toggle: relay

switch:
  - platform: gpio
    name: "Sonoff Basic Relay"
    pin: GPIO12
    id: relay

status_led:
  pin:
    number: GPIO13
    inverted: yes

ota:
  password: !secret 'ota_password'

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  domain: .home
  ap:
    ssid: ${devicename}_ap
    password: !secret wifi_ap_password

logger:
  level: DEBUG

api:

```

closed 2022-10-12 by github-actions[bot]