esphome/issues #3236: Climate component doesn't 'toggle' switches?

gvdhoven, 2022-04-23:
### The problem

I have added a dallas temperature sensor and switch to my esphome board. Both work independently and can be controlled from Home Assistant, but combining them in a thermostat or bang_bang doesn't work.

**What i want to achieve:**
Basically what i want to achieve is that the switch turns ON when the temp is above 25 degrees celcius, but it doesn’t turn on whereas i am expecting it that it should based on above logs and config?

My base config:
```
esphome:
  name: "esphome-beneden"

# Configure board it runs on
esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:
  level: DEBUG

# WiFi settings
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

# Enable captive portal (fallback hotspot) in case WiFi connection fails
  ap:
    ssid: "ESPHome beneden"
    password: "-REDACTED-"

captive_portal:

# Enable OTA updates
ota:
  password: "-REDACTED-"

# Enable Home Assistant API
api:

# Enable Dallas temperature monitoring
dallas:
  - pin: GPIO16

# Enable Bluetooth scanning for this ESP32
esp32_ble_tracker:

# Enable CRON
time:
  - platform: homeassistant
    on_time:
      - seconds: 0
        minutes: 0
        hours: 4
        days_of_week: MON-SUN
        then:
           - switch.toggle: restart_switch

# Configure switches
switch:
  - platform: gpio
    name: "Vloerverwarming pomp"
    id: vloerverwarming_pomp
    pin: GPIO17
  - platform: gpio
    name: "Onboard LED"
    pin: 2
    inverted: True
  - platform: restart
    name: "Restart toggle"
    id: restart_switch

# Configure sensors
sensor:
  # Uptime sensor
  - platform: uptime
    name: "Uptime"

  # Wifi Signal
  - platform: wifi_signal
    name: "WiFi Signal"
    update_interval: 60s

  # Xiaomi Mijia device using pvvx custom firmware and "Custom" advertising
  - platform: pvvx_mithermometer
    mac_address: "A4:C1:38:50:F2:E7"
    temperature:
      name: "Badkamer temperature"
    humidity:
      name: "Badkamer humidity"
    battery_level:
      name: "Badkamer battery"

  # Vloerverwarming
  - platform: dallas
    name: "Vloerverwarming aanvoer"
    id: vloerverwarming_aanvoer
    address: 0xd602161a580dee28
```

I have tried this:
```
# Vloerverwarming controller
climate:
  - platform: thermostat
    name: "Vloerverwarming controller"
    sensor: vloerverwarming_aanvoer
    default_mode: cool
    default_target_temperature_high: 25 °C
    min_cooling_off_time: 0s
    min_cooling_run_time: 0s
    min_idle_time: 0s
    cool_action:
      - switch.turn_on: vloerverwarming_pomp
    idle_action:
      - switch.turn_off: vloerverwarming_pomp
```

And even this:
```
# Vloerverwarming controller
climate:
  - platform: bang_bang
    name: "Vloerverwarming controller"
    sensor: vloerverwarming_aanvoer
    default_target_temperature_low: 30 °C
    default_target_temperature_high: 32 °C
    cool_action:
      - switch.turn_on: vloerverwarming_pomp
    idle_action:
      - switch.turn_off: vloerverwarming_pomp
```

But the logs still shown this (when using the thermostat method):
```
[16:54:37][D][sensor:125]: 'Vloerverwarming aanvoer': Sending state 28.50000 °C with 1 decimals of accuracy
[16:54:37][D][climate:385]: 'Vloerverwarming controller' - Sending state:
[16:54:37][D][climate:388]:   Mode: OFF
[16:54:37][D][climate:390]:   Action: OFF
[16:54:37][D][climate:408]:   Current Temperature: 28.50°C
[16:54:37][D][climate:414]:   Target Temperature: 25.00°C
```

I was expecting to see here :
MODE: ON
ACTION: COOL

(and that the switch would turn on, which doesn't)

What am i missing?

### Which version of ESPHome has the issue?

latest

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

latest

### What platform are you using?

ESP32

### Board

wemos d1

### Component causing the issue

climate

### Example YAML snippet

```yaml
See above
```

### Anything in the logs that might be useful for us?

```txt
See above
```

gvdhoven, 2022-04-23:
@kbx81 i think you might be able to assist/help me?

bkaufx, 2022-04-23:
Can you turn the climate entity on in Home Assistant?  Should basically look like the below.  That is how you turn the "Mode" you are seeing in the logs from OFF to COOL or HEAT or whatever.

[image]

gvdhoven, 2022-04-23:
Well, yes i indeed could, but i was expecting this to be 'default' on so to say. e.g. that it could operate without home assistant running and me having to switch it on once?

bkaufx, 2022-04-23:
There is a default mode setting: https://esphome.io/components/climate/thermostat.html?highlight=default_mode

kbx81, 2022-04-24:
The climate mode should be restored when the ESP resets; see [here](https://github.com/esphome/esphome/blob/757b98748b7e6d13cf21ecf379870eb4efe94fa4/esphome/components/thermostat/thermostat_climate.cpp#L28-L36).

The climate **action** will remain `off` whenever the climate **mode** is `off`; the climate **action** will not be changed/updated as long as the climate **mode** is `off`. There is no "on" mode. If you use `default_mode`, it must be one of the modes shown [here](https://esphome.io/components/climate/thermostat.html?highlight=default_mode#default-target-temperatures-and-mode). Note that `default_mode` is only applied when the mode cannot be restored from memory. If you set it to `off` at some point and reset the device, it will remain `off` at least until you change it to `cool`.

If you'd like a live example of the `thermostat` component, see my [repo](https://github.com/kbx81/esphome-configs/blob/62ad8ff9d2ad4e69b03b16aabeda80ac4ae8baa2/shared/esp_thermostat_nextion_common_config.yaml#L223-L317).

gvdhoven, 2022-04-25:
Thank you both, it was unclear from the docs that by default the component is off (e.g. the top 3 examples could be extended with the default_mode parameter)

gvdhoven, 2022-04-25:
@kbx81 created https://github.com/esphome/esphome-docs/pull/2041 but not sure if i've done it correctly.

kbx81, 2022-05-11:
@gvdhoven were you able to get to where you want to be with this? Shall we close the issue? 🙂

closed 2022-09-15 by ssieb