esphome/issues #3239: Lilygo mini D1 plus v1.1

prophetmaster, 2022-04-24:
### The problem

how to compile ?

**the board:**

[http://www.lilygo.cn/claprod_view.aspx?TypeId=21&Id=1365&FId=t28:21:28](url)

### Which version of ESPHome has the issue?

2022.3.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.4.6

### What platform are you using?

ESP32-IDF

### Board

Lilygo mini D1 plus v1.1

### Example YAML snippet

```yaml
esphome:
  name: esp32c3
esp32:
  board: esp32-c3-devkitm-1
  framework:
    type: esp-idf
    version: latest
    # Custom sdkconfig options
    sdkconfig_options:
      CONFIG_COMPILER_OPTIMIZATION_SIZE: n
    # Advanced tweaking options
    advanced:
      ignore_efuse_mac_crc: true

# Enable logging
logger:
```

### Anything in the logs that might be useful for us?

```txt
10:30:28.922 -> ESP-ROM:esp32c3-api1-20210207
10:30:28.956 -> Build:Feb  7 2021
10:30:28.956 -> rst:0x1 (POWERON),boot:0xd (SPI_FAST_FLASH_BOOT)
10:30:28.956 -> invalid header: 0xffffffff
10:30:28.956 -> invalid header: 0xffffffff
10:30:28.956 -> invalid header: 0xffffffff
10:30:28.956 -> invalid header: 0xffffffff
10:30:28.956 -> invalid header: 0xffffffff
10:30:28.956 -> invalid header: 0xffffffff
10:30:28.956 -> invalid header: 0xffffffff
10:30:28.956 -> invalid header: 0xffffffff
10:30:28.956 -> invalid header: 0xffffffff
10:30:28.956 -> invalid header: 0xffffffff
```

ertw, 2022-06-28:
Hi, I'm running into a similar issue, and I hope this feedback is helpful. I'm able to start flashing the board, but then it gets stuck in a loop:

```
[02:51:00]ets_loader.c 79
[02:51:01]ESP-ROM:esp32c3-api1-20210207
[02:51:01]Build:Feb  7 2021
[02:51:01]rst:0x7 (TG0WDT_SYS_RST),boot:0xd (SPI_FAST_FLASH_BOOT)
[02:51:01]Saved PC:0x40049a64
[02:51:01]SPIWP:0xee
[02:51:01]mode:QIO, clock div:1
[02:51:01]load:0x3fcd6100,len:0x19c0
[02:51:01]ets_loader.c 79
[02:51:02]ESP-ROM:esp32c3-api1-20210207
[02:51:02]Build:Feb  7 2021
[02:51:02]rst:0x10 (RTCWDT_RTC_RST),boot:0xd (SPI_FAST_FLASH_BOOT)
[02:51:02]SPIWP:0xee
[02:51:02]mode:QIO, clock div:1
[02:51:02]load:0x3fcd6100,len:0x19c0
[02:51:02]ets_loader.c 79

```

ertw, 2022-06-28:
It appears that the board ID for this is `ttgo-t-oi-plus`: https://docs.platformio.org/en/stable/boards/espressif32/ttgo-t-oi-plus.html

However, I can't get that working, even directly from platformio, but I thin that is an issue on my side. Were you able to get this working @prophetmaster

closed 2022-11-11 by github-actions[bot]