esphome/issues #3241: Dallas sensors no longer found after upgrading from ESPHome version 2022.2.6 to 2022.3.1

quizzical86, 2022-04-24:
### The problem

I have an ESP8266 with 2 Dallas temperature sensors connected (and some other unrelated sensors). These have been working very reliably without problems so far.
Yesterday I upgraded this ESP8266 to run ESPHome 2022.3.1 and then these sensors stopped working. See below for the errors in the logs.
Fortunately I still had a snapshot of Home Assistant where I'm using ESPHome version 2022.2.6 so I launched that one and then reinstalled the ESP8266 with that version. That works fine (see the logs below).

I suspect it has to do with these changes made in version 2022.3: https://github.com/esphome/esphome/pull/3181
Obviously I'd like to be able to make changes in the YAML of this ESP8266 and also use the latest version of ESPHome.
How could I fix this problem?

### Which version of ESPHome has the issue?

2022.3.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.4

### What platform are you using?

ESP8266

### Board

D1 Mini ESP8266

### Component causing the issue

Dallas

### Example YAML snippet

```yaml
# Define the Dallas hub which allows reading the temperature sensor
dallas:
  - pin: GPIO13

sensor:
  - platform: dallas
    address: 0x163C01E07626F828
    name: "Temperature Buiten"
    filters:
      - offset: +0.6
  - platform: dallas
    address: 0x0700000003DF8D28
    name: "Temperature Zwemvijver"
    filters:
      - offset: -0.4
```

### Anything in the logs that might be useful for us?

```txt
Relevant elements in the log in the problematic version 2022.3.1:
[15:45:45][I][app:102]: ESPHome version 2022.3.1 compiled on Apr 23 2022, 21:45:41
[14:12:54][W][dallas.sensor:080]:   Found no sensors!
[14:13:42][W][dallas.sensor:137]: 'Temperature Buiten' - Scratch pad checksum invalid!
[14:13:42][W][dallas.sensor:137]: 'Temperature Zwemvijver' - Scratch pad checksum invalid!

Relevant elements in the log in the working version 2022.2.6:
[15:48:30][I][app:102]: ESPHome version 2022.2.6 compiled on Apr 24 2022, 15:47:11
[15:48:31][C][dallas.sensor:070]: DallasComponent:
[15:48:31][C][dallas.sensor:071]:   Pin: GPIO13
[15:48:31][C][dallas.sensor:072]:   Update Interval: 60.0s
[15:48:31][D][dallas.sensor:077]:   Found sensors:
[15:48:31][D][dallas.sensor:079]:     0x163c01e07626f828
[15:48:31][D][dallas.sensor:079]:     0x0700000003df8d28
```

mbrunton77, 2022-04-26:
I am seeing the same error message with a Sonoff Basic when I upgraded to 2022.3.0.

[17:24:05][C][dallas.sensor:075]: DallasComponent:
[17:24:05][C][dallas.sensor:076]:   Pin: GPIO14
[17:24:05][C][dallas.sensor:077]:   Update Interval: 60.0s
[17:24:05][W][dallas.sensor:080]:   Found no sensors!

quizzical86, 2022-04-27:
I noticed there are multiple issues open for this problem:
https://github.com/esphome/issues/issues/3153
https://github.com/esphome/issues/issues/3165
There seems to be a possibility to solve it by changing the pull-up resistors (and it seems to apply mostly to people who use long cables – mine are indeed over 20 meters long) but in my view it would be strange to solve what looks like a software problem by making hardware changes so hopefully this problem introduced by https://github.com/esphome/esphome/pull/3181 can be solved in a future release of ESPHome.

quizzical86, 2022-05-06:
I can  now confirm that switching to a 3.3K resistor, the values get reported again. Still it remains strange that moving from the 4.7K resistors as specified in the ESPHome documentation to a 3.3K resistor is the way to solve a problem introduced by a software update https://github.com/esphome/esphome/pull/3181.

quizzical86, 2022-05-26:
@edenhaus sure I'll close it although I don't understand your explanation.
You are saying we were all using the wrong pull-up resistor and that it is explained in the docs we need a different resistor if we have long wires.
However what the docs actually say is the following:
`For this, connect a resistor of about 4.7KΩ between 3.3V and the data pin. Values ± 1KΩ will, in most cases, work fine as well, if you don’t have massively long wires.`
So what the docs are saying is that for normal lengths you can use 3.7KΩ-4.7KΩ but for very long wire you need to use exactly 4.7KΩ.
The reality is that with this new update those with long wires had to replace their 4.7KΩ resistor for a 3.3KΩ resistor to get it working again.

closed 2022-05-26 by quizzical86

edenhaus, 2022-05-26:
I understand it different. You used use a pull-up resistor around 4.7k (+-1k), except you have massive long wires. in that case you need to find it by your own as it depends a lot on your setup (which wire, how long, which voltage are used…)

But you are also right, that it can be understand in your way😊 Please feel free to open a PR to improve the docs

Stokstaart, 2022-07-02:
So let me get this straigth, i have a D1 mini from a friend here which i am supposed to update (add 2nd DS18B20, 2 in total) but now this doesn't work anymore with 4K7 resistor?
Both sensors have about 50cm cable length, is that long? How do i determine what resistor would work??

quizzical86, 2022-07-02:
no 50cm is not long :) so use the recommended 4K7