esphome/issues #3245: TMP117 Component Doesn't Compile

mankyd, 2022-04-27:
### The problem

Copy and pasted the example BME280 sample config and get this compilation error:

https://esphome.io/components/sensor/tmp117.html

```
Linking /data/air-main/.pioenvs/air-main/firmware.elf
/data/cache/platformio/packages/toolchain-xtensa-esp32s2/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld: /data/air-main/.pioenvs/air-main/src/main.o:(.literal._Z5setupv+0x84): undefined reference to `vtable for esphome::tmp117::TMP117Component'
/data/cache/platformio/packages/toolchain-xtensa-esp32s2/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld: /data/air-main/.pioenvs/air-main/src/main.o:(.literal._Z5setupv+0x88): undefined reference to `vtable for esphome::tmp117::TMP117Component'
collect2: error: ld returned 1 exit status
*** [/data/air-main/.pioenvs/air-main/firmware.elf] Error 1
```

### Which version of ESPHome has the issue?

2022.4.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.4.6

### What platform are you using?

ESP32

### Board

SparkFun Thing Plus - ESP32-S2 WROOM

### Component causing the issue

TMP117

### Example YAML snippet

```yaml
# Example configuration entry
sensor:
  - platform: tmp117
    name: "Living Room Temperature"
    update_interval: 60s
```

chrishoage, 2022-05-06:
I am encountering a similar error with a different sensor. 

Oddly, when I use 

```
docker run --rm -v "${PWD}":/config --device=/dev/ttyUSB0 -it esphome/esphome run livingroom.yaml
```

On my workstation it compiles fine and is able to install. 

The error below only happens on my server when using the web portal of esphome

```yaml
esphome:
  name: livingroom

esp32:
  board: esp32-s2-saola-1
  framework:
    type: esp-idf

i2c:
  sda: 01
  scl: 02

sensor:
  - platform: tsl2591
    name: "Living Room Light Sensor"
    visible:
      name: "Living Room Visible Light"
    infrared:
      name: "Living Room Infrared Light"
    full_spectrum:
      name: "Living Room Full Spectrum Light"
    calculated_lux:
      name: "Living Room Lux"
  - platform: sht4x
    temperature:
      name: "Living Room Temperature"
    humidity:
      name: "Living Room Relative Humidity"

```
/src/main.cpp:286: undefined reference to `esphome::sensor::Sensor::set_accuracy_decimals(signed char)'
/config/.esphome/platformio/packages/toolchain-xtensa-esp32s2/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld: .pioenvs/livingroom/src/esphome/core/controller.o:(.literal._ZN7esphome10Controller16setup_controllerEb+0x8): undefined reference to `esphome::sensor::Sensor::add_on_state_callback(std::function<void (float)>&&)'
/config/.esphome/platformio/packages/toolchain-xtensa-esp32s2/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld: .pioenvs/livingroom/src/esphome/core/controller.o: in function `esphome::Controller::setup_controller(bool)':
/config/.esphome/build/livingroom/src/esphome/core/controller.cpp:29: undefined reference to `esphome::sensor::Sensor::add_on_state_callback(std::function<void (float)>&&)'
collect2: error: ld returned 1 exit status
*** [.pioenvs/livingroom/firmware.elf] Error 1
========================= [FAILED] Took 24.66 seconds =========================
```

Please let me know if there is any further information that might be helpful!

chrishoage, 2022-05-06:
@mankyd Just a heads up on a hunch I ran "clean build files" and after that I was able to successfully build.

Perhaps give it a try, can't hurt! 

[image]

mankyd, 2022-05-06:
Cleaning the build did seem to make the error go away. Thanks!

Not sure if this bug should stay open or not.

chrishoage, 2022-05-09:
Due to the nature of the bug I thing unless either of us can come up with a reliable repo case the issue can probably be closed.

closed 2022-09-12 by ssieb