esphome/issues #3247: On_press in binary sensor working as on_release.

melwinek, 2022-04-27:
### The problem

My switch changes state when released, rather than when touched.

### Which version of ESPHome has the issue?

2022.4.0

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2022.4.7

### What platform are you using?

ESP8266

### Board

Wemos D1 mini

### Example YAML snippet

```yaml
wifi:
  ssid: xxxxxx
  password: xxxxxx
captive_portal:
logger:
api:
  password: xxxxxx
ota:
  password: xxxxxx
esphome:
  name: myswitch1
  platform: ESP8266
  board: esp01_1m

binary_sensor: 
- platform: gpio
  pin:
    number: GPIO16
  internal: True
  name: My Button 1
  on_press:
    - switch.toggle: relay1
    
switch: 
- platform: gpio
  name: My Relay 1
  pin: GPIO12
  id: relay1
```

bkaufx, 2022-04-27:
How is it wired up?  Probably need to add inverted: true to the binary sensor definition.  https://esphome.io/components/binary_sensor/gpio.html#inverting-values

melwinek, 2022-04-28:
My mistake. I don't even know when I typed into yaml "inverted: True". Eclipse of the mind. Sorry.

mmakaay, 2022-04-28:
Heh,  it happens. No problem with that.
I'll close this issue then.

closed 2022-04-28 by mmakaay