esphome/issues #3248: Hydreon Rain sensor RG-15

mikesjogren, 2022-04-27:
### The problem

There seems to be a problem with the communication with the sensor. I get a lot of error messages in the log

### Which version of ESPHome has the issue?

2022.4.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.4.7

### What platform are you using?

ESP8266

### Board

NodeMCU

### Component causing the issue

hydreon_rgxx

### Anything in the logs that might be useful for us?

```txt
INFO Reading configuration /config/esphome/rainsensor-ir.yaml...
INFO Starting log output from 10.100.210.145 using esphome API
INFO Successfully connected to 10.100.210.145
[09:53:36][I][app:102]: ESPHome version 2022.4.0 compiled on Apr 26 2022, 20:11:08
[09:53:36][C][wifi:491]: WiFi:
[09:53:36][C][wifi:353]:   Local MAC: 40:91:51:4F:86:59
[09:53:36][C][wifi:354]:   SSID: 'VVK2'[redacted]
[09:53:36][C][wifi:355]:   IP Address: 10.100.210.145
[09:53:36][C][wifi:356]:   BSSID: D2:21:F9:12:7B:75[redacted]
[09:53:36][C][wifi:358]:   Hostname: 'rainsensor-ir'
[09:53:36][C][wifi:360]:   Signal strength: -64 dB ▂▄▆█
[09:53:36][C][wifi:364]:   Channel: 13
[09:53:36][C][wifi:365]:   Subnet: 255.255.255.0
[09:53:36][C][wifi:366]:   Gateway: 10.100.210.1
[09:53:36][C][wifi:367]:   DNS1: 10.100.210.8
[09:53:36][C][wifi:368]:   DNS2: 10.100.210.2
[09:53:36][C][logger:233]: Logger:
[09:53:36][C][logger:234]:   Level: DEBUG
[09:53:36][C][logger:235]:   Log Baud Rate: 0
[09:53:36][C][logger:236]:   Hardware UART: UART0
[09:53:36][C][uart.arduino_esp8266:102]: UART Bus:
[09:53:36][C][uart.arduino_esp8266:103]:   TX Pin: GPIO1
[09:53:36][C][uart.arduino_esp8266:104]:   RX Pin: GPIO3
[09:53:36][C][uart.arduino_esp8266:106]:   RX Buffer Size: 256
[09:53:36][C][uart.arduino_esp8266:108]:   Baud Rate: 9600 baud
[09:53:36][C][uart.arduino_esp8266:109]:   Data Bits: 8
[09:53:36][C][uart.arduino_esp8266:110]:   Parity: NONE
[09:53:36][C][uart.arduino_esp8266:111]:   Stop bits: 1
[09:53:36][C][uart.arduino_esp8266:113]:   Using hardware serial interface.
[09:53:36][C][hydreon_rgxx.sensor:015]: hydreon_rgxx:
[09:53:36][C][hydreon_rgxx.sensor:019]:   Update Interval: 30.0s
[09:53:36][C][hydreon_rgxx.sensor:026]:   "Acc" 'rain'
[09:53:36][C][hydreon_rgxx.sensor:026]:     Device Class: 'humidity'
[09:53:36][C][hydreon_rgxx.sensor:026]:     State Class: 'measurement'
[09:53:36][C][hydreon_rgxx.sensor:026]:     Unit of Measurement: 'mm'
[09:53:36][C][hydreon_rgxx.sensor:026]:     Accuracy Decimals: 2
[09:53:36][C][hydreon_rgxx.sensor:026]:   "Rint" 'rain intensity'
[09:53:36][C][hydreon_rgxx.sensor:026]:     Device Class: 'humidity'
[09:53:36][C][hydreon_rgxx.sensor:026]:     State Class: 'measurement'
[09:53:36][C][hydreon_rgxx.sensor:026]:     Unit of Measurement: 'mm/h'
[09:53:36][C][hydreon_rgxx.sensor:026]:     Accuracy Decimals: 2
[09:53:36][C][hydreon_rgxx.sensor:026]:   "EventAcc" 'rain event'
[09:53:36][C][hydreon_rgxx.sensor:026]:     Device Class: 'humidity'
[09:53:36][C][hydreon_rgxx.sensor:026]:     State Class: 'measurement'
[09:53:36][C][hydreon_rgxx.sensor:026]:     Unit of Measurement: 'mm'
[09:53:36][C][hydreon_rgxx.sensor:026]:     Accuracy Decimals: 2
[09:53:36][C][hydreon_rgxx.sensor:026]:   "TotalAcc" 'rain total'
[09:53:36][C][hydreon_rgxx.sensor:026]:     Device Class: 'humidity'
[09:53:36][C][hydreon_rgxx.sensor:026]:     State Class: 'measurement'
[09:53:36][C][hydreon_rgxx.sensor:026]:     Unit of Measurement: 'mm'
[09:53:36][C][hydreon_rgxx.sensor:026]:     Accuracy Decimals: 2
[09:53:37][C][captive_portal:088]: Captive Portal:
[09:53:37][C][mdns:084]: mDNS:
[09:53:37][C][mdns:085]:   Hostname: rainsensor-ir
[09:53:37][C][ota:085]: Over-The-Air Updates:
[09:53:37][C][ota:086]:   Address: 10.100.210.145:8266
[09:53:37][C][ota:089]:   Using Password.
[09:53:37][C][api:138]: API Server:
[09:53:37][C][api:139]:   Address: 10.100.210.145:6053
[09:53:37][C][api:143]:   Using noise encryption: NO
[09:53:44][W][hydreon_rgxx.sensor:051]: Missing Rint
[09:53:44][E][hydreon_rgxx.sensor:063]: data missing 12 times
[09:53:44][D][sensor:124]: 'rain': Sending state 0.00000 mm with 2 decimals of accuracy
[09:53:44][D][hydreon_rgxx.sensor:200]: Received Acc: 0.000000
[09:53:44][D][sensor:124]: 'rain event': Sending state 0.00000 mm with 2 decimals of accuracy
[09:53:44][D][hydreon_rgxx.sensor:200]: Received EventAcc: 0.000000
[09:53:44][D][sensor:124]: 'rain total': Sending state 1.00000 mm with 2 decimals of accuracy
[09:53:44][D][hydreon_rgxx.sensor:200]: Received TotalAcc: 1.000000
[09:54:14][W][hydreon_rgxx.sensor:051]: Missing Rint
[09:54:14][E][hydreon_rgxx.sensor:063]: data missing 13 times
[09:54:14][D][sensor:124]: 'rain': Sending state 0.00000 mm with 2 decimals of accuracy
[09:54:14][D][hydreon_rgxx.sensor:200]: Received Acc: 0.000000
[09:54:14][D][sensor:124]: 'rain event': Sending state 0.00000 mm with 2 decimals of accuracy
[09:54:14][D][hydreon_rgxx.sensor:200]: Received EventAcc: 0.000000
[09:54:14][D][sensor:124]: 'rain total': Sending state 1.00000 mm with 2 decimals of accuracy
[09:54:14][D][hydreon_rgxx.sensor:200]: Received TotalAcc: 1.000000
[09:54:44][W][hydreon_rgxx.sensor:051]: Missing Rint
[09:54:44][E][hydreon_rgxx.sensor:063]: data missing 14 times
[09:54:44][D][sensor:124]: 'rain': Sending state 0.00000 mm with 2 decimals of accuracy
[09:54:44][D][hydreon_rgxx.sensor:200]: Received Acc: 0.000000
[09:54:44][D][sensor:124]: 'rain event': Sending state 0.00000 mm with 2 decimals of accuracy
[09:54:44][D][hydreon_rgxx.sensor:200]: Received EventAcc: 0.000000
[09:54:44][D][sensor:124]: 'rain total': Sending state 1.00000 mm with 2 decimals of accuracy
[09:54:44][D][hydreon_rgxx.sensor:200]: Received TotalAcc: 1.000000
[09:55:14][W][hydreon_rgxx.sensor:051]: Missing Rint
[09:55:14][E][hydreon_rgxx.sensor:063]: data missing 15 times
[09:55:14][D][sensor:124]: 'rain': Sending state 0.00000 mm with 2 decimals of accuracy
[09:55:14][D][hydreon_rgxx.sensor:200]: Received Acc: 0.000000
[09:55:14][D][sensor:124]: 'rain event': Sending state 0.00000 mm with 2 decimals of accuracy
[09:55:14][D][hydreon_rgxx.sensor:200]: Received EventAcc: 0.000000
[09:55:14][D][sensor:124]: 'rain total': Sending state 1.00000 mm with 2 decimals of accuracy
[09:55:14][D][hydreon_rgxx.sensor:200]: Received TotalAcc: 1.000000
[09:55:44][W][hydreon_rgxx.sensor:051]: Missing Rint
[09:55:44][E][hydreon_rgxx.sensor:063]: data missing 16 times
[09:55:44][E][hydreon_rgxx.sensor:065]: asking sensor to reboot
[09:55:44][D][sensor:124]: 'rain': Sending state nan mm with 2 decimals of accuracy
[09:55:44][D][sensor:124]: 'rain intensity': Sending state nan mm/h with 2 decimals of accuracy
[09:55:44][D][sensor:124]: 'rain event': Sending state nan mm with 2 decimals of accuracy
[09:55:44][D][sensor:124]: 'rain total': Sending state nan mm with 2 decimals of accuracy
[09:55:45][I][hydreon_rgxx.sensor:204]: Got unknown line: Reset D

[09:55:46][I][hydreon_rgxx.sensor:204]: Got unknown line: 

[09:55:46][I][hydreon_rgxx.sensor:146]: Comment: ;************************************************************
[09:55:46][I][hydreon_rgxx.sensor:146]: Comment: ; HYDREON MODEL RG-15 RAIN GAUGE
[09:55:46][I][hydreon_rgxx.sensor:146]: Comment: ; SOFTWARE COPYRIGHT (C) 2020 HYDREON CORPORATION
[09:55:46][I][hydreon_rgxx.sensor:174]: detected sw version 1000
[09:55:46][I][hydreon_rgxx.sensor:204]: Got unknown line: Emitters 8 8, NormEm 17, EmTotal 16

[09:55:46][I][hydreon_rgxx.sensor:157]: Boot detected: PwrDays 0
[09:55:46][I][hydreon_rgxx.sensor:204]: Got unknown line: 

[09:55:49][I][hydreon_rgxx.sensor:204]: Got unknown line: p

[09:55:49][I][hydreon_rgxx.sensor:204]: Got unknown line: h

[09:55:49][I][hydreon_rgxx.sensor:204]: Got unknown line: m

[09:56:14][W][hydreon_rgxx.sensor:051]: Missing Rint
[09:56:14][E][hydreon_rgxx.sensor:063]: data missing 1 times
[09:56:14][D][sensor:124]: 'rain': Sending state 0.00000 mm with 2 decimals of accuracy

This then repeats "indefinitely"
```

### Additional information

It says Missing **Rint** but the documentation says **RInt**

mikesjogren, 2022-04-27:
I did some testing and can confirm that Rint is the problem.

Changed my sensors.py and now it works fine!

PROTOCOL_NAMES = {
    CONF_MOISTURE: "R",
    CONF_ACC: "Acc",
    CONF_R_INT: "RInt",
    CONF_EVENT_ACC: "EventAcc",
    CONF_TOTAL_ACC: "TotalAcc",
}

emdekext, 2022-05-04:
same problem here, also missing Rint. Where can I find sensors.py on a Hass OS installation for the HA add-on?
Thanks!

mikesjogren, 2022-05-04:
You have to connect to the ESP docker container and then find the file
_root@5c53de3b-esphome:/esphome/esphome/components/hydreon_rgxx#_
Edit it and update your ESP. The sensor.py file will by updated by Home Assistant so don't update your ESP until the problem is solved or you have to do the correction each time you update the ESP.

emdekext, 2022-05-07:
Thanks! I will give this a shot, have tried ssh to port 22222 before but ran into problems with the key, and didn't push on then. Enough reason to sort this out now...

wtremmel, 2022-05-24:
Same problem here. Pull request?

juhanjuku, 2022-05-28:
Missing Rint is only part of problem. Actually seems that whole message is parsed wrong. I have tested with my RG-15 sensor and only 'TotalAcc' gets value and it's without decimals.
I have raised issue https://github.com/home-assistant/core/issues/72643
Developer jesserockz admits that he had no real RG-15 for testing, so I'm ready to help whatever he needs.
https://github.com/esphome/esphome/pull/2711

functionpointer, 2022-05-29:
Hi, developer of the `hydreon_rgxx` integration here. Unfortunately i don't know about this huge issue ☹️  
I finally got notified when @wtremmel stepped up (thank you ❤️ ) and submitted a pull request.

juhanjuku, 2022-05-29:
> Hi, developer of the `hydreon_rgxx` integration here. Unfortunately i don't know about this huge issue ☹️ I finally got notified when @wtremmel stepped up (thank you ❤️ ) and submitted a pull request.

Please let me know how can I assist you with testing. I have RG-15.

functionpointer, 2022-05-29:
You could test the changes. You can add the changes to your Home-Assistant OS-based install (as already discussed in https://github.com/home-assistant/core/issues/72643#issuecomment-1140444856).

Alternatively you can leave your Raspberry PI alone and install esphome on your PC. Basically you follow this: https://esphome.io/guides/installing_esphome.html
However, you **don't** run `pip install esphome`. Instead you [install git](https://git-scm.com/book/en/v2/Getting-Started-Installing-Git). Then open a command line and run

```
git clone https://github.com/wtremmel/esphome.git
cd esphome
pip install -e .
```
Then create a new file and paste the contents the yaml file for your esphome-based device in there (from your Home-Assistant).
Then run
```
esphome run <yourfilename>.yaml
```

Esphome should compile the new firmware and upload it to your device.

juhanjuku, 2022-05-29:
I installed esphome on PC and followed your instructions. From first look all went good, except result seems to be same as it was.
TotalAcc still 8.00 mm, but RG-15 sent 8.60
and I can't see RInt

Have you made changes in https://github.com/wtremmel/esphome/tree/dev/esphome/components/hydreon_rgxx?

[image]

functionpointer, 2022-05-29:
Seems like you are on the wrong branch. Try
```
git checkout hydreon-rg15-rainIntensity
esphome run <yourfilename>.yaml
```

functionpointer, 2022-05-29:
The decimal point changes are in another branch, called `wtremmel-hydreon-rg15-data`.
They will get combined once the two pull requests are merged (https://github.com/esphome/esphome/pull/3511 and https://github.com/esphome/esphome/pull/3512).
You can already combine them on your machine by calling
```
git merge <the other branch>
```
So if you are on `hydreon-rg15-rainIntensity` you would call
```
git merge origin/wtremmel-hydreon-rg15-data
```

juhanjuku, 2022-05-29:
I can't say that I understood all what I did, first time with git, but I can confirm
IT WORKS NOW :)

[image]

[image]

juhanjuku, 2022-05-29:
I have set update_interval: 10 sec
Can you comment why lovelace values are not updated?
WiFi signal rssi updates frequently from same esp

[image]

functionpointer, 2022-05-29:
Nice to see that it works!

Lovelace should be updating just fine, especially if you see the updates in the logs (`esphome <filename> logs` to bring it back up without re-flashing). I am running `update_interval: 5s` and i used to run `update_interval: 1s` earlier.

Maybe the id has changed? Do you have two rain sensors showing up now?

juhanjuku, 2022-05-30:
No, there are no duplicated sensors. 
I deleted hydreon device from HA and restarted. It was discovered, I added again, but still no refresh for rain entities. Values update only on restart. 
Strange is that wifi sensor from same esp updates normally.
Tomorrow will be rainy day, so I can make more tests.

closed 2022-05-31 by jesserockz

juhanjuku, 2022-05-31:
Despite to strange loevlace behaivor (item last update shows actually last change) 
I can confirm that esp code works now correctly
[image]

roger-, 2022-07-17:
Testing a RG-15 which *I think* is working, but I get some weird messages after I trigger an event by dripping some water on the lens:

```
[15:32:44][I][hydreon_rgxx.sensor:146]: Comment: ;----------------------------------------
[15:32:44][I][hydreon_rgxx.sensor:204]: Got unknown line: Event
[15:32:44]
[15:32:44][I][hydreon_rgxx.sensor:174]: detected sw version 1000
[15:32:44][I][hydreon_rgxx.sensor:204]: Got unknown line:
[15:32:44]
[15:32:44][D][sensor:124]: 'Rainfall accumulation increment': Sending state 0.00000 mm with 2 decimals of accuracy
[15:32:44][D][hydreon_rgxx.sensor:200]: Received Acc: 0.000000
[15:32:44][D][sensor:124]: 'Rainfall intensity': Sending state 0.55000 mm/h with 2 decimals of accuracy
[15:32:44][D][hydreon_rgxx.sensor:200]: Received RInt: 0.550000
[15:32:44][D][sensor:124]: 'Rainfall accumulation event': Sending state 0.00000 mm with 2 decimals of accuracy
[15:32:44][D][hydreon_rgxx.sensor:200]: Received EventAcc: 0.000000
[15:32:44][D][sensor:124]: 'Rainfall accumulation total': Sending state 0.00000 mm with 2 decimals of accuracy
[15:32:44][D][hydreon_rgxx.sensor:200]: Received TotalAcc: 0.000000
```

Seems to register an increase in the intensity (0.55 from 0.00) but I'm not sure what the `Event` line is about (I don't see a command like that in the documentation).

functionpointer, 2022-07-17:
Could you enable uart logging and try again?

It works like this:
```
uart:
  rx_pin: GPIO16
  tx_pin: GPIO17
  baud_rate: 9600
  id: uart_port
  debug:
    direction: BOTH
    dummy_receiver: false
    after:
      delimiter: "\n"
    sequence:
      - lambda: UARTDebug::log_string(direction, bytes);
```

roger-, 2022-07-17:
Sure — does that work with the software UART BTW?

functionpointer, 2022-07-17:
Uhh...probably? I have never used software uart in esphome.

roger-, 2022-07-18:
Yup logging works fine on the software UART!

Here's the output I'm getting:
```
[08:49:50][D][uart_debug:158]: >>> "R\n"
[08:49:50][D][uart_debug:158]: <<< "\r\n"
[08:49:50][I][hydreon_rgxx.sensor:204]: Got unknown line:
[08:49:50]
[08:49:50][D][uart_debug:158]: <<< ";----------------------------------------\r\n"
[08:49:50][I][hydreon_rgxx.sensor:146]: Comment: ;----------------------------------------
[08:49:50][D][uart_debug:158]: <<< "Event\r\n"
[08:49:50][I][hydreon_rgxx.sensor:204]: Got unknown line: Event
[08:49:50]
[08:49:50][D][uart_debug:158]: <<< "SW 1.000 2020.07.06\r\n"
[08:49:50][I][hydreon_rgxx.sensor:174]: detected sw version 1000
[08:49:50][D][uart_debug:158]: <<< "\r\n"
[08:49:50][I][hydreon_rgxx.sensor:204]: Got unknown line:
[08:49:50]
[08:49:50][D][uart_debug:158]: <<< "Acc  0.03 mm, EventAcc  0.03 mm, TotalAcc  0.03 mm, RInt  2.23 mmph\r\n"
[08:49:50][D][sensor:124]: 'Rainfall accumulation increment': Sending state 0.03000 mm with 2 decimals of accuracy
[08:49:50][D][hydreon_rgxx.sensor:200]: Received Acc: 0.030000
[08:49:50][D][sensor:124]: 'Rainfall intensity': Sending state 2.23000 mm/h with 2 decimals of accuracy
[08:49:50][D][hydreon_rgxx.sensor:200]: Received RInt: 2.230000
[08:49:50][D][sensor:124]: 'Rainfall accumulation event': Sending state 0.03000 mm with 2 decimals of accuracy
[08:49:50][D][hydreon_rgxx.sensor:200]: Received EventAcc: 0.030000
[08:49:50][D][sensor:124]: 'Rainfall accumulation total': Sending state 0.03000 mm with 2 decimals of accuracy
[08:49:50][D][hydreon_rgxx.sensor:200]: Received TotalAcc: 0.030000
```

Seems like:
* blank lines should be ignored entirely
* `Event\r\n` is sent (asynchronously?) when rain is (first?) detected

These could probably be ignored, but it would be good to handle them more cleanly.

functionpointer, 2022-07-18:
Very nice logs and good find on the documentation.

I wonder if `Event` could be useful for anything. But we should already see `EventAcc` being reset so i guess not.

roger-, 2022-07-18:
Yeah `Event` is redundant and probably should be cleanly ignored in the code.

But it could be useful to parse status messages like `LensBad` and `EmSat`. I also noticed there's a temperature command, would be nice to support that.

roger-, 2022-07-18:
Thanks! Did some quick testing, first off I noticed this:

```
[16:32:30][D][uart_debug:158]: <<< ";************************************************************\r\n"
[16:32:30][I][hydreon_rgxx.sensor:157]: Comment: ;************************************************************
[16:32:30][D][uart_debug:158]: <<< "; HYDREON MODEL RG-15 RAIN GAUGE\r\n"
[16:32:30][I][hydreon_rgxx.sensor:157]: Comment: ; HYDREON MODEL RG-15 RAIN GAUGE
[16:32:30][D][uart_debug:158]: <<< "; SOFTWARE COPYRIGHT (C) 2020 HYDREON CORPORATION\r\n"
[16:32:30][I][hydreon_rgxx.sensor:157]: Comment: ; SOFTWARE COPYRIGHT (C) 2020 HYDREON CORPORATION
[16:32:30][D][uart_debug:158]: <<< "SW 1.000 2020.07.06\r\n"
[16:32:30][I][hydreon_rgxx.sensor:198]: detected sw version 1000
[16:32:30][D][uart_debug:158]: <<< "Emitters 11 12, NormEm 24, EmTotal 23\r\n"
[16:32:30][I][hydreon_rgxx.sensor:230]: Got unknown line: Emitters 11 12, NormEm 24, EmTotal 23
[16:32:30]
[16:32:30][D][uart_debug:158]: <<< "PwrDays 0\r\n"
[16:32:30][I][hydreon_rgxx.sensor:181]: Boot detected: PwrDays 0
[16:32:30][D][uart_debug:158]: >>> "P\n"
[16:32:30][D][uart_debug:158]: >>> "H\n"
[16:32:30][D][uart_debug:158]: >>> "M\n"
[16:32:30][D][uart_debug:158]: <<< "\r\n"
[16:32:30][D][hydreon_rgxx.sensor:162]: Received empty line
[16:32:33][D][uart_debug:158]: <<< "p\r\n"
[16:32:33][I][hydreon_rgxx.sensor:230]: Got unknown line: p
[16:32:33]
[16:32:33][D][uart_debug:158]: <<< "h\r\n"
[16:32:33][I][hydreon_rgxx.sensor:230]: Got unknown line: h
[16:32:33]
[16:32:33][D][uart_debug:158]: <<< "m\r\n"
[16:32:33][I][hydreon_rgxx.sensor:230]: Got unknown line: m
[16:32:33]
[16:32:37][D][uart_debug:158]: >>> "R\n"
[16:32:37][D][uart_debug:158]: >>> "T\n"
```

Looks like the acknowledgements (`p`, `h`, etc.) aren't handled correctly. Same for `Emitters`.

Right after I get this:

```
[16:32:37][D][binary_sensor:036]: 'RG-15 too cold': Sending state OFF
[16:32:37][D][binary_sensor:036]: 'RG-15 lens bad': Sending state OFF
[16:32:37][D][binary_sensor:036]: 'RG-15 emitter saturated': Sending state ON
``` 

Seems like `em_sat` gets initialized to ON? Not sure how to test it, so it stays on.

Later on I see this:

```
[16:42:07][D][uart_debug:158]: >>> "R\n"
[16:42:07][D][uart_debug:158]: >>> "T\n"
[16:42:07][D][uart_debug:158]: <<< "Acc  0.00 mm,ventAcc  0.00 mm, TotalAcc  0.03 mm, RInt  0.00 mmph\r\n"
[16:42:07][D][sensor:124]: 'RG-15 rainfall accumulation increment': Sending state 0.00000 mm with 2 decimals of accuracy[16:42:07][D][hydreon_rgxx.sensor:224]: Received Acc: 0.000000
[16:42:07][D][sensor:124]: 'RG-15 rainfall intensity': Sending state 0.00000 mm/h with 2 decimals of accuracy
[16:42:07][D][hydreon_rgxx.sensor:224]: Received RInt: 0.000000
[16:42:07][D][sensor:124]: 'RG-15 rainfall accumulation total': Sending state 0.03000 mm with 2 decimals of accuracy
[16:42:07][D][hydreon_rgxx.sensor:224]: Received TotalAcc: 0.030000
[16:42:07][D][sensor:124]: 'RG-15 temperature': Sending state 0.00000 °C with 0 decimals of accuracy
[16:42:07][D][hydreon_rgxx.sensor:224]: Received t: 0.000000
[16:42:23][D][sensor:124]: 'WiFi signal strength': Sending state -43.00000 dBm with 0 decimals of accuracy
[16:42:37][W][hydreon_rgxx.sensor:051]: Missing EventAcc
[16:42:37][E][hydreon_rgxx.sensor:063]: data missing 1 times
```

Seems like the temperature command doesn't work as I don't see a response and it gets stuck on 0.0. Also notice how ` mm,EventAcc` gets corrupted to ` mm,ventAcc`, causing the parser to fail. Not sure if this is a hardware issue (maybe software serial?)

functionpointer, 2022-07-18:
Very good! Looks like I have to fix some bugs.

And they removed the temperature from the documentation again?
Okay then, I will maybe lock that feature to 1.200 and later. Good thing I wrote a parser for the version.
The broken response is also annoying. Maybe I shouldn't send two requests at once.

functionpointer, 2022-07-19:
> **EDIT**: spoke to Hydreon and temperature only works on a new revision (1.200) of the RG-9, so I can't test it.

Temperature works on RG-9? Or RG-15? We saw it in RG-15 documentation. Did they tell you it works on RG-9 1.200?

wtremmel, 2022-07-19:
> EDIT: spoke to Hydreon and temperature only works on a new revision (1.200) of the RG-9, so I can't test it.

Sorry to hijack this - but if you have a support contact at Hydreon can you post it here?
I still have the problem of erroneous rain reports if the sun is directly over head

roger-, 2022-07-19:
> > **EDIT**: spoke to Hydreon and temperature only works on a new revision (1.200) of the RG-9, so I can't test it.
> 
> Temperature works on RG-9? Or RG-15? We saw it in RG-15 documentation. Did they tell you it works on RG-9 1.200?

It was an error in the documentation. It only applies to RG-9.

roger-, 2022-07-19:
> > EDIT: spoke to Hydreon and temperature only works on a new revision (1.200) of the RG-9, so I can't test it.
> 
> Sorry to hijack this - but if you have a support contact at Hydreon can you post it here? I still have the problem of erroneous rain reports if the sun is directly over head

Sure: info@hydreon.com

roger-, 2022-07-19:
Another one:

```
[08:05:37][W][hydreon_rgxx.sensor:131]: hydreon_rgxx failed to boot 2 times
[08:05:37][D][uart_debug:158]: >>> "K\n"
[08:05:38][D][uart_debug:158]: <<< "Reset D\r\n"
[08:05:38][I][hydreon_rgxx.sensor:230]: Got unknown line: Reset D
[08:05:38]
[08:05:40][D][uart_debug:158]: <<< "\r\n"
```

Besides the failing to boot 2 times, looks like `Reset` isn't parsed either.

roger-, 2022-07-20:
Seems to address all the issues, but I did notice some weird corruption again:

```
[18:16:06][D] [uart_debug:158]: >>> "R\n"
[18:16:06][D] [binary_sensor:036]: 'RG-15 too cold': Sending state OFF
[18:16:06][D] [binary_sensor:036]: 'RG-15 lens bad': Sending state OFF
[18:16:06][D] [binary_sensor:036]: 'RG-15 emitter saturated': Sending state OFF
[18:16:06][D] [uart_debug:158]: <<< "Acc 0.00 m\xB6\t*|xD9\x95\xB9\xD1\x05\x8D\x8D\x81 0.00 mm, TotalAcc 0.03 mm, RInt 0.0 0 mmph\r\n"
[18:16:06][D] [sensor:124]: 'RG-15 rainfall accumulation increment': Sending state 0.00000 mm with 2 decimals of accuracy
[18:16:06][D] [hydreon_rgxx.sensor:232]: Received Acc : 0.000000
[18:16:06][D] [sensor:124]: 'RG-15 rainfall intensity': Sending state 0.00000 mm/h with 2 decimals of accuracy
[18:16:06][D] [hydreon_rgxx.sensor:232]: Received RInt : 0.000000
[18:16:06][D] [sensor: 124]: 'RG-15 rainfall accumulation total': Sending state 0.03000 mm with 2 decimals of accuracy
[18:16:06][D] [hydreon_rgxx.sensor:232]: Received TotalAcc : 0.030006
[18:16:36][W] [hydreon_rgxx.sensor:063]: Missing EventAcc
[18:16:36][E] [hydreon_rgxx.sensor:068]: missing 1 sensors; 1 times in a row
```

Notice line 5. Only seen it at boot up so far.

functionpointer, 2022-07-21:
Weird. Is right after the options (`p`, `h`,`m`) get set? 
They are set all at once, maybe that confuses it.

If not then I don't think we can do anything about this. Since it only happens once at bootup it's not a big problem anyway.

roger-, 2022-07-23:
Haven't noticed it again, but the init commands (P H R) get sent right after one another and before the individual acknowledgements, so perhaps that's related. They do eventually get acknowledged in the correct order so perhaps not...

Also the first messages I see are this:

```
[10:54:09][D][uart_debug:158]: >>> "K\n"                                                                                
[10:54:12][W][hydreon_rgxx.sensor:133]: hydreon_rgxx failed to boot 1 times                                             
[10:54:12][D][uart_debug:158]: >>> "K\n"                                                                                
[10:54:13][D][uart_debug:158]: <<< "Reset D\r\n"                                                                        
[10:54:13][I][hydreon_rgxx.sensor:241]: Ignoring Reset D                                                                
[10:54:15][D][uart_debug:158]: <<< "\r\n"         
```

Didn't look into the logic, but is the boot failure message expected?

Otherwise seems to be working pretty good!

functionpointer, 2022-07-23:
The boot failure is a real failure here. The sensor is commanded to reboot (`K`) but we don't get a response in 4 seconds.
So we correctly deduce the reboot attempt has failed and try again.

Did this happen after powerup? Maybe the sensor isn't ready to receive commands yet.

functionpointer, 2022-07-23:
Since the issues seem to be fixed for the most part, i have marked [the PR](https://github.com/esphome/esphome/pull/3642) as ready to review now

roger-, 2022-07-24:
> The boot failure is a real failure here. The sensor is commanded to reboot (`K`) but we don't get a response in 4 seconds. So we correctly deduce the reboot attempt has failed and try again.
> 
> Did this happen after powerup? Maybe the sensor isn't ready to receive commands yet.

This was after rebooting the ESP (a warm restart).

roger-, 2022-07-24:
Another useful feature would be to expose switches for the accumulation reset command (O) and possibly the reboot command (K). 

Alternatively perhaps the O command can be issued at boot so that rebooting ESPHome accomplishes both?

roger-, 2022-07-28:
> > EDIT: spoke to Hydreon and temperature only works on a new revision (1.200) of the RG-9, so I can't test it.
> 
> Sorry to hijack this - but if you have a support contact at Hydreon can you post it here? I still have the problem of erroneous rain reports if the sun is directly over head

I'm noticing the same thing -- have you gotten a response from them?