esphome/issues #3250: No more Climate in homeassistant

gefkuz, 2022-04-28:
### The problem

Hello,

Since last update of EspHome (2022.3.2 to 2022.4.0) my old climate pid esphome are unvailable in homeassistant.... With the same code, just an update.
Please what's wrong with this update?

### Which version of ESPHome has the issue?

2022.4.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.4.7

### What platform are you using?

ESP8266

### Board

wemos D1

### Component causing the issue

climate pid

### Example YAML snippet

```yaml
esphome:
  name: thermostat-chambre-bas

esp8266:
  board: d1_mini

# Enable logging
logger:

# Enable Home Assistant API
api:
  reboot_timeout: 0s

ota:
  password: "4b8e58c46c4370c6e7a26d5a74d01cac"

wifi:
  ssid: !secret wifi_ssid_terrasse
  password: !secret wifi_password
#  use_address: thermostat-chambre-bas-2.local

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Thermostat-Chambre-Bas"
    password: "H0zJpMi7wn1w"

captive_portal:
    
 
mqtt:
  broker: 192.168.1.15
  on_json_message:
    topic: tele/sonoff_chambrebas/SENSOR
    then:
      - sensor.template.publish:
          id: template_temperature
[... 219 lines cut ...]
[10:08:20][C][mqtt:076]:   Log Topic: 'thermostat-chambre-bas/debug'
[10:08:20][C][mqtt:079]:   Availability: 'thermostat-chambre-bas/status'
[10:08:21][C][mqtt.sensor:027]: MQTT Sensor 'chambre bas temperature sensor':
[10:08:21][C][mqtt.sensor:031]:   State Topic: 'thermostat-chambre-bas/sensor/chambre_bas_temperature_sensor/state'
[10:08:21][C][mqtt.sensor:027]: MQTT Sensor 'chambre bas temperature':
[10:08:21][C][mqtt.sensor:031]:   State Topic: 'thermostat-chambre-bas/sensor/chambre_bas_temperature/state'
[10:08:21][C][mqtt_subscribe.sensor:031]: MQTT Subscribe 'chambre bas temperature'
[10:08:21][C][mqtt_subscribe.sensor:031]:   State Class: ''
[10:08:21][C][mqtt_subscribe.sensor:031]:   Unit of Measurement: ''
[10:08:21][C][mqtt_subscribe.sensor:031]:   Accuracy Decimals: 1
[10:08:21][C][mqtt_subscribe.sensor:032]:   Topic: Thermostat-Chambre-Bas/sensor
[10:08:21][C][mqtt.sensor:027]: MQTT Sensor 'Chambre bas vanne':
[10:08:21][C][mqtt.sensor:031]:   State Topic: 'thermostat-chambre-bas/sensor/chambre_bas_vanne/state'
[10:08:21][C][mqtt.sensor:027]: MQTT Sensor 'Chambre bas vanne calculated':
[10:08:21][C][mqtt.sensor:031]:   State Topic: 'thermostat-chambre-bas/sensor/chambre_bas_vanne_calculated/state'
[10:08:21][C][mqtt.switch:041]: MQTT Switch 'chambre bas heater': 
[10:08:21][C][mqtt.switch:042]:   State Topic: 'thermostat-chambre-bas/switch/chambre_bas_heater/state'
[10:08:21][C][mqtt.switch:042]:   Command Topic: 'thermostat-chambre-bas/switch/chambre_bas_heater/command'
[10:08:21][C][mqtt.sensor:027]: MQTT Sensor 'chambre bas proportional':
[10:08:21][C][mqtt.sensor:031]:   State Topic: 'thermostat-chambre-bas/sensor/chambre_bas_proportional/state'
[10:08:21][C][mqtt.sensor:027]: MQTT Sensor 'chambre bas integral':
[10:08:21][C][mqtt.sensor:031]:   State Topic: 'thermostat-chambre-bas/sensor/chambre_bas_integral/state'
[10:08:21][C][mqtt.sensor:027]: MQTT Sensor 'chambre bas derivative':
[10:08:21][C][mqtt.sensor:031]:   State Topic: 'thermostat-chambre-bas/sensor/chambre_bas_derivative/state'
```

Esc4pe, 2022-04-28:
Hi.
I saw the same issue with my climate entity but as well with my lights. 
Buttons, switches, sensors, binary_sensors kept working.

What I found out so far:
- when I downgrade back to 2022.3.2 it works again, so it must be the update
- the discovery message on MQTT looks different

2022.3.2
---------------
{"schema":"json","clrm":true,"supported_color_modes":["rgb"],"brightness":true,"effect":true,"fx_list":["Slow Random","Fast Random","Flicker Effect With Custom Values","None"],"name":"Frederick Nachtlicht","stat_t":"danziger/frederick/nachtlicht/light/frederick_nachtlicht/state","cmd_t":"danziger/frederick/nachtlicht/light/frederick_nachtlicht/command","avty_t":"danziger/frederick/nachtlicht/status","uniq_id":"ESPlightfrederick_nachtlicht","dev":{"ids":"84f3ebe421b4","name":"fredericknachtlicht","sw":"esphome v2022.3.2 Apr 28 2022, 19:39:06","mdl":"nodemcuv2","mf":"espressif"}}

2022.4.0
---------------
{"schema":"json","clrm":true,"supported_color_modes":["rgb"],"brightness":true,"effect":true,"fx_list":["Slow Random","Fast Random","Flicker Effect With Custom Values","None"],"name":"Frederick Nachtlicht","stat_t":"danziger/frederick/nachtlicht/light/frederick_nachtlicht/state","cmd_t":"danziger/frederick/nachtlicht/light/frederick_nachtlicht/command","avty_t":"danziger/frederick/nachtlicht/status","uniq_id":null}

As you can see there is a whole bunch of things missing in the new version starting with "uniq_id":null 

Hope that helps.

gefkuz, 2022-04-28:
How Can you downgrade in ha?

Esc4pe, 2022-04-28:
Not sure. 
I am using docker-compose for HA and ESPhome (+ a few other things). There I can set the specific version to be used.

itkfilelor, 2022-05-02:
> Not sure. I am using docker-compose for HA and ESPhome (+ a few other things). There I can set the specific version to be used.

you can do the same with docker run commands as well with the same image syntax 
`esphome/esphome:2022.3.2`

kquinsland, 2022-05-05:
Your slightly mutilated autoconfig payload looks similar to mine. See #3264 . It could be that #3256 is also related.

kquinsland, 2022-05-05:
> How Can you downgrade in ha?

How did you install it? If you are using very new version of python/pip, you can `❯ pip install esphome==2022.3.2`

norriscm, 2022-05-06:
I was able to resolve a related issue by reverting https://github.com/esphome/esphome/pull/3384/commits/3a9410f5f6342d56e0a752ac3123b5ab66012882

That commit causes the json component to allocate a buffer that is too small to hold the config message, which ends up truncated and invalid as a result.

closed 2022-09-18 by github-actions[bot]