esphome/issues #3252: bblanchon/ArduinoJson@6.18.5 lib dep has changed names breaking builds

blutack, 2022-04-28:
### The problem

I am trying to build my previously working ESPHome project, after cleaning the project.

I now get the following error whenever attempting to build:
```
Library Manager: Installing bblanchon/ArduinoJson @ 6.18.5
Warning! Could not find the package with 'bblanchon/ArduinoJson @ 6.18.5' requirements for your system 'linux_x86_64'
```

And indeed, visiting https://registry.platformio.org/packages/libraries/bblanchon/ArduinoJson gives a 404. Instead, it appears that platformio or bblanchon have moved the library to https://registry.platformio.org/libraries/platformio@benoitblanchon.fr/ArduinoJson.

Thanks for esphome and for taking a look!

### Which version of ESPHome has the issue?

2022.4.0

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

n/a

### What platform are you using?

ESP32

### Board

firebeetle

### Component causing the issue

bblanchon/ArduinoJson

gazoodle, 2022-04-28:
I dislike "me too" posts, but in this case it's pulled me to a full stop, so "me too". I'm looking for a workaround and if I find anything I'll post it here.

I did happen to notice that there is an OttoWinter fork of that library that states it's for ESPHome to make the build process easier ... ottowinter/ArduinoJson-esphomelib@^6.15.2

mmakaay, 2022-04-28:
That's what I an into as well this morning.
I hot patched it by using simply:

```python
cg.add_library("ArduinoJson", "6.18.5")
```

mmakaay, 2022-05-01:
Can you retry again without patching the json component?
I noticed that on my setup, I can now compile the code without any changes.
So might have been a temporary glitch.

Mirdinus, 2022-05-01:
Hi @mmakaay, I can confirm that it's working on clean installation once again

blutack, 2022-05-02:
Likewise, this is now working for me as well with no changes - presumably it was fixed by either PIO or upstream.

Thanks everyone!

closed 2022-05-02 by blutack