esphome/issues #3256: MQTT Config on LED strip light incomplete

dashdrum, 2022-04-30:
### The problem

I have an LED strip light using the partition platform (underlyling platform is neopixelbus) and MQTT.  With the April '22 update the MQTT config payload is incomplete, to the point that configuration at Hass restart throws errors.  I have tested using the neopixelbus platform and the fastled platform (with the prescribed framework downgrade).  I have confirmed that the March '22 version did not have this problem.

I have many MQTT devices in ESPHome and haven't noticed any problems with others.  This is my only MQTT light at this time (the rest are Xmas decorations).

### Which version of ESPHome has the issue?

2022.4.0

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2022.4.7

### What platform are you using?

ESP8266

### Board

Wemos D1 Mini

### Component causing the issue

mqtt

### Example YAML snippet

```yaml
mqtt: 
  broker: !secret mqtt_broker
  username: !secret mqtt_username
  password: !secret mqtt_password
  birth_message:
    topic: $device_name/availability
    payload: online
  will_message:
    topic: $device_name/availability
    payload: offline
  on_message:
    topic: $device_name/markers/command
    then:
      - lambda: |-
          if(x=="0") id(marker_setting) = 0;
          if(x=="1") id(marker_setting) = 1;
          if(x=="2") id(marker_setting) = 2;

light:
  - platform: neopixelbus
    type: GRB
    variant: WS2811
    pin: GPIO5
    num_leds: 60
    id: led_circle
    internal: true

  - platform: partition
    id: $device_name
    name: $upper_device_name
    segments:
      - id: led_circle
        from: 30
        to: 59
      - id: led_circle
        from: 0
        to: 29
[... 80 lines cut ...]
    "schema": "json",
    "stat_t": "wall_clock/light/wall_clock/state",
    "supported_color_modes": [
        "rgb"
    ],
    "uniq_id": "ESPlightwall_clock"
}

From Hass:
Logger: homeassistant.util.logging
Source: util/logging.py:114
First occurred: April 29, 2022, 10:02:28 PM (4 occurrences)
Last logged: 9:06:38 AM

Exception in async_discover when dispatching 'mqtt_discovery_new_light_mqtt': ({'brightness': True, 'color_mode': True, 'effect': True, 'effect_list': ['Clock', 'Clock Backfill', 'Clock Backwards', 'Clock Basic', 'Clock Candy Cane', 'Clock Comet', 'Clock Flicker', 'Clock Fly', 'Clock Glitter', 'Clock Inverse', 'Clock Marquee', 'Clock Tourbillon', 'Color Test'], 'schema': 'json', 'supported_color_modes': ['rgb'], 'platform': 'mqtt'},) Traceback (most recent call last): File "/usr/src/homeassistant/homeassistant/components/mqtt/mixins.py", line 246, in async_discover config = schema(discovery_payload) File "/usr/local/lib/python3.9/site-packages/voluptuous/validators.py", line 218, in __call__ return self._exec((Schema(val) for val in self.validators), v) File "/usr/local/lib/python3.9/site-packages/voluptuous/validators.py", line 341, in _exec raise e if self.msg is None else AllInvalid(self.msg, path=path) File "/usr/local/lib/python3.9/site-packages/voluptuous/validators.py", line 337, in _exec v = func(v) File "/usr/local/lib/python3.9/site-packages/voluptuous/schema_builder.py", line 272, in __call__ return self._compiled([], data) File "/usr/local/lib/python3.9/site-packages/voluptuous/schema_builder.py", line 817, in validate_callable return schema(data) File "/usr/src/homeassistant/homeassistant/components/mqtt/light/__init__.py", line 39, in validate_mqtt_light_discovery return schemas[value[CONF_SCHEMA]](value) File "/usr/local/lib/python3.9/site-packages/voluptuous/validators.py", line 218, in __call__ return self._exec((Schema(val) for val in self.validators), v) File "/usr/local/lib/python3.9/site-packages/voluptuous/validators.py", line 341, in _exec raise e if self.msg is None else AllInvalid(self.msg, path=path) File "/usr/local/lib/python3.9/site-packages/voluptuous/validators.py", line 337, in _exec v = func(v) File "/usr/local/lib/python3.9/site-packages/voluptuous/schema_builder.py", line 272, in __call__ return self._compiled([], data) File "/usr/local/lib/python3.9/site-packages/voluptuous/schema_builder.py", line 817, in validate_callable return schema(data) File "/usr/local/lib/python3.9/site-packages/voluptuous/schema_builder.py", line 272, in __call__ return self._compiled([], data) File "/usr/local/lib/python3.9/site-packages/voluptuous/schema_builder.py", line 594, in validate_dict return base_validate(path, iteritems(data), out) File "/usr/local/lib/python3.9/site-packages/voluptuous/schema_builder.py", line 432, in validate_mapping raise er.MultipleInvalid(errors) voluptuous.error.MultipleInvalid: required key not provided @ data['command_topic']
Exception in async_discover when dispatching 'mqtt_discovery_new_light_mqtt': ({'schema': 'json', 'color_mode': True, 'supported_color_modes': ['rgb'], 'brightness': True, 'effect': True, 'effect_list': ['Clock', 'Clock Backfill', 'Clock Backwards', 'Clock Basic', 'Clock Candy Cane', 'Clock Comet', 'Clock Flicker', 'Clock Fly', 'Clock Glitter', 'Clock Inverse', 'Clock Marquee', 'Clock Tourbillon', 'Color Test', None], 'platform': 'mqtt'},) Traceback (most recent call last): File "/usr/src/homeassistant/homeassistant/components/mqtt/mixins.py", line 246, in async_discover config = schema(discovery_payload) File "/usr/local/lib/python3.9/site-packages/voluptuous/validators.py", line 218, in __call__ return self._exec((Schema(val) for val in self.validators), v) File "/usr/local/lib/python3.9/site-packages/voluptuous/validators.py", line 341, in _exec raise e if self.msg is None else AllInvalid(self.msg, path=path) File "/usr/local/lib/python3.9/site-packages/voluptuous/validators.py", line 337, in _exec v = func(v) File "/usr/local/lib/python3.9/site-packages/voluptuous/schema_builder.py", line 272, in __call__ return self._compiled([], data) File "/usr/local/lib/python3.9/site-packages/voluptuous/schema_builder.py", line 817, in validate_callable return schema(data) File "/usr/src/homeassistant/homeassistant/components/mqtt/light/__init__.py", line 39, in validate_mqtt_light_discovery return schemas[value[CONF_SCHEMA]](value) File "/usr/local/lib/python3.9/site-packages/voluptuous/validators.py", line 218, in __call__ return self._exec((Schema(val) for val in self.validators), v) File "/usr/local/lib/python3.9/site-packages/voluptuous/validators.py", line 341, in _exec raise e if self.msg is None else AllInvalid(self.msg, path=path) File "/usr/local/lib/python3.9/site-packages/voluptuous/validators.py", line 337, in _exec v = func(v) File "/usr/local/lib/python3.9/site-packages/voluptuous/schema_builder.py", line 272, in __call__ return self._compiled([], data) File "/usr/local/lib/python3.9/site-packages/voluptuous/schema_builder.py", line 817, in validate_callable return schema(data) File "/usr/local/lib/python3.9/site-packages/voluptuous/schema_builder.py", line 272, in __call__ return self._compiled([], data) File "/usr/local/lib/python3.9/site-packages/voluptuous/schema_builder.py", line 594, in validate_dict return base_validate(path, iteritems(data), out) File "/usr/local/lib/python3.9/site-packages/voluptuous/schema_builder.py", line 432, in validate_mapping raise er.MultipleInvalid(errors) voluptuous.error.MultipleInvalid: string value is None @ data['effect_list'][13]
Exception in discovery_callback when dispatching 'mqtt_discovery_updated_('light', 'wall_clock wall_clock')': ({'schema': 'json', 'color_mode': True, 'supported_color_modes': ['rgb'], 'brightness': True, 'effect': True, 'effect_list': ['Clock', 'Clock Backfill', 'Clock Backwards', 'Clock Basic', 'Clock Candy Cane', 'Clock Comet', 'Clock Flicker', 'Clock Fly', 'Clock Glitter', 'Clock Inverse', 'Clock Marquee', 'Clock Tourbillon', 'Color Test', None], 'platform': 'mqtt'},) Traceback (most recent call last): File "/usr/src/homeassistant/homeassistant/components/mqtt/mixins.py", line 571, in discovery_callback await self._discovery_update(payload) File "/usr/src/homeassistant/homeassistant/components/mqtt/mixins.py", line 750, in discovery_update config = self.config_schema()(discovery_payload) File "/usr/local/lib/python3.9/site-packages/voluptuous/validators.py", line 218, in __call__ return self._exec((Schema(val) for val in self.validators), v) File "/usr/local/lib/python3.9/site-packages/voluptuous/validators.py", line 341, in _exec raise e if self.msg is None else AllInvalid(self.msg, path=path) File "/usr/local/lib/python3.9/site-packages/voluptuous/validators.py", line 337, [... cut ...] = self.raw_decode(s, idx=_w(s, 0).end()) File "/usr/local/lib/python3.9/json/decoder.py", line 355, in raw_decode raise JSONDecodeError("Expecting value", s, err.value) from None json.decoder.JSONDecodeError: Expecting value: line 1 column 1 (char 0)
```

### Additional information

The full source code for the light is avaiable at: [https://github.com/dashdrum/esphome/blob/master/wall_clock.yaml](https://github.com/dashdrum/esphome/blob/master/wall_clock.yaml)

Thanks for taking a look at this!

dashdrum, 2022-04-30:
Just found this error in the device boot:

**[json:061] | Could not allocate memory for JSON document! Requested 0 bytes, free heap: 24152**

This could be related to the ongoing JSON work.

JayWll, 2022-05-01:
Can confirm I'm experiencing the same issue, and the JSON object sent over mqtt to homeassistant doesn't seem to contain enough information for the entities to be created/available in HASS. I also had the full strip set to internal, and when I turned this off it showed up as expected, however the partitions did not. Since I'm also running esphome in docker I rolled back to 2022.3.0 and re-flashed the device, which resolved the issue.

ShellAddicted, 2022-05-01:
I also can confirm the issue, in my case the MQTT discovery config was incomplete for a
```yaml
sensor:
  - platform: ntc
```
but ok for several other `switch`.

As a workaround I restored the old "allocate everything and shrink later" logic here:
https://github.com/esphome/esphome/blob/dev/esphome/components/json/json_util.cpp#L29

norriscm, 2022-05-06:
I also have this issue, triggered by a light component with many effects that resulted in a config message well over 512 bytes. I traced the regression to the commit referenced below, which shrank the buffer allocated for JSON output from 5120 to 512; reverting it fixed the issue.

https://github.com/esphome/esphome/pull/3384/commits/3a9410f5f6342d56e0a752ac3123b5ab66012882

dashdrum, 2022-06-10:
Is there anything I can do to help move this issue forward?

closed 2022-10-24 by github-actions[bot]

ssieb, 2022-10-25:
Did you test this after the json changes?

dashdrum, 2022-10-30:
I have tested this using the device from my original post, and it is now working as expected.  The entire MQTT config is included in the message.  Fixed!