esphome/issues #3259: ESP32 projects will not compile

omegacore, 2022-05-01:
### The problem

I started today with building for an esp8266. Then I switched to an esp32 and now all esp32 builds fail.

### Which version of ESPHome has the issue?

2022.4.0

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Anything in the logs that might be useful for us?

```txt
INFO Reading configuration /config/esphome/ctrl-thtr.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing ctrl-thtr (board: esp-wrover-kit; framework: arduino; platform: platformio/espressif32 @ 3.5.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- <AsyncTCP-esphome> 1.2.2
|-- <WiFi> 1.0
|-- <FS> 1.0
|-- <Update> 1.0
|-- <ESPAsyncWebServer-esphome> 2.1.0
|   |-- <AsyncTCP-esphome> 1.2.2
|-- <DNSServer> 1.1.0
|-- <ESPmDNS> 1.0
Compiling /data/ctrl-thtr/.pioenvs/ctrl-thtr/src/esphome/components/binary_sensor/automation.cpp.o
Compiling /data/ctrl-thtr/.pioenvs/ctrl-thtr/src/esphome/components/binary_sensor/binary_sensor.cpp.o
Compiling /data/ctrl-thtr/.pioenvs/ctrl-thtr/src/esphome/components/binary_sensor/filter.cpp.o
Compiling /data/ctrl-thtr/.pioenvs/ctrl-thtr/src/esphome/components/captive_portal/captive_portal.cpp.o
xtensa-esp32-elf-g++: error trying to exec 'as': execvp: No such file or directory
*** [/data/ctrl-thtr/.pioenvs/ctrl-thtr/src/esphome/components/binary_sensor/binary_sensor.cpp.o] Error 1
xtensa-esp32-elf-g++: error trying to exec 'as': execvp: No such file or directory
*** [/data/ctrl-thtr/.pioenvs/ctrl-thtr/src/esphome/components/binary_sensor/automation.cpp.o] Error 1
xtensa-esp32-elf-g++: error trying to exec 'as': execvp: No such file or directory
*** [/data/ctrl-thtr/.pioenvs/ctrl-thtr/src/esphome/components/binary_sensor/filter.cpp.o] Error 1
xtensa-esp32-elf-g++: error trying to exec 'as': execvp: No such file or directory
*** [/data/ctrl-thtr/.pioenvs/ctrl-thtr/src/esphome/components/captive_portal/captive_portal.cpp.o] Error 1
========================== [FAILED] Took 8.14 seconds ==========================
```

martgras, 2022-05-01:
Try if cleaning the build folder resolves the problem

omegacore, 2022-05-01:
I used the clean option from the webui to no effect.
`INFO Reading configuration /config/esphome/ctrl-thtr.yaml...
INFO Deleting /data/ctrl-thtr/.pioenvs
INFO Done!`

omegacore, 2022-05-01:
Another note on this issue, previously built projects using esp32 now fail to compile.

martgras, 2022-05-02:
sounds like a python related issue. 
Personally I prefer using a virtual envorionment for esphome under Windows

For troubleshooting the Discord Chat: https://discord.gg/KhAMKrd is the best place.

closed 2022-09-15 by github-actions[bot]